riscv-collab/riscv-gnu-toolchain #1210: MUSL build fails outside of /tmp directory

rkabrick, 2023-03-10:
Hello,

I've had issues trying to build the toolchain using `musl` on several different machines (ubuntu-20.04. ubuntu-22.04, debian sid, arch). 

### Error
```
In file included from -------------------------------/riscv-gnu-toolchain/gcc/libgcc/unwind-dw2.c:412:
./md-unwind-support.h: In function 'riscv_fallback_frame_state':
./md-unwind-support.h:67:6: warning: assignment to 'struct sigcontext *' from incompatible pointer type 'mcontext_t *' [-Wincompatible-pointer-types]
   67 |   sc = &rt_->uc.uc_mcontext;
      |      ^
Makefile:875: warning: overriding recipe for target 'all-multi'
Makefile:866: warning: ignoring old recipe for target 'all-multi'
Makefile:875: warning: overriding recipe for target 'all-multi'
Makefile:866: warning: ignoring old recipe for target 'all-multi'
Makefile:875: warning: overriding recipe for target 'all-multi'
Makefile:866: warning: ignoring old recipe for target 'all-multi'
-------------------------------/riscv-gnu-toolchain/gcc/libgomp/config/linux/affinity.c: In function 'gomp_init_thread_affinity':
-------------------------------/riscv-gnu-toolchain/gcc/libgomp/config/linux/affinity.c:69:3: error: implicit declaration of function 'pthread_attr_setaffinity_np'; did you mean 'pthread_setaffinity_np'? [-Werror=implicit-function-declaration]
   69 |   pthread_attr_setaffinity_np (attr, gomp_cpuset_size,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   pthread_setaffinity_np
cc1: all warnings being treated as errors
make[5]: *** [Makefile:800: affinity.lo] Error 1
make[5]: *** Waiting for unfinished jobs....
configure: WARNING: No native atomic operations are provided for this platform.
configure: WARNING: They will be faked using a mutex.
configure: WARNING: Performance of certain classes will degrade as a result.
make[4]: *** [Makefile:1022: all-recursive] Error 1
make[3]: *** [Makefile:622: all] Error 2
make[2]: *** [Makefile:14999: all-target-libgomp] Error 2
make[2]: *** Waiting for unfinished jobs....
configure: WARNING: execinfo.h: present but cannot be compiled
configure: WARNING: execinfo.h:     check for missing prerequisite headers?
configure: WARNING: execinfo.h: see the Autoconf documentation
configure: WARNING: execinfo.h:     section "Present But Cannot Be Compiled"
configure: WARNING: execinfo.h: proceeding with the compiler's result
make[3]: [Makefile:1785: riscv64-unknown-linux-musl/bits/largefile-config.h] Error 1 (ignored)
make[3]: [Makefile:1786: riscv64-unknown-linux-musl/bits/largefile-config.h] Error 1 (ignored)
make[1]: *** [Makefile:1030: all] Error 2
make: *** [Makefile:805: stamps/build-gcc-musl-stage2] Error 2
```

*Other Error*
```
cc1: error: no include path in which to search for stdc-predef.h
```

### Steps to Reproduce (Fresh Install of Ubuntu/Debian/Arch)
1. Install dependencies outlined in `README.md`
2. `./configure --prefix=$HOME/.local/opt`
3. `make musl`

### Notes
- Upon writing this bug report I pulled the repo into my /tmp directory and built with `./configure --prefix=/tmp/riscv`. To my surprise this built successfully. This also remedied the issue on my colleagues machine and other VMs. 
- I am able to build everything but musl (ie. `make linux` and plain old `make`) just fine no matter where I clone the repo to and where I set the `--prefix` to. 
- If I clone the repo inside my home directory and set `--prefix=/tmp/riscv` it fails as well. 
- I replaced my home directory path with `--------------` in the above error message btw.

I cannot for the life of me figure out why this could be happening. Thank you in advance for any guidance you can offer.

TommyMurphyTM1234, 2023-03-10:
Please post/attach your full `configure` and `make` logs.

TommyMurphyTM1234, 2023-03-11:
I did the following and it worked fine.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1210
cd issue-1210
./configure --prefix=`pwd`/installed-tools
date > build.log && make musl 2>&1 | tee -a build.log && date >> build.log
```
* Config log: [config.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10948145/config.log.tar.gz)
* Build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/10948146/build.log.tar.gz)

TommyMurphyTM1234, 2023-03-13:
Closing as invalid since no reproducible test case has been provided and there is nothing to suggest that this is an issue with the `riscv-gnu-toolchain` build mechanism.