riscv-collab/riscv-gnu-toolchain #1212: make :error in the pull function

peter-samir-luis, 2023-03-13:
i am using ubuntu 22.04
i run the following commands

git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git gcc-demo
sudo apt-get install autoconf automake autotools-dev curl python3 libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build
cd gcc-demo/
mkdir build
cd build/
../configure --prefix=/home/peter/Graduation-Project/riscv-demo --with-arch=rv32gc --with-abi=ilp32d
export PATH="$HOME/Graduation-Project/riscv-demo/bin:$PATH"
sudo make

i got error when i run "sudo make"

[image]

TommyMurphyTM1234, 2023-03-13:
I cannot reproduce this using your instructions so I can only assume that it's a local or transient repo issue as it was here?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480#issuecomment-564410068

Try again?

peter-samir-luis, 2023-03-13:
i tried this command like 3 days ago, and still have the same error

TommyMurphyTM1234, 2023-03-13:
What happens if you try to clone recursively and with verbose progress info?
```
git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain.git
```

TommyMurphyTM1234, 2023-03-13:
And maybe try on a different host and/or network connection?

TommyMurphyTM1234, 2023-03-13:
I've run your commands on my Zorin OS 16.2 Lite (Ubuntu 22.04 derivative) and the build process proceeds as expected.

TommyMurphyTM1234, 2023-03-15:
Closing this issue as it's not reproducible.
If you have a reproducible test case then please provide it and reopen the issue.

closed 2023-03-15 by TommyMurphyTM1234

peter-samir-luis, 2023-05-06:
I ran the command you gave me "git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain.git" instead of the last command "sudo make", is this ok? or do I have to re-configure again?

[image]

TommyMurphyTM1234, 2023-05-06:
> I ran the command you gave me "git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain.git" instead of the last command "sudo make", is this ok? or do I have to re-configure again?

I don't understand what you mean.
To build the toolchain you obviously have to `clone`, `configure` and `make`.
Cloning does just that - clones the repo - and nothing more - in particular configuration and building.

So, to build the toolchain, you need to do this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools ...
make 
```
or this:
```
git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools ...
make
```
or something similar.

peter-samir-luis, 2023-05-06:
please i want to use riscv32 with zicsr extension (in order to access CSR registers )
is this the right configuration
./configure --prefix=$RISCV --with-arch=rv32izicsr --with-abi=ilp32

TommyMurphyTM1234, 2023-05-06:
> please i want to use riscv32 with zicsr extension (in order to access CSR registers ) is this the right configuration ./configure --prefix=$RISCV --with-arch=rv32izicsr --with-abi=ilp32

I don't understand what this has got to do with the original issue reported here?

In any case, you should use:
```
./configure --prefix=$RISCV --with-arch=rv32i_zicsr --with-abi=ilp32
```
Please only report suspected bugs here and provide adequate info to reproduce and investigate them.
For general questions and support you should use the mailing lists and other fora.