riscv-collab/riscv-gnu-toolchain #1213: Using libopcodes

ImanHosseini, 2023-03-14:
I was following this work: https://blog.yossarian.net/2019/05/18/Basic-disassembly-with-libopcodes 
On using libopcodes to disassemble raw bytes. I have added new instructions to my fork of riscv-gnu and I wanted to be able to disassemble raw bytes, taking into account those instructions I added, and this looked like the best solution. TLDR: I am trying to build a "X86" elf which uses "dis-asm.h" header and the "libopcode.a" to disassemble raw bytes into RISCV32 instructions.
The problem though is I can't get to compile my code, in that article I was reading the compilation goes like:
```
clang test.c -lopcodes -o test
```
How I am doing it is:
```
gcc test.c ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a -lbfd -o test
```
But I get:
```
/usr/bin/ld: ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a(riscv-dis.o): in function `parse_riscv_dis_option':
/home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:117: undefined reference to `riscv_priv_specs'
/usr/bin/ld: ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a(riscv-dis.o): in function `riscv_disassemble_insn':
/home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:641: undefined reference to `riscv_subset_supports'
/usr/bin/ld: /home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:656: undefined reference to `riscv_multi_subset_supports'
/usr/bin/ld: ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a(riscv-dis.o): in function `riscv_get_disassembler':
/home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1022: undefined reference to `riscv_release_subset_list'
/usr/bin/ld: /home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1023: undefined reference to `riscv_parse_subset'
/usr/bin/ld: /home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1014: undefined reference to `riscv_get_priv_spec_class_from_numbers'
/usr/bin/ld: ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a(riscv-dis.o): in function `riscv_symbol_is_valid':
/home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1042: undefined reference to `riscv_elf_is_mapping_symbols'
/usr/bin/ld: ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a(riscv-dis.o): in function `disassembler_options_riscv':
/home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1102: undefined reference to `riscv_priv_specs'
/usr/bin/ld: /home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1102: undefined reference to `riscv_priv_specs'
/usr/bin/ld: /home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1102: undefined reference to `riscv_priv_specs'
/usr/bin/ld: /home/iman/projs/riscv-gnu-toolchain/binutils/opcodes/riscv-dis.c:1102: undefined reference to `riscv_priv_specs
```

TommyMurphyTM1234, 2023-03-15:
Please provide a reproducible test case including code that can be compiled to reproduce this issue - e.g. `test.c` here or at least a stub that reproduces the issue.
> `gcc test.c ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a -lbfd -o test`

ImanHosseini, 2023-03-15:
the code is the same as the sample in the blogpost, except I changed it to disassemble RISCV:
```
#define _GNU_SOURCE /* asprintf, vasprintf */

#include <stdarg.h>
#include <stdbool.h>
#include <stddef.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>

#include <dis-asm.h>

// clang test.c -lopcodes -o test

typedef struct {
  char *insn_buffer;
  bool reenter;
} stream_state;

/* This approach isn't very memory efficient or clear,
 * but it avoids external size/buffer tracking in this
 * example.
 */
static int dis_fprintf(void *stream, const char *fmt, ...) {
  stream_state *ss = (stream_state *)stream;

  va_list arg;
  va_start(arg, fmt);
  if (!ss->reenter) {
    vasprintf(&ss->insn_buffer, fmt, arg);
    ss->reenter = true;
  } else {
    char *tmp;
    vasprintf(&tmp, fmt, arg);

    char *tmp2;
    asprintf(&tmp2, "%s%s", ss->insn_buffer, tmp);
    free(ss->insn_buffer);
    free(tmp);
    ss->insn_buffer = tmp2;
  }
  va_end(arg);

  return 0;
}

char *disassemble_raw(uint8_t *input_buffer, size_t input_buffer_size) {
  char *disassembled = NULL;
  stream_state ss = {};

  disassemble_info disasm_info = {};
  init_disassemble_info(&disasm_info, &ss, dis_fprintf);
// bfd_mach_riscv32, bfd_arch_riscv
  disasm_info.arch = bfd_arch_riscv;
  disasm_info.mach = bfd_mach_riscv32;
  disasm_info.read_memory_func = buffer_read_memory;
  disasm_info.buffer = input_buffer;
  disasm_info.buffer_vma = 0;
  disasm_info.buffer_length = input_buffer_size;
  disassemble_init_for_target(&disasm_info);

  disassembler_ftype disasm;
  disasm = disassembler(bfd_arch_riscv, false, bfd_mach_riscv32, NULL);

  size_t pc = 0;
  while (pc < input_buffer_size) {
    size_t insn_size = disasm(pc, &disasm_info);
    pc += insn_size;
    printf("pc += %zu\n", insn_size);

    if (disassembled == NULL) {
      asprintf(&disassembled, "%s", ss.insn_buffer);
    } else {
      char *tmp;
      asprintf(&tmp, "%s\n%s", disassembled, ss.insn_buffer);
      free(disassembled);
      disassembled = tmp;
    }

    /* Reset the stream state after each instruction decode.
     */
    free(ss.insn_buffer);
    ss.reenter = false;
  }

  return disassembled;
}

int main(int argc, char const *argv[]) {
  uint8_t input_buffer[] = {
      0x55,             /* push rbp */
      0x48, 0x89, 0xe5, /* mov rbp, rsp */
      0x89, 0x7d, 0xfc, /* mov DWORD PTR [rbp-0x4], edi */
      0x8b, 0x45, 0xfc, /* mov eax, DWORD PTR [rbp-0x4] */
      0x0f, 0xaf, 0xc0, /* imul eax, rax */
      0x5d,             /* pop ebp */
      0xc3,             /* ret */
  };
  size_t input_buffer_size = sizeof(input_buffer);

  char *disassembled = disassemble_raw(input_buffer, input_buffer_size);
  puts(disassembled);
  free(disassembled);

  return 0;
}
```
TLDR: its just calling into libbfd's "disassembler" stuff. O0 does not matter, the part that is in _my_ code has sufficient info, the '??' part is where it tries to jump into the code in libbfd/libopcodes stuff - which is already compiled, and I am just linking it in -

TommyMurphyTM1234, 2023-03-15:
You don't clarify what `dis-asm.h` you are using - e.g. one of these or something else:
```
tommy@kingfisher:~/Downloads/riscv-gnu-toolchain$ find . -name "dis-asm.h" -print
./binutils/include/dis-asm.h
./gdb/include/dis-asm.h
./newlib/include/dis-asm.h
```
You really need to provide a self-contained reproducible test case if you want assistance here.
Expecting others to reverse engineer your example is not really fair.

ImanHosseini, 2023-03-16:
Sorry if the question was not clear & thank you for your response. It is self-contained though, the example code is there, and there is no makefile, just that 'gcc' command. The example code shouldn't require reverse engineering, it's short and clear: it uses libopcodes to disassemble raw bytes. It should build on any system, and compiled binary then lead to the segfault. I discussed the motivation and the context, I thought maybe someone here knows how to use libopcodes to disassemble raw bytes or has any leads on why it works for '-lopcodes' and not the riscv-gnu libopcodes.a.

Debugging, by nature, involves ambiguity, I would not be asking if I knew what exactly was wrong. The example above already works for x86 with '-lopcodes' (instead of specifying the libopcodes.a under /riscv-gnu-toolchain). When I run the compilation command above, the header that is used is the one under '/usr/include' and not any of the headers inside riscv-gnu-toolchain. I changed it to use the headers you mentioned (only 'riscv-gdb/include/dis-asm.h' compiles) and it still segfaults.

TommyMurphyTM1234, 2023-03-16:
The test case is NOT self contained.
Please provide a self contained test case that reproduces the problem.

ImanHosseini, 2023-03-16:
a.c:
```
#define _GNU_SOURCE /* asprintf, vasprintf */

#include <stdarg.h>
#include <stdbool.h>
#include <stddef.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>

#include <dis-asm.h>

// clang test.c -lopcodes -o test

typedef struct {
  char *insn_buffer;
  bool reenter;
} stream_state;

/* This approach isn't very memory efficient or clear,
 * but it avoids external size/buffer tracking in this
 * example.
 */
static int dis_fprintf(void *stream, const char *fmt, ...) {
  stream_state *ss = (stream_state *)stream;

  va_list arg;
  va_start(arg, fmt);
  if (!ss->reenter) {
    vasprintf(&ss->insn_buffer, fmt, arg);
    ss->reenter = true;
  } else {
    char *tmp;
    vasprintf(&tmp, fmt, arg);

    char *tmp2;
    asprintf(&tmp2, "%s%s", ss->insn_buffer, tmp);
    free(ss->insn_buffer);
    free(tmp);
    ss->insn_buffer = tmp2;
  }
  va_end(arg);

  return 0;
}

char *disassemble_raw(uint8_t *input_buffer, size_t input_buffer_size) {
  char *disassembled = NULL;
  stream_state ss = {};

  disassemble_info disasm_info = {};
  init_disassemble_info(&disasm_info, &ss, dis_fprintf);
// bfd_mach_riscv32, bfd_arch_riscv
  disasm_info.arch = bfd_arch_riscv;
  disasm_info.mach = bfd_mach_riscv32;
  disasm_info.read_memory_func = buffer_read_memory;
  disasm_info.buffer = input_buffer;
  disasm_info.buffer_vma = 0;
  disasm_info.buffer_length = input_buffer_size;
  disassemble_init_for_target(&disasm_info);

  disassembler_ftype disasm;
  disasm = disassembler(bfd_arch_riscv, false, bfd_mach_riscv32, NULL);

  size_t pc = 0;
  while (pc < input_buffer_size) {
    size_t insn_size = disasm(pc, &disasm_info);
    pc += insn_size;
    printf("pc += %zu\n", insn_size);

    if (disassembled == NULL) {
      asprintf(&disassembled, "%s", ss.insn_buffer);
    } else {
      char *tmp;
      asprintf(&tmp, "%s\n%s", disassembled, ss.insn_buffer);
      free(disassembled);
      disassembled = tmp;
    }

    /* Reset the stream state after each instruction decode.
     */
    free(ss.insn_buffer);
    ss.reenter = false;
  }

  return disassembled;
}

int main(int argc, char const *argv[]) {
  uint8_t input_buffer[] = {
      0x55,             /* push rbp */
      0x48, 0x89, 0xe5, /* mov rbp, rsp */
      0x89, 0x7d, 0xfc, /* mov DWORD PTR [rbp-0x4], edi */
      0x8b, 0x45, 0xfc, /* mov eax, DWORD PTR [rbp-0x4] */
      0x0f, 0xaf, 0xc0, /* imul eax, rax */
      0x5d,             /* pop ebp */
      0xc3,             /* ret */
  };
  size_t input_buffer_size = sizeof(input_buffer);

  char *disassembled = disassemble_raw(input_buffer, input_buffer_size);
  puts(disassembled);
  free(disassembled);

  return 0;
}
```
compile as:
```
gcc a.c ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a ./riscv-gnu-toolchain/build-binutils-newlib/bfd/libbfd.a ./riscv-gnu-toolchain/build-binutils-newlib/libiberty/libiberty.a -lz -ldl -o a
```
Which part of it is not clear? Did it compile? Did it not compile? Did it segfault?

TommyMurphyTM1234, 2023-03-16:
```
tommy@kingfisher:~/Downloads/riscv-gnu-toolchain/installed-tools/bin$ gcc a.c ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a ./riscv-gnu-toolchain/build-binutils-newlib/bfd/libbfd.a ./riscv-gnu-toolchain/build-binutils-newlib/libiberty/libiberty.a -lz -ldl -o a
gcc: error: ./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a: No such file or directory
gcc: error: ./riscv-gnu-toolchain/build-binutils-newlib/bfd/libbfd.a: No such file or directory
gcc: error: ./riscv-gnu-toolchain/build-binutils-newlib/libiberty/libiberty.a: No such file or directory
```
Even if I adjust the paths...
```
tommy@kingfisher:~/Downloads/riscv-gnu-toolchain/installed-tools/bin$ gcc a.c ../../../riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a ../../../riscv-gnu-toolchain/build-binutils-newlib/bfd/libbfd.a ../../../riscv-gnu-toolchain/build-binutils-newlib/libiberty/libiberty.a -lz -ldl -o a
a.c:10:10: fatal error: dis-asm.h: No such file or directory
   10 | #include <dis-asm.h>
      |          ^~~~~~~~~~~
compilation terminated.
```
So you are NOT providing sufficient information to reproduce your issue but you seem OK with pushing responsibility onto others to try to reverse engineer it. Totally unacceptable.

ImanHosseini, 2023-03-17:
I understand, but I am only assuming that people reproducing this adjust the paths to the libraries (i.e. 'libopcodes.a'), I cannot possibly know where they have their build files and "./riscv-gnu-toolchain/build-binutils-newlib/opcodes/libopcodes.a" should be sufficient, they just adjust it to point to wherever their 'riscv-gnu-toolchain' is located. It's not some dark magic reverse-engineering vodoo. 'dis-asm.h' is located at './riscv-gnu-toolchain/riscv-gdb/include/dis-asm.h' which again, has to be adjusted according to where your toolchain is located.

closed 2023-03-20 by TommyMurphyTM1234

kito-cheng, 2023-03-20:
@ImanHosseini We have a small project are reusing libopcode, and I spend few time to make it workable with your codes, the code work to me, so...no warranty :P

Makefile
```
BINUTILS_SRC:=$(CURDIR)/binutils
BINUTILS_REPO:=git@github.com:bminor/binutils-gdb.git
BINUTILS_BRANCH:=master
CXXFLAGS:=-I $(BINUTILS_SRC)/include -O -g -std=c++17 -I build-binutils/bfd
LDFLAGS:=-L build-binutils/zlib -L build-binutils/libiberty  build-binutils/opcodes/libopcodes.a build-binutils/bfd//.libs/libbfd.a build-binutils/libsframe/.libs/libsframe.a -liberty -lz -ldl

all: dump-opcode
        ./dump-opcode

stamp/clone-binutils:
        git clone $(BINUTILS_REPO) -b $(BINUTILS_BRANCH) $(BINUTILS_SRC)
        mkdir -p $(dir $@)
        touch $@

stamp/build-binutils: stamp/clone-binutils
        mkdir build-binutils
        cd build-binutils && \
                $(BINUTILS_SRC)/configure \
                        --prefix=$(CURDIR)/install-binutils \
                        --target=riscv64-elf \
                        --disable-gdb \
                        --enable-install-libbfd \
                        CFLAGS="-g3 -O0" CXXFLAGS="-g3 -O0"
        cd build-binutils && $(MAKE)
        cd build-binutils/bfd && $(MAKE) headers
        mkdir -p $(dir $@)
        touch $@

clean:
        rm -rf stamp build-binutils install-binutils \
            dump-opcode dump-opcode.o $(BINUTILS_SRC)

dump-opcode: dump-opcode.o stamp/build-binutils
        $(CXX) $< $(LDFLAGS) -o $@

dump-opcode.o: dump-opcode.cc stamp/build-binutils
        $(CXX) $(CXXFLAGS) -c $< -o $@

```

dump-opcode.cc (mostly copy from your code.)
```
#define _GNU_SOURCE /* asprintf, vasprintf */

#include <config.h>
#include <dis-asm.h>

#include <stdarg.h>
#include <stdbool.h>
#include <stddef.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>

// clang test.c -lopcodes -o test

typedef struct {
  char *insn_buffer;
  bool reenter;
} stream_state;

/* This approach isn't very memory efficient or clear,
 * but it avoids external size/buffer tracking in this
 * example.
 */
static int dis_fprintf(void *stream, const char *fmt, ...) {
  stream_state *ss = (stream_state *)stream;

  va_list arg;
  va_start(arg, fmt);
  if (!ss->reenter) {
    vasprintf(&ss->insn_buffer, fmt, arg);
    ss->reenter = true;
  } else {
    char *tmp;
    vasprintf(&tmp, fmt, arg);

    char *tmp2;
    asprintf(&tmp2, "%s%s", ss->insn_buffer, tmp);
    free(ss->insn_buffer);
    free(tmp);
    ss->insn_buffer = tmp2;
  }
  va_end(arg);

  return 0;
}

static int dis_fprintf_style(void *stream, enum disassembler_style, const char *fmt, ...) {
  stream_state *ss = (stream_state *)stream;

  va_list arg;
  va_start(arg, fmt);
  if (!ss->reenter) {
    vasprintf(&ss->insn_buffer, fmt, arg);
    ss->reenter = true;
  } else {
    char *tmp;
    vasprintf(&tmp, fmt, arg);

    char *tmp2;
    asprintf(&tmp2, "%s%s", ss->insn_buffer, tmp);
    free(ss->insn_buffer);
    free(tmp);
    ss->insn_buffer = tmp2;
  }
  va_end(arg);

  return 0;
}

char *disassemble_raw(uint8_t *input_buffer, size_t input_buffer_size) {
  char *disassembled = NULL;
  stream_state ss = {};

  disassemble_info disasm_info = {};
  init_disassemble_info(&disasm_info, &ss, dis_fprintf, dis_fprintf_style);
// bfd_mach_riscv32, bfd_arch_riscv
  disasm_info.arch = bfd_arch_riscv;
  disasm_info.mach = bfd_mach_riscv32;
  disasm_info.read_memory_func = buffer_read_memory;
  disasm_info.buffer = input_buffer;
  disasm_info.buffer_vma = 0;
  disasm_info.buffer_length = input_buffer_size;
  disassemble_init_for_target(&disasm_info);

  disassembler_ftype disasm;
  disasm = disassembler(bfd_arch_riscv, false, bfd_mach_riscv32, NULL);

  size_t pc = 0;
  while (pc < input_buffer_size) {
    size_t insn_size = disasm(pc, &disasm_info);
    pc += insn_size;
    printf("pc += %zu\n", insn_size);

    if (disassembled == NULL) {
      asprintf(&disassembled, "%s", ss.insn_buffer);
    } else {
      char *tmp;
      asprintf(&tmp, "%s\n%s", disassembled, ss.insn_buffer);
      free(disassembled);
      disassembled = tmp;
    }

    /* Reset the stream state after each instruction decode.
     */
    free(ss.insn_buffer);
    ss.reenter = false;
  }

  return disassembled;
}

int main(int argc, char const *argv[]) {
  uint8_t input_buffer[] = {
      0x55,             /* push rbp */
      0x48, 0x89, 0xe5, /* mov rbp, rsp */
      0x89, 0x7d, 0xfc, /* mov DWORD PTR [rbp-0x4], edi */
      0x8b, 0x45, 0xfc, /* mov eax, DWORD PTR [rbp-0x4] */
      0x0f, 0xaf, 0xc0, /* imul eax, rax */
      0x5d,             /* pop ebp */
      0xc3,             /* ret */
  };
  size_t input_buffer_size = sizeof(input_buffer);

  char *disassembled = disassemble_raw(input_buffer, input_buffer_size);
  puts(disassembled);
  free(disassembled);

  return 0;
}
```

ImanHosseini, 2023-03-20:
It does work, this is great! Thank you so much @kito-cheng !