riscv-collab/riscv-gnu-toolchain #1214: Error in compiling empty c file with main only

Hager44, 2023-03-15:
[image]

this is the error

TommyMurphyTM1234, 2023-03-15:
How on earth do you expect anybody to do anything with that issue report?
No details of where you got the toolchain or how you configured and built it.
No details of how it's installed or what platform you're using.
No `-v` verbose output to aid with investigation.
Sigh! :-|

I just built a toolchain as follows on Zorin OS Lite 16.2 (a Ubuntu 20.04 derivative):
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib
date > build.log && make 2>&1 | tee -a build.log && date >> build.log
```
And then I used this simple code:
```
/* test.c */
#include <stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello, World\n");
    return 0;
}
```
And compiled it successfully as follows:
```
riscv64-unknown-elf-gcc -v -o test.exe test.c
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/tommy/Downloads/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/tommy/Downloads/riscv-gnu-toolchain/gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.0 () 
COLLECT_GCC_OPTIONS='-v' '-o' 'test.exe' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d' '-misa-spec=2.2' '-march=rv64imafdc'
 /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/cc1 -quiet -v test.c -quiet -dumpbase test.c -dumpbase-ext .c -mtune=rocket -march=rv64imafdc -mabi=lp64d -misa-spec=2.2 -march=rv64imafdc -version -o /tmp/ccBRWMsa.s
GNU C17 () version 12.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 9.4.0, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/include
 /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/include-fixed
 /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C17 () version 12.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 9.4.0, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: ff619fb268449bb4c4a1f26e53ca6669
COLLECT_GCC_OPTIONS='-v' '-o' 'test.exe' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d' '-misa-spec=2.2' '-march=rv64imafdc'
 /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/as -v --traditional-format -march=rv64imafdc -march=rv64imafdc -mabi=lp64d -misa-spec=2.2 -o /tmp/ccOcCDrd.o /tmp/ccBRWMsa.s
GNU assembler version 2.40.0 (riscv64-unknown-elf) using BFD version (GNU Binutils) 2.40.0.20230214
COMPILER_PATH=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/lib/:/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'test.exe' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d' '-misa-spec=2.2' '-march=rv64imafdc' '-dumpdir' 'test.'
 /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/collect2 -plugin /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/liblto_plugin.so -plugin-opt=/home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/12.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/cc1pQYBa.res -plugin-opt=-pass-t [... cut ...] wnloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib /tmp/ccOcCDrd.o -lgcc --start-group -lc -lgloss --end-group -lgcc /home/tommy/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv64-unknown-elf/12.2.0/crtend.o
COLLECT_GCC_OPTIONS='-v' '-o' 'test.exe' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d' '-misa-spec=2.2' '-march=rv64imafdc' '-dumpdir' 'test.'

file test.exe
test.exe: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), statically linked, not stripped
```

Hager44, 2023-03-16:
I do the same as you did but still the same error in the compiling this is the last command that ran successfully 
date > build.log && make 2>&1 | tee -a build.log && date >> build.log
Then the compiling command failed 
[image]![Uploading 225658421...]

TommyMurphyTM1234, 2023-03-16:
Please provide logs and not screenshots.

TommyMurphyTM1234, 2023-03-16:
Looks to me like your toolchain didn't install properly or something.
It cannot find include directories that should exist.
But, again, you're not really providing much useful info.

Hager44, 2023-03-16:
eman@eman:~/Documents/project/riscv-gnu-toolchain$ riscv64-unknown-elf-gcc -v -o test.exe test.c
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 () 
COLLECT_GCC_OPTIONS='-v' '-o' 'test.exe' '-march=rv64imafdc' '-mabi=lp64d'
 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/cc1 -quiet -v -imultilib rv64imafdc/lp64d test.c -quiet -dumpbase test.c -march=rv64imafdc -mabi=lp64d -auxbase test -version -o /tmp/ccYFZ81F.s
GNU C17 () version 10.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 10.2.0, GMP version 6.2.0, MPFR version 4.1.0, MPC version 1.2.0-rc1, isl version none
warning: GMP header version 6.2.0 differs from library version 6.2.1.
warning: MPC header version 1.2.0-rc1 differs from library version 1.2.1.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/sys-include"
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/include
 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/include-fixed
End of search list.
GNU C17 () version 10.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 10.2.0, GMP version 6.2.0, MPFR version 4.1.0, MPC version 1.2.0-rc1, isl version none
warning: GMP header version 6.2.0 differs from library version 6.2.1.
warning: MPC header version 1.2.0-rc1 differs from library version 1.2.1.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: 25503746c4b264fb4a430029e3344b15
test.c:1:10: fatal error: stdio.h: No such file or directory
    1 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.

Hager44, 2023-03-16:
Please could you provide me any video or instructions to install it prperly as I follow this video but I have been trying for more than 4 days but useless could you please help and thanks in advance 

https://youtu.be/sPvMXGFTC2U

TommyMurphyTM1234, 2023-03-16:
I've already posted how to build the toolchain.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1214#issuecomment-1470435974

If your toolchain isn't working as expected then you need to capture, zip, and attach your toolchain build (`git clove`, `configure` and `make`) details.

TommyMurphyTM1234, 2023-03-20:
No further update so I am closing this as no issue with the `riscv-gnu-toolchain` has been identified.

closed 2023-03-20 by TommyMurphyTM1234