riscv-collab/riscv-gnu-toolchain #1215: Can't build bitmanip multilib with linux 64 bit toolchain

amitch1999, 2023-03-19:
Hi, I'm trying to build a linux 64 bit toolchain with the following multilib permutations: rv64imac-lp64, rv64imac_zbs_zbc_zbb_zba-lp64, rv64imafdc-lp64d, rv64imafdc_zbs_zbc_zbb_zba-lp64d.
Build from source hash: `170a9a3045207e3ed4d396f6c4f03e1c7978b0e4`
my config line:
`../riscv-gnu-toolchain/configure --with-arch=rv64imafdc --with-abi=lp64d --enable-multilib --with-multilib-generator="rv64imac-lp64-- rv64imac_zbs_zbc_zbb_zba-lp64-- rv64imafdc-lp64d-- rv64imafdc_zbs_zbc_zbb_zba-lp64d" --prefix=/scratch/riscv/gnu-workspace/install-collab-linux-64-bitmanip`
And `make linux` after.
The build finishes fine with no errors, but when I use `-print-multi-lib` on gcc all I get is:
`.;
lib32/ilp32;@march=rv32imac@mabi=ilp32
lib32/ilp32d;@march=rv32imafdc@mabi=ilp32d
lib64/lp64;@march=rv64imac@mabi=lp64`
Without the bitmanip multilib I requested in the config line, and with 32 bit libraries that were not listed.
This is also the same output I get from `-print-multi-lib` when I config and build with no `--with-multilib-generator` at all, and only `--enable-multilib` as though it ignores `--with-multilib-generator`

TommyMurphyTM1234, 2023-03-19:
`--with-multilib-generator` only works for the bare metal toolchain at the moment. See here:

* https://github.com/riscv-collab/riscv-gnu-toolchain#build-with-customized-multi-lib-configure

For the Linux toolchain you'll have to edit/override the `t-linux-multilib` file and use `configure ... --enable-multilib`.

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/t-elf-multilib

TommyMurphyTM1234, 2023-03-19:
> my config line: `../riscv-gnu-toolchain/configure --with-arch=rv64imafdc --with-abi=lp64d --enable-multilib --with-multilib-generator="rv64imac-lp64-- rv64imac_zbs_zbc_zbb_zba-lp64-- [...]

Note that `--with-arch=rv64imafdc --with-abi=lp64d ` is the default native `arch/abi` for the toolchain so you don't actually have to specify them at configuration time.

TommyMurphyTM1234, 2023-03-20:
This is how I would approach building the multilib toolchain that you need:
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1215
cd issue-1215
pushd
cd gcc/gcc/config/riscv
./multilib-generator rv64imac-lp64-- rv64imac_zbs_zbc_zbb_zba-lp64-- rv64imafdc_zbs_zbc_zbb_zba-lp64d-- > t-linux-multilib
popd
./configure --prefix=`pwd`/installed-tools --enable-multilib
date > build.log && make linux 2>&1 | tee -a build.log && date >> build.log
```
Note that there is no need to specify the multilib `rv64imafdc-lp64d--` since `rv64imafdc/lp64d` aka `rv64gc/lp64d` is the default `arch/abi` for the toolchain in the absence of this being overridden at `configure` time using `--with-arch=... --with-abi=...` - so the libraries/startup code/etc. for `rv64gc/lp64d` are generated automatically anyway.

TommyMurphyTM1234, 2023-03-20:
Hmmmm... I tried that myself and it failed for some reason.
Oddly attempting to build some rv32 stuff which doesn't make sense to me since no rv32 multilibs were specified.
```
a - elf/libc_early_init.os
for symbol in __GI___pthread_disable_asynccancel __GI___pthread_enable_asynccancel __libc_assert_fail __pthread_disable_asynccancel __pthread_enable_asynccancel calloc free malloc realloc  __stack_chk_fail __stack_chk_fail_local; do \
	echo ".globl $symbol"; \
	echo "$symbol:"; \
done | riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32 -o /home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/elf/librtld.mapT.o -g -Werror=undef -Wa,--noexecstack  -c -x assembler -
riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32   -nostdlib -nostartfiles -r -o /home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/elf/librtld.map.o /home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/elf/librtld.mapT.o '-Wl,-(' /home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/elf/dl-allobjs.os /home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/libc_pic.a -lgcc '-Wl,-)' -Wl,-Map,/home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/elf/librtld.mapT
/home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/libgcc.a(div.o)
/home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: attempt to do relocatable link with elf64-littleriscv input and elf32-littleriscv output
/home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/libgcc.a(div.o): file class ELFCLASS64 incompatible with ELFCLASS32
/home/tommy/Downloads/issue-1215/installed-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:1289: /home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32/elf/librtld.map] Error 1
make[3]: Leaving directory '/home/tommy/Downloads/issue-1215/glibc/elf'
make[2]: *** [Makefile:484: elf/subdir_lib] Error 2
make[2]: Leaving directory '/home/tommy/Downloads/issue-1215/glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/home/tommy/Downloads/issue-1215/build-glibc-linux-rv32imac-ilp32'
```
Full `build.log`: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11021080/build.log.tar.gz)

TommyMurphyTM1234, 2023-03-20:
Maybe I'm hitting the same or similar issues here as described in this issue?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1126

amitch1999, 2023-03-21:
Hi Tommy, thanks for the quick response.
After creating a custom `t-linux-multilib` file I encountered the same error as you did, and after looking in issue #1126 you mentioned I've manged to get it build by editing `configure` and changing `glibc_multilib_names` and `newlib_multilib_names' to list my wanted permutations.
And it looks like it did build glibc with my permutations (I saw it created folders named "build-glibc-linux-rv64i...") but now when I invoke `riscv64-unknown-linux-gnu-gcc -print-multi-lib` it's printing only `lib64/lp64;@march=rv64imac@mabi=lp64`.
Now, I'm not sure how to check if it actually built them, and where they are suppose to be in my install folder. Maybe only the print is wrong? and they are actually available?

TommyMurphyTM1234, 2023-03-21:
> Hi Tommy, thanks for the quick response. After creating a custom `t-linux-multilib` file I encountered the same error as you did, and after looking in issue #1126 you mentioned I've manged to get it [...]

If you are building a Linux toolchain then why do you think that you need to change the latter? Newlib has nothing to do with the Linux toolchain.

> And it looks like it did build glibc with my permutations (I saw it created folders named "build-glibc-linux-rv64i...") but now when I invoke `riscv64-unknown-linux-gnu-gcc -print-multi-lib` it's [...]

If `-print-multi-lib` is not listing your required multilibs then, even if multilibs got built, it sounds like the toolchain knowns nothing about them.

Unfortunately, there seem to be several fundamental problems with building certain multilib toolchains and I don't understand what they are.

TommyMurphyTM1234, 2023-03-23:
> Now, I'm not sure how to check if it actually built them, and where they are suppose to be in my install folder.

For the Linux toolchain multilibs should be in 

* `<prefix-directory>/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/lib32`
* `<prefix-directory>/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/lib64`

TommyMurphyTM1234, 2023-04-01:
Any update @amitch1999?

TommyMurphyTM1234, 2023-10-16:
My suggestion here for overriding the default set of multilibs seems to be no longer valid:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1215#issuecomment-1476338836

See here for further background on this:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1354

Instead I took this, arguably slightly hacky, approach:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1215
cd issue-1215
./configure --prefix=`pwd`/installed-tools --enable-multilib
```
I then edited the `Makefile` generated by the `configure` step to change the default set of multilibs:
```
GLIBC_MULTILIB_NAMES := rv32imac-ilp32 rv32imafdc-ilp32d rv64imac-lp64 rv64imafdc-lp64d
```
to this alternative set, appending the bitmanip multilibs to the default set:
```
GLIBC_MULTILIB_NAMES := rv32imac-ilp32 rv32imafdc-ilp32d rv64imac-lp64 rv64imafdc-lp64d rv64imac_zbs_zbc_zbb_zba-lp64 rv64imafdc_zbs_zbc_zbb_zba-lp64d
```
and then ran the build:
```
make linux 2>&1 | tee build.log

./installed-tools/bin/riscv64-unknown-linux-gnu-gcc -dumpspecs

...

*multilib:
. !march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 !mabi=ilp32d !mabi=lp64 !mabi=lp64d;lib32/ilp32:../lib32/ilp32 march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs mabi=ilp32 !mabi=ilp32d !mabi=lp64 !mabi=lp64d;lib32/ilp32d:../lib32/ilp32d !march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 mabi=ilp32d !mabi=lp64 !mabi=lp64d;lib64/lp64:../lib64/lp64 !march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 !mabi=ilp32d mabi=lp64 !mabi=lp64d;lib64/lp64d:../lib64/lp64d !march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 !mabi=ilp32d !mabi=lp64 mabi=lp64d;lib64/lp64:../lib64/lp64 !march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 !mabi=ilp32d mabi=lp64 !mabi=lp64d;lib64/lp64d:../lib64/lp64d !march=rv32imac !march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 !mabi=ilp32d !mabi=lp64 mabi=lp64d;

*multilib_defaults:
march=rv64imafdc_zicsr mabi=lp64d

*multilib_extra:

*multilib_matches:
march=rv32imac march=rv32imac;march=rv32ima march=rv32ima;march=rv32imaf_zicsr march=rv32imaf_zicsr;march=rv32imafc_zicsr march=rv32imafc_zicsr;march=rv32imafd_zicsr march=rv32imafd_zicsr;march=rv32imafdc_zicsr march=rv32imafdc_zicsr;march=rv64imac march=rv64imac;march=rv64ima march=rv64ima;march=rv64imaf_zicsr march=rv64imaf_zicsr;march=rv64imafc_zicsr march=rv64imafc_zicsr;march=rv64imafd_zicsr march=rv64imafd_zicsr;march=rv64imafdc_zicsr march=rv64imafdc_zicsr;march=rv64imac_zba_zbb_zbc_zbs march=rv64imac_zba_zbb_zbc_zbs;march=rv64imafdc_zicsr_zba_zbb_zbc_zbs march=rv64imafdc_zicsr_zba_zbb_zbc_zbs;mabi=ilp32 mabi=ilp32;mabi=ilp32d mabi=ilp32d;mabi=lp64 mabi=lp64;mabi=lp64d mabi=lp64d;

*multilib_exclusions:

*multilib_options:
march=rv32imac/march=rv32ima/march=rv32imaf_zicsr/march=rv32imafc_zicsr/march=rv32imafd_zicsr/march=rv32imafdc_zicsr/march=rv64imac/march=rv64ima/march=rv64imaf_zicsr/march=rv64imafc_zicsr/march=rv64imafd_zicsr/march=rv64imafdc_zicsr/march=rv64imac_zba_zbb_zbc_zbs/march=rv64imafdc_zicsr_zba_zbb_zbc_zbs mabi=ilp32/mabi=ilp32d/mabi=lp64/mabi=lp64d

*multilib_reuse:
lib32/ilp32:../lib32/ilp32 !march=rv32imac march=rv32ima !march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32imafd_zicsr !march=rv32imafdc_zicsr !march=rv64imac !march=rv64ima !march=rv64imaf_zicsr !march=rv64imafc_zicsr !march=rv64imafd_zicsr !march=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs mabi=ilp32 !mabi=ilp32d !mabi=lp64 !mabi=lp64d;lib32/ilp32:../lib32/ilp32 !march=rv32imac !march=rv32ima march=rv32imaf_zicsr !march=rv32imafc_zicsr !march=rv32im [... cut ...] arch=rv64imafdc_zicsr !march=rv64imac_zba_zbb_zbc_zbs !march=rv64imafdc_zicsr_zba_zbb_zbc_zbs !mabi=ilp32 !mabi=ilp32d !mabi=lp64 mabi=lp64d;

...

```
So, using this approach, it is possible to build a multilib toolchain with the required bitmanip multilibs in addition to the default base set of multilibs.

I think that there may still be other issues with the way that multilibs are configured/built, but these are separate from the original issue raised here which I consider to be resolved now.

closed 2023-10-16 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-10-17:
Actually, this summarises the multilib capabilities of the resulting toolchain more clearly/concisely.
Rememer that `.` refers to the toolchain default `arch/abi` - by default, and in this case, `rv64gc/lp64d`.
```
./installed-tools/bin/riscv64-unknown-linux-gnu-gcc -print-multi-lib
.;
lib32/ilp32;@march=rv32imac@mabi=ilp32
lib32/ilp32d;@march=rv32imafdc_zicsr@mabi=ilp32d
lib64/lp64;@march=rv64imac@mabi=lp64
lib64/lp64;@march=rv64imac_zba_zbb_zbc_zbs@mabi=lp64
lib64/lp64d;@march=rv64imafdc_zicsr_zba_zbb_zbc_zbs@mabi=lp64d
```