riscv-collab/riscv-gnu-toolchain #1218: Update configure defaults: `with-isa-spec` to `20191213` and `march` to `rv64gc`

atrib, 2023-03-23:
I suggest changing the default `with-isa-spec` to support the latest version `20191213`, and change the default `march` to `rv64gc` instead of `rv64imafdc`. 

Reasons:
1) This should keep compatibility with programs using `csrr`/`csrw`/`fence.i`, as the definition of `G` is `IMAFDZicsr Zifencei` which includes all of the aforementioned instructions. 
2) This change is in line with the specifications of popular RISC-V cores, including the default `rocketchip`.
3) The current default for `with-isa-spec` option introduced with commit `44fedc5e1a8f202248fc0f2b39c84226d8f3d1e6` is `v2.2`. The commit message says "once we bump to GCC 12 we bump that to 20191213". The gcc submodule has long been bumped to 12.

If the current toolchain has trouble correctly supporting this configuration, can you point me some of the issues so I can try to help iron them out?

pz9115, 2023-04-20:
Hello, I had updated it in https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1229, thanks for your suggestions.

kito-cheng, 2023-04-20:
Thanks @pz9115 :)

closed 2023-04-20 by kito-cheng

atrib, 2023-04-20:
Thanks @kito-cheng @pz9115

tienntr, 2023-05-06:
`README.md` should be updated to reflect new default ISA spec, too.