riscv-collab/riscv-gnu-toolchain #1222: Reopen to issue #264

amitch1999, 2023-03-26:
[issue #264](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/264) was about the build failing on windows msys2.
The fixes [3211aa](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/3211aa154fc41c64db8532a8ea3a993003556841) and [e708c7](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/e708c776877aba27c9a3b659ac539a7e38553a27) seems to relay on the fact that the gcc folder was named `riscv-gcc` but since [bbc2db](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/bbc2dbb6b3e1909bf4c94fa66f125e9b27d3e7a8) the folder is named `gcc` so the issue appears again.

TommyMurphyTM1234, 2023-03-28:
@amitch1999 , can you submit a PR to address this please?

amitch1999, 2023-03-29:
I'm not sure that's this small fix will make the build work on mingw, but this should be fixed nonetheless, so I will submit a PR.

amitch1999, 2023-04-20:
Fix was merged - [PR](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1224)

closed 2023-04-20 by amitch1999