riscv-collab/riscv-gnu-toolchain #1223: Should march=rv64iv be possible if toolchain is built for rv64gc?

angsch, 2023-03-29:
Hello,
The nightly tarball riscv64-glibc-ubuntu-20.04 of this toolchain is built using the default `--with-abi=lp64d --with-arch=rv64gc`. 

I assumed that `rv64gc` would only enable the extensions subsumed by `g` and `c`. However, by setting `-march=rv64iv`, vector instructions can be enabled anyway. Is this the expected behavior?

## Example

```
//main.c
#include <stdio.h>

extern void fcn(float *v);

int main() {
  float v[20];
  for (int i = 0; i < 20; i++)
    v[i] = 1.0;

  fcn(v);

  printf("v[0] = %.6e\n", v[0]);

#if __riscv_v
  printf("v extension enabled\n");
#endif

  return 0;
}
```
```
// zero.S
    .text
    .global fcn

// void fcn(float *v)
//
// register arguments:
// a0:   v

fcn:
    vsetvli t6, zero, e32
    vxor.vv v0, v0, v0
    vse32.v v0, (a0)
    ret
```
### Behavior with march=rv64iv
```
./riscv64-unknown-linux-gnu-gcc -march=rv64iv zero.S main.c -static
qemu-riscv64 -cpu rv64,vext_spec=v1.0,v=true,vlen=128 -B 0x100000 ./a.out
v[0] = 0.000000e+00
v extension enabled
```
The vector assembly instructions are executed and the test flag `__riscv_v` is set.

### Behavior without specifying march
```
./riscv64-unknown-linux-gnu-gcc zero.S main.c -static
zero.S: Assembler messages:
zero.S:10: Error: unrecognized opcode `vsetvli t6,zero,e32', extension `v' or `zve64x' or `zve32x' required
zero.S:11: Error: unrecognized opcode `vxor.vv v0,v0,v0', extension `v' or `zve64x' or `zve32x' required
zero.S:12: Error: unrecognized opcode `vse32.v v0,(a0)', extension `v' or `zve64x' or `zve32x' required
```
Without specifying march, the vector instructions are not recognized, which is what I would have expected.

**Should it be possible to use the vector extension by setting `march` even though the toolchain is built without it?**

## Version info
Used tarball: https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2023.03.14/riscv64-glibc-ubuntu-20.04-nightly-2023.03.14-nightly.tar.gz

```
./riscv64-unknown-linux-gnu-gcc -v
Target: riscv64-unknown-linux-gnu
Configured with: /home/runner/work/riscv-gnu-toolchain/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-pkgversion=g2ee5e430018 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/home/runner/work/riscv-gnu-toolchain/riscv-gnu-toolchain/gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (g2ee5e430018) 
```

TommyMurphyTM1234, 2023-03-29:
> I assumed that `rv64gc` would only enable the extensions subsumed by `g` and `c`. However, by setting `-march=rv64iv`, vector instructions can be enabled anyway. Is this the expected behavior?

Yes. `--with-arch` and `--with-abi` simply specify the **default** `arch/abi` for code generation. But all other formalised/implemented extensions are also supported by the toolchain code generation. (In some cases, e.g. vector extension, the toolchain needs to be built from a specific work in progress branch). However, without multilib support for required `arch/abi` combinations, there will be no startup code or standard libraries to link with. So, in your example, it's expected and not anomalous that the toolchain that defaults to `rv64gc/lp64d` can also generate code for, say, `rv64iv/lp64`. However, if you specified `-march=rv64iv` and not also `-mabi=lp64` then I believe that the effective `arch/abi` would be `rv64iv/lp64d` which is an incompatible combination and should have generated an error?

TommyMurphyTM1234, 2023-03-29:
> @TommyMurphyTM1234 Thank you for the excellent explanation. 👍 

You're welcome.

> The `v` extension implies the `d` extension.

I couldn't see that stated in the V spec.
Can you clarify where you see this stated please?

> ```
> $CC main.c -march=rv64im`
> requested ABI requires '-march' to subsume the 'D' extension
> ```
> 
> I don't have a problem with setting `D`, but I find the error unexpected. Why does integer arithmetic require double-precision instructions?

That's not what the error is trying to say.
You're specifying `-march=rv64im`.
You don't specify any ABI so the default, `lp64d`, is used.
So the effective `arch/abi` that you're using is `rv64im/lp64d`.
This is incompatible - you're asking to use an ABI that depends on the D extension with an architecture that doesn't include the D extension.
> 
> Is there some documentation (for example in the form of a directed acyclic graph) that illustrates all dependencies between extensions?

TommyMurphyTM1234, 2023-03-29:
> The compiler uses instructions such as `fmadd.d` that belong to the `D` extension if `march=rv64iv`. I also recall having seen an issue that an early version of the compiler throws an error if `V` is enabled, but `D` is not. I couldn't find it in my quick search, though.

Ok, I guess that this is it:

* https://github.com/riscv/riscv-v-spec/releases/tag/v1.0

> _The V extension requires the scalar processor implements the F and D extensions, and implements all vector floating-point
 instructions_

> I forgot about the ABI. Oops. Sorry. So `D` has to be set explicitly here, but `$CC main.c -march=rv64iv` works, so `D` must be set implicitly by `V`.

Personally, I dislike the implicit stuff in this context as it leads confusion the like of which is seen in this thread.

angsch, 2023-03-29:
> Personally, I dislike the implicit stuff in this context as it leads confusion the like of which is seen in this thread.

Ditto.

Thank you again for your explanation, really appreciated.

closed 2023-03-29 by angsch

TommyMurphyTM1234, 2023-03-29:
> Thank you again for your explanation, really appreciated.

You're welcome.