riscv-collab/riscv-gnu-toolchain #1225: Why are there:  segment fault?

24geng, 2023-03-31:
The following are error alerts:

/usr/local/include/eigen3/Eigen/src/Core/GenericPacketMath.h:1014:71: internal compiler error: Segmentation fault
 1014 | template <typename Packet,int N=unpacket_traits<Packet>::size> struct PacketBlock {
      |                                                                       ^~~~~~~~~~~
0xf494ff crash_signal
	../.././riscv-gcc/gcc/toplev.cc:322
0x1248ef3 selt
	../.././riscv-gcc/gcc/wide-int.cc:418
0x1248ef3 wi::lts_p_large(long const*, unsigned int, unsigned int, long const*, unsigned int)
	../.././riscv-gcc/gcc/wide-int.cc:494
0x861d14 bool wi::lts_p<generic_wide_int<wi::extended_tree<192> >, generic_wide_int<wi::extended_tree<192> > >(generic_wide_int<wi::extended_tree<192> > const&, generic_wide_int<wi::extended_tree<192> > const&)
	../.././riscv-gcc/gcc/wide-int.h:1927
0x861d14 wi::binary_traits<generic_wide_int<wi::extended_tree<192> >, generic_wide_int<wi::extended_tree<192> >, wi::int_traits<generic_wide_int<wi::extended_tree<192> > >::precision_type, wi::int_traits<generic_wide_int<wi::extended_tree<192> > >::precision_type>::signed_predicate_result operator< <generic_wide_int<wi::extended_tree<192> >, generic_wide_int<wi::extended_tree<192> > >(generic_wide_int<wi::extended_tree<192> > const&, generic_wide_int<wi::extended_tree<192> > const&)
	../.././riscv-gcc/gcc/wide-int.h:3250
0x861d14 tree_int_cst_lt(tree_node const*, tree_node const*)
	../.././riscv-gcc/gcc/tree.h:6341
0x861d14 end_of_class
	../.././riscv-gcc/gcc/cp/class.cc:6440
0x86ada4 include_empty_classes
	../.././riscv-gcc/gcc/cp/class.cc:6573
0x86ada4 layout_class_type
	../.././riscv-gcc/gcc/cp/class.cc:6953
0x86ada4 finish_struct_1(tree_node*)
	../.././riscv-gcc/gcc/cp/class.cc:7502
0x9c80cf instantiate_class_template_1
	../.././riscv-gcc/gcc/cp/pt.cc:12349
0x9c80cf instantiate_class_template(tree_node*)
	../.././riscv-gcc/gcc/cp/pt.cc:12399
0xa02003 complete_type(tree_node*)
	../.././riscv-gcc/gcc/cp/typeck.cc:143
0x9b6d60 tsubst_decl
	../.././riscv-gcc/gcc/cp/pt.cc:14990
0x9ac257 tsubst(tree_node*, tree_node*, int, tree_node*)
	../.././riscv-gcc/gcc/cp/pt.cc:15582
0x9bbb83 tsubst_expr(tree_node*, tree_node*, int, tree_node*, bool)
	../.././riscv-gcc/gcc/cp/pt.cc:18590
0x9b93dc tsubst_expr(tree_node*, tree_node*, int, tree_node*, bool)
	../.././riscv-gcc/gcc/cp/pt.cc:18476
0x9b911f tsubst_expr(tree_node*, tree_node*, int, tree_node*, bool)
	../.././riscv-gcc/gcc/cp/pt.cc:18735
0x9b9b6f tsubst_expr(tree_node*, tree_node*, int, tree_node*, bool)
	../.././riscv-gcc/gcc/cp/pt.cc:18833
0x9bb7b3 tsubst_expr(tree_node*, tree_node*, int, tree_node*, bool)
	../.././riscv-gcc/gcc/cp/pt.cc:18802
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.

TommyMurphyTM1234, 2023-03-31:
You haven't provided sufficient details. For example:

* A reproducible test case
* The code that you are trying to compile
* Where did you get the toolchain or how did you build it
* How you are invoking the toolchain to compile your code
* The OS on which you are working
* Etc.

TommyMurphyTM1234, 2023-04-01:
Any update @24geng?

TommyMurphyTM1234, 2023-04-03:
No update so I'm closing as invalid.
If there's an update with the sort of info that I mentioned above then please reopen it.

closed 2023-04-03 by TommyMurphyTM1234