riscv-collab/riscv-gnu-toolchain #1226: execvp: /bin/sh: Argument list too long error when using with --with-multilib-generator

fanghuaqi, 2023-04-03:
Hello, I want to report an issue with build the compiler using with the `--with-multilib-generator`.

Here is my configure command:

~~~shell
mkdir build && cd build
../configure  --prefix=/work/LocalInstall/v3nlbmulti --enable-multilib  --with-multilib-generator="rv32emc-ilp32e--;rv32emac-ilp32e--;rv32imc-ilp32--;rv32imac-ilp32--zfh*zfhmin*zve32x;rv32imafc-ilp32f--zfh*zfhmin*zve32f;rv32imafdc-ilp32d--zfh*zfhmin*zve32f;rv32imac_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt-ilp32--zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt-ilp32f--zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd-ilp32d--zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imac_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imafc_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafdc_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imaczba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv64imac-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafc-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafdc-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imac_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafc_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafdc_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64ima_zca_zcb_zcmp_zcmt-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafd_zca_zcb_zcf_zcd-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v" --with-arch=rv64ima --with-abi=lp64 --with-cmodel=medany
make -j16
~~~

Then it shows following error when build gcc in build-gcc-newlib-stage1

~~~
if test yes = yes \
   || test -n ""; then \
  /bin/sh /work/gcc/gcc/genmultilib \
...... too long to show, ignore it .........
_zbs_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b_xxldsp/mabi.lp64d march.rv64imafd_zca_zcb_zcd_zcf_zba_zbb_zbc_zbs/mabi.lp64d=march.rv64imafdv_zfhmin_zca_zcb_zcd_zcf_zba_zbb_zbc_zbs_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b_xxldsp_xxldspn1/mabi.lp64d march.rv64imafd_zca_zcb_zcd_zcf_zba_zbb_zbc_zbs/mabi.lp64d=march.rv64imafdv_zfhmin_zca_zcb_zcd_zcf_zba_zbb_zbc_zbs_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b_xxldspn1/mabi.lp64d" \
    "yes" \
    > tmp-mlib.h; \
else \
  /bin/sh /work/gcc/gcc/genmultilib '' '' '' '' '' '' '' '' \
    "" '' no \
    > tmp-mlib.h; \
fi
make[2]: execvp: /bin/sh: Argument list too long
make[2]: *** [Makefile:2252: s-mlib] Error 127
~~~

I searched and found it is caused by the command line is too long exceed linux `ARGMAX` limit, the command itself is defined in Makefile https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/Makefile.in;h=d8b76d83d6850c3ce5318f3acd7cdc2a8cbc140b;hb=refs/heads/master#l2215. 

Because we have a complex multilib generation definition, so the generated multilib configuration variables constructed command exceed the `ARGMAX` limit, so can we have other ways to define the multilib and allow complex multilib generation?

Thanks

TommyMurphyTM1234, 2023-04-03:
Seems like an issue alright.
For the moment can you try overriding the default multilib list

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/t-elf-multilib

by running the `multilib-generator` script

* https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/multilib-generator

to generate your desired multilib list?

TommyMurphyTM1234, 2023-04-03:
> Thanks for your help, I checked here https://github.com/gcc-mirror/gcc/blob/master/gcc/Makefile.in#L1881, even I generate manually by myself, the `t-elf-multilib` will be include as Makefile to provide make variables like **MULTILIB**_* used by https://github.com/gcc-mirror/gcc/blob/master/gcc/Makefile.in#L2215-L2230

Ah, ok. I didn't realise that the info ended up in the same place regardless of its origin.

I'm not familiar with `ARGMAX` and if/how that limit can be extended/overcome.

As a stopgap, maybe it would be possible to do multiple builds with subsets of the multilibs required and then manually merge them? But I've never tried this and I don't know if just dropping the additional multilib folders into a toolchain build will work....

fanghuaqi, 2023-04-03:
> Ah, ok. I didn't realise that the info ended up in the same place regardless of its origin.
>
> I'm not familiar with `ARGMAX` and if/how that limit can be extended/overcome.
>
> As a stopgap, maybe it [...]

Thanks @TommyMurphyTM1234 , since it will need to generate a mutlilib.h used to compile gcc compiler , so make the complex multilib to small subsets maybe not help, since the compiled gcc might not be able to choose correct library

kito-cheng, 2023-04-03:
Hi @fanghuaqi :

I would suggest you can back port following two patches to improving the multi-lib handling:

https://github.com/gcc-mirror/gcc/commit/5ca9980fc86242505ffdaaf62bca1fd5db26550b
https://github.com/gcc-mirror/gcc/commit/d72ca12b846a9f5c01674b280b1817876c77888f

It introduce a new scheme to select the most compatible multi-lib, so you can focus on how many different multi-lib you want, and then forgot the reuse rule part.

I know how pain about handling huge number of multi-lib, we've build near 50 different multilib in our toolchian release and the reuse rule is really terrible without this patch.

fanghuaqi, 2023-04-03:
> 

Thanks a lot @kito-cheng , I will take a lot at these patches, does these patches already in gcc 13 upstream?

TommyMurphyTM1234, 2023-04-03:
From visual observation of the GCC repo those changes seem to be in the master branch so maybe you could just rebuilt with that (e.g. by modifying `.gitmodules`)?

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/170a9a3045207e3ed4d396f6c4f03e1c7978b0e4/.gitmodules#L8

TommyMurphyTM1234, 2023-04-03:
> > 
> 
> Thanks a lot @kito-cheng , I will take a lot at these patches, does these patches already in gcc 13 upstream?
> 
> I have checked the two commits are already in gcc upstream master branch(gcc 13 latest), which is now I am using.

From visual observation of the GCC repo

* https://github.com/gcc-mirror/gcc

 those changes seem to be in the master branch so maybe you could just rebuild with that (e.g. by modifying `.gitmodules`)?

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/170a9a3045207e3ed4d396f6c4f03e1c7978b0e4/.gitmodules#L8

TommyMurphyTM1234, 2023-04-03:
I just tried this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
mkdir build
cd build
../configure  --prefix=`pwd`/../installed-tools --enable-multilib  --with-multilib-generator="rv32emc-ilp32e--;rv32emac-ilp32e--;rv32imc-ilp32--;rv32imac-ilp32--zfh*zfhmin*zve32x;rv32imafc-ilp32f--zfh*zfhmin*zve32f;rv32imafdc-ilp32d--zfh*zfhmin*zve32f;rv32imac_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt-ilp32--zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt-ilp32f--zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd-ilp32d--zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imac_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imafc_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafdc_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imaczba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv64imac-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafc-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafdc-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imac_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafc_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafdc_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64ima_zca_zcb_zcmp_zcmt-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafd_zca_zcb_zcf_zcd-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v" --with-arch=rv64ima --with-abi=lp64 --with-cmodel=medany
```
I then edited `.gitmodules` to change
```
[submodule "gcc"]
	path = gcc
	url = https://gcc.gnu.org/git/gcc.git
	branch = releases/gcc-12
```
to
```
[submodule "glibc"]
	path = glibc
	url = https://sourceware.org/git/glibc.git
	branch = master
```
Then
```
make 2>&1 | tee build.log
```
but it failed.
```
checking for _LK_LOCK... no
checking if mkdir takes one argument... no
/usr/bin/env: 'python': No such file or directory
invalid option for --with-multilib-generator
make[1]: *** [Makefile:4549: configure-gcc] Error 1
make[1]: Leaving directory '/home/tommy/Downloads/riscv-gnu-toolchain/build/build-gcc-newlib-stage1'
make: *** [Makefile:571: stamps/build-gcc-newlib-stage1] Error 2
```

`build.log`: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11142548/build.log.tar.gz)

fanghuaqi, 2023-04-04:
> 

You need to install python3, and change like this `scripts/march-to-cpu-opt`

~~~diff
diff --git a/scripts/march-to-cpu-opt b/scripts/march-to-cpu-opt
index f85bd26..7e1064a 100755
--- a/scripts/march-to-cpu-opt
+++ b/scripts/march-to-cpu-opt
@@ -1,4 +1,4 @@
-#!/usr/bin/python3
+#!/usr/bin/env python3
~~~

cmuellner, 2023-04-04:
> > 
> 
> You need to install python3, and change `scripts/march-to-cpu-opt` like this
> 
> ```diff
> diff --git a/scripts/march-to-cpu-opt b/scripts/march-to-cpu-opt
> index f85bd26..7e1064a 100755
> --- a/scripts/march-to-cpu-opt
> +++ b/scripts/march-to-cpu-opt
> @@ -1,4 +1,4 @@
> -#!/usr/bin/python3
> +#!/usr/bin/env python3
> ```

This would be a nice PR!

TommyMurphyTM1234, 2023-04-07:
Can anybody tell me what I'm doing wrong:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
mkdir build
cd build
../configure  --prefix=`pwd`/../installed-tools --enable-multilib  --with-multilib-generator="rv32emc-ilp32e--;rv32emac-ilp32e--;rv32imc-ilp32--;rv32imac-ilp32--zfh*zfhmin*zve32x;rv32imafc-ilp32f--zfh*zfhmin*zve32f;rv32imafdc-ilp32d--zfh*zfhmin*zve32f;rv32imac_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt-ilp32--zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt-ilp32f--zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd-ilp32d--zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imac_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imafc_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafdc_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imaczba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv64imac-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafc-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafdc-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imac_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafc_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafdc_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64ima_zca_zcb_zcmp_zcmt-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafd_zca_zcb_zcf_zcd-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v" --with-arch=rv64ima --with-abi=lp64 --with-cmodel=medany
```
Change `.gitmodules` to use upstream GCC master:
```
[submodule "gcc"]
	path = gcc
	url = https://gcc.gnu.org/git/gcc.git
	branch = master
```
Edit `scripts/march-to-cpu-opt` as outlined above:
```
#!/usr/bin/env python3
...
```
But the build fails:
```
...
a_zbb_zbc_zbs_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b_xxldspn1/mabi.lp64d" \
    "yes" \
    > tmp-mlib.h; \
else \
  /bin/bash /home/tommy/Downloads/issue-1226/gcc/gcc/genmultilib '' '' '' '' '' '' '' '' \
    "" '' no \
    > tmp-mlib.h; \
fi
make[2]: execvp: /bin/bash: Argument list too long
make[2]: *** [Makefile:2223: s-mlib] Error 127
make[2]: Leaving directory '/home/tommy/Downloads/issue-1226/build/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4607: all-gcc] Error 2
make[1]: Leaving directory '/home/tommy/Downloads/issue-1226/build/build-gcc-newlib-stage1'
make: *** [Makefile:571: stamps/build-gcc-newlib-stage1] Error 2
```
Full build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11180311/build.log.tar.gz)

fanghuaqi, 2023-04-10:
> 

Hi @TommyMurphyTM1234 , you are doing correct, and you repeat the error message I met and mentioned in this issue description, this is what I want to get help for riscv community.

Thanks

TommyMurphyTM1234, 2023-04-10:
> > 
> 
> Hi @TommyMurphyTM1234 , you are doing correct, and you repeat the error message I met and mentioned in this issue description, this is what I want to get help for riscv community.
> 
> Thanks

Sorry - what I mean is why is the problem still exant when (I believe) I am using the fixes mentioned here?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1226#issuecomment-1494042755
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1226#issuecomment-1495222585

TommyMurphyTM1234, 2023-04-10:
I was editing `.gitmodules` as explained here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1226#issuecomment-1494992029

expecting the build to then use the `master` branch instead of the `releases/gcc-12` branch.
But it seems to me that this does not happen. Should it?

Or do I have to use `--with-gcc-src` to do a build with the upstream GCC `master` branch?

* https://github.com/riscv-collab/riscv-gnu-toolchain#update-source-tree

TommyMurphyTM1234, 2023-04-10:
> OK - I'm trying this instead...
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1226
> cd issue-1226
> git clone https://github.com/gcc-mirror/gcc gcc-master
> mkdir build
> cd build
> ../configure  --prefix=`pwd`/installed-tools --with-multilib-generator="rv32emc-ilp32e--;rv32emac-ilp32e--;rv32imc-ilp32--;rv32imac-ilp32--zfh*zfhmin*zve32x;rv32imafc-ilp32f--zfh*zfhmin*zve32f;rv32imafdc-ilp32d--zfh*zfhmin*zve32f;rv32imac_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt-ilp32--zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt-ilp32f--zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd-ilp32d--zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-ilp32f--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-ilp32d--zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imac_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imafc_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafdc_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imaczba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imafc_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafdc_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32ima_zca_zcb_zcmp_zcmt_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32imafd_zca_zcb_zcf_zcd_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zve32f;rv32ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32x;rv32imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs_xxldsp-ilp32f--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv32imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs_xxldsp-ilp32d--xxldspn1*xxldspn2*xxldspn3*zfh*zfhmin*zbkb_zbkc_zbkx_zk*zve32f;rv64imac-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafc-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafdc-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imac_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafc_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafdc_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64ima_zca_zcb_zcmp_zcmt-lp64--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt-lp64f--xxldsp*xxldspn1*zfhmin*zfh*v;rv64imafd_zca_zcb_zcf_zcd-lp64d--xxldsp*xxldspn1*zfhmin*zfh*v;rv64ima_zca_zcb_zcmp_zcmt_zba_zbb_zbc_zbs-lp64--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imaf_zca_zcb_zcf_zcmp_zcmt_zba_zbb_zbc_zbs-lp64f--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v;rv64imafd_zca_zcb_zcf_zcd_zba_zbb_zbc_zbs-lp64d--xxldsp*xxldspn1*zfhmin*zfh*zbkb_zbkc_zbkx_zk*v" --with-arch=rv64ima --with-abi=lp64 --with-cmodel=medany --with-gcc-src=`pwd`/../gcc-master
> ```
> 
> Edit
> 
> https://github.com/riscv-collab/riscv-gnu-toolchain/blob/170a9a3045207e3ed4d396f6c4f03e1c7978b0e4/scripts/march-to-cpu-opt#L1
> 
> to read `#!/usr/bin/env python3` then build:
> ```
> make 2>&1 | tee build.log
> ```
> 
> I'll post back when I have an update...

Ok - same failure as before even though I am definitely using the GCC source base with the patches mentioned earlier.
```
...
64x_zvl128b_zvl32b_zvl64b_xxldsp_xxldspn1/mabi.lp64d march.rv64imafd_zca_zcb_zcd_zcf_zba_zbb_zbc_zbs/mabi.lp64d=march.rv64imafdv_zfhmin_zca_zcb_zcd_zcf_zba_zbb_zbc_zbs_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b_xxldspn1/mabi.lp64d" \
    "yes" \
    > tmp-mlib.h; \
else \
  /bin/bash /home/tommy/Downloads/issue-1226/gcc-master/gcc/genmultilib '' '' '' '' '' '' '' '' \
    "" '' no \
    > tmp-mlib.h; \
fi
make[2]: execvp: /bin/bash: Argument list too long
make[2]: *** [Makefile:2254: s-mlib] Error 127
make[2]: Leaving directory '/home/tommy/Downloads/issue-1226/build/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4615: all-gcc] Error 2
make[1]: Leaving directory '/home/tommy/Downloads/issue-1226/build/build-gcc-newlib-stage1'
make: *** [Makefile:571: stamps/build-gcc-newlib-stage1] Error 2
```

fanghuaqi, 2023-04-11:
Hi @TommyMurphyTM1234 ，according to the replies here https://gcc.gnu.org/bugzilla/show_bug.cgi?id=109461#c2 by @kito-cheng , I think I need to make simple multilib configurations, since the two commits introduced by him simply the multilib choice with easier multilib configuration rules.

[image]

And kito will write some doc in future to cover how to use the new multilib reuse scheme.

Thanks

TommyMurphyTM1234, 2023-04-11:
> Hi @TommyMurphyTM1234 ，according to the replies here https://gcc.gnu.org/bugzilla/show_bug.cgi?id=109461#c2 by @kito-cheng , I think I need to make simple multilib configurations, since the two [...]

But the actual content of the `--with -multilib-generator` string is irrelevant to the failure which is because the argument is too long and not because of what the string actually contains. Even if the contents of the string is changed it's still going to fail. So I'm still confused as to what the intended fix is here and how to use it?

TommyMurphyTM1234, 2023-04-12:
> Sorry I guess I didn't explain the patch I posted clearly, after those two patch the `--with-multilib-generator` can be simplified a lots, the argument of `--with-multilib-generator` is a list of `<arch>-<abi>-<reuse-arch>-<reuse-subarch>`, and we can just simplified that as `<arch>-<abi>--` since reuse rule will automatically compute now.
> 
> So for example, `rv32imafc-ilp32f--zfh*zfhmin*zve32f` can be simplified to `rv32imafc-ilp32f--` now, and the this simplification can also applied on other item in the list - then the argument length will be significantly reduced :)

Thanks @kito-cheng - that make a bit more sense now although I haven't actually tried out the changes yet.
But, as I mentioned, this is not a fix for the substantive issue but rather a mitigation that maybe helps to avoid it.
And, as I said, with the preponderance, and ever increasing number, of extensions that issue will inevitably be hit again.
BTW - as I said before, I personally don't really like the multilib "reuse" idea as it can simply cause confusion and potentially arcane errors.

kito-cheng, 2023-04-13:
> But maybe, if the issue is an OS (?) length limitation then there's no pragmatic "fix"?

One possible improvement is make those config pass by file, rather than expand on command line I think.

> BTW - as I said before, I personally don't really like the multilib "reuse" idea as it can simply cause confusion and potentially arcane errors.

I hate multi-lib and reuse rules too, we already deliver almost ~50 different set of multilib, we might deliver more than 100+ different set of multilib if there is no reuse rule...

TommyMurphyTM1234, 2024-04-10:
Just revisiting this...
>```
> make[2]: execvp: /bin/sh: Argument list too long
> make[2]: *** [Makefile:2252: s-mlib] Error 127
>```
>I searched and found it is caused by the command line is too long exceed linux ARGMAX limit ...

I think that `ARGMAX` should actually be `ARG_MAX` which is the command line limit?
But I don't think that `ARG_MAX` is the issue here even though the error message might suggest that.
I think that it's the fact that some of the indivudal arguments to the `genmultilib` script exceed `MAX_ARG_STRLEN`.

Explanation...

From the build log it seems that the [`genmultilib` script](https://github.com/gcc-mirror/gcc/blob/19b98410eb43be60f7e2673afeaabd016321c625/gcc/genmultilib#L131-L142) is being called with the following parameters:

1. `options` - length = 158,794 bytes
2. `dirnames` - length = 148,085
3. `matches` - length = 0
4. `exceptions` - length = 0
5. `extra` - length = 0
6. `exclusions` - length = 0
7. `osdirnames` - length = 0
8. `multilib_required` - length = 1.895
9. `multiarch` - length = 0
10. `multilib_reuse` - length = 291,107
11. `enable_multilib`- length = 3

And the full command line passed to `bash` is a little over 600,000 characters long.

On Ubuntu (and maybe most or all modern Linux distros) `ARG_MAX` is the max stack size (`ulimit -s`) divided by 4 and dictates the maximum length of all command line parameters (including the command itself I think).

By default I see:
```
# ulimit reports the max stack size in KiB
ulimit -s
8192
# 8192 * 1024 = 8388608 bytes
# 8388608 / 4 = 2097152 

# getconf reports ARG_MAX in bytes
getconf ARG_MAX
2097152
```
So, our long command line of ~600,000 characters is still a lot less than `ARG_MAX` = 2,097,152.

However, individual arguments are also limited by `MAX_ARG_STRLEN` and for Ubuntu for example (may also apply to most or all other modern distros):
```
# MAX_ARG_STRLEN = PAGE_SIZE * 32

getconf PAGE_SIZE
4096

# So MAX_ARG_STRLEN = 131072
```
And, from the list above we can see that two individual arguments exceed this limit of 131,072 bytes which is what I think triggers the error.

Unlike `ARG_MAX` which can be dynamically increased by increasing the max stack size using `ulimit -s <new-size>`, I don't think that `MAX_ARG_STRLEN` can be increased dynamically and probably only by recompiling the kernel.

As such, there isn't really a workaround for the original issue unfortunately.

On the other hand:

1. This isn't really a `riscv-gnu-toolchain` issue per se but rather a limit(ation) of most or all Linux distros
2. The more recent multilib reuse logic in the toolchain (mentioned above) mitigates this issue and often obviates the need for such excessively long multilib patterns

Given all of this I suggest that we close this issue.

fanghuaqi, 2024-04-11:
Okay, I think current this issue is resolved, close it now

closed 2024-04-11 by fanghuaqi