riscv-collab/riscv-gnu-toolchain #1227: how to replace current glibc with an old one

JunnanLi, 2023-04-14:
I met a problem ("/lib/libc.so.6: version `GLIBC_2.27' not found") when running ariane core (cva6) on FPGA. And I checked the glibc version of ariane's linux, which is glibc-2.26.  My question is how to replace current glibc with an old one?
Thanks

kito-cheng, 2023-04-14:
Two possible way:

1) go to glibc folder and `git checkout glibc-2.26`
2) Download glibc 2.26 and use `--with-glibc-src` in configure script to specify that.

TommyMurphyTM1234, 2023-04-14:
> Thanks. Do I also need to replace the gcc with an old one

Only if you need an older GCC.
Closing this as it's not an issue with the `riscv-gnu-toolchain`.
Please use other fora such as the mailing lists for general support questions.
This area is for reporting issues/bugs.

closed 2023-04-14 by TommyMurphyTM1234