riscv-collab/riscv-gnu-toolchain #1230: Is there some method to build successful when opcode bits 1..0 is not 3 ?

dshm, 2023-04-18:
I am now doing some vliw extension with riscv and using bit 1 in opcode to indicate whether it is the end of packet. e.g. the opcode addvliwpacketend     rd rs1 rs2 31..25=0  14..12=0 6..2=0x0C 1..0=1   is the add instruction with packet end info.   However I generate the encoding.out.h using riscv-opcodes and add MASK and MATCH to binutils. but when I make the toolchain it gives 
Error: internal: bad RISC-V opcode (bits 0xffffffffffff0000 undefined): addvliwnotdelay d,s,t
Fatal error: internal: broken assembler.  No assembly attempted
Makefile:481: recipe for target 'lib_a-sprintf_chk.o' failed
make[9]: *** [lib_a-sprintf_chk.o] Error 1

I google and find https://github.com/riscv-software-src/riscv-tools/issues/233
it says “I think this is happening because 0x74 is not a valid opcode for a 32-bit-long RISC-V instruction. If instruction bits 1..0 are not equal to 3, then it's a 16-bit-long instruction. So you should pick another opcode for which bits 1..0 equal 3. (Or, if you intended to make a 16-bit-long instruction, change the mask to 0xffff.)”

It‘s there way to solve this？ Or for 32bit riscv instruction the bits 1..0 must equal 3 and these two bits can't be used for vliw extension?

aswaterman, 2023-04-18:
The standard length encoding reserves bits 1..0 != 3 for 16-bit long instructions; see https://github.com/riscv/riscv-isa-manual/blob/1f038182810727f5feca311072e630d6baac51da/src/intro.tex#L478-L482

Since your approach conflicts with the standard, you'd probably have to make a variety of hacks to the toolchain and other components to make this work.  It may not be straightforward to accomplish.

dshm, 2023-04-19:
Is there some work to handle this successful as far as you know? thus do some vliw extension with riscv. We want to use bits 1..0 for vliw extension as said in riscv-spec-20191213 chapter 26 Extending RISC-V section 26.5 Supporting VLIW encodings, it said as follow:

"End-of-Group bits in Prefix
None of the above approaches retains the RISC-V encoding for the individual operations within
a VLIW instruction. Yet another approach is to repurpose the two prefix bits in the fixed-width
32-bit encoding. One prefix bit can be used to signal “end-of-group” if set, while the second bit
could indicate execution under a predicate if clear. Standard RISC-V 32-bit instructions generated
by tools unaware of the VLIW extension would have both prefix bits set (11) and thus have the
correct semantics, with each instruction at the end of a group and not predicated.
The main disadvantage of this approach is that the base ISA lacks the complex predication support
usually required in an aggressive VLIW system, and it is difficult to add space to specify more
predicate registers in the standard 30-bit encoding space."

If the current software implemetation does not leave much place to handle this extension, how many place in the software may need to change to make it work as far as you can estimate?   @aswaterman

TommyMurphyTM1234, 2023-04-26:
I've seen mention of some people attempting VLIW using RISC-V but nothing tangible and nothing in relation to toolchains such as GCC or LLVM. Might be best to take this issue to the mailing lists as it's not really an issue with the `riscv-gnu-toolchain` repo per se?

closed 2023-04-26 by TommyMurphyTM1234