riscv-collab/riscv-gnu-toolchain #1231: Compilation problems after calling libgcc __bswapdi2 return

xzt777, 2023-04-19:
hello, I am using the newlib bare metal toolchain to run libgcc __bswapdi2 api through qemu，and I found a problem with code compilation after calling __bswapdi2, which led to a result error. 

   Here are my tests, compilations, and results：

````
  volatile int64_t swap64Val;

  for (int64_t i = 1;i < 0x100000000;i = i << 1) {
      swap64Val = __bswapdi2((int64_t)i);
      printf("swap64Val(0x%016llx):0x%016llx\n",i ,swap64Val);
  }
````

The result of __bswapdi2 is stored in a0 and a1, but __bswapdi2 returns only sw a0, and a1 is not processed, instead it is compiled with the strange instruction srai a5,a5,0x1f. I don't know why gcc behaves this way.

````
800011aa <__bswapdi2>:
800011f2:	00f6e533          	or	a0,a3,a5
800011f6:	8dd1                	or	a1,a1,a2
800011f8:	8082                	ret
````

````
8000112a <main>:
...
80001142:	a881                	j	80001192 <main+0x68>
80001144:	fe842503          	lw	a0,-24(s0)
80001148:	fec42583          	lw	a1,-20(s0)
8000114c:	28b9                	jal	800011aa <__bswapdi2>
8000114e:	87aa                	mv	a5,a0
80001150:	893e                	mv	s2,a5
80001152:	87fd                	        srai	a5,a5,0x1f
80001154:	89be                	mv	s3,a5
80001156:	ff242023          	sw	s2,-32(s0)
8000115a:	ff342223          	sw	s3,-28(s0)
8000115e:	fe042703          	lw	a4,-32(s0)
80001162:	fe442783          	lw	a5,-28(s0)
80001166:	fe842603          	lw	a2,-24(s0)
8000116a:	fec42683          	lw	a3,-20(s0)
8000116e:	9f418513          	addi	a0,gp,-1548 # 90000b94 <_global_impure_ptr+0x1c4>
80001172:	7e6000ef          	jal	ra,80001958 <printf>
...
````

The results are all zeros due to compilation problems, which obviously do not meet expectations：

````
swap64Val(0x0000000000000001):0x0000000000000000
swap64Val(0x0000000000000002):0x0000000000000000
swap64Val(0x0000000000000004):0x0000000000000000
swap64Val(0x0000000000000008):0x0000000000000000
swap64Val(0x0000000000000010):0x0000000000000000
swap64Val(0x0000000000000020):0x0000000000000000
swap64Val(0x0000000000000040):0x0000000000000000
swap64Val(0x0000000000000080):0x0000000000000000
swap64Val(0x0000000000000100):0x0000000000000000
swap64Val(0x0000000000000200):0x0000000000000000
swap64Val(0x0000000000000400):0x0000000000000000
swap64Val(0x0000000000000800):0x0000000000000000
swap64Val(0x0000000000001000):0x0000000000000000
swap64Val(0x0000000000002000):0x0000000000000000
swap64Val(0x0000000000004000):0x0000000000000000
swap64Val(0x0000000000008000):0x0000000000000000
swap64Val(0x0000000000010000):0x0000000000000000
swap64Val(0x0000000000020000):0x0000000000000000
swap64Val(0x0000000000040000):0x0000000000000000
swap64Val(0x0000000000080000):0x0000000000000000
swap64Val(0x0000000000100000):0x0000000000000000
swap64Val(0x0000000000200000):0x0000000000000000
swap64Val(0x0000000000400000):0x0000000000000000
swap64Val(0x0000000000800000):0x0000000000000000
swap64Val(0x0000000001000000):0x0000000000000000
swap64Val(0x0000000002000000):0x0000000000000000
swap64Val(0x0000000004000000):0x0000000000000000
swap64Val(0x0000000008000000):0x0000000000000000
swap64Val(0x0000000010000000):0x0000000000000000
swap64Val(0x0000000020000000):0x0000000000000000
swap64Val(0x0000000040000000):0x0000000000000000
swap64Val(0x0000000080000000):0x0000000000000000
````

Thanks
Zitai

xzt777, 2023-04-19:
The compile option used is riscv64-unknown-elf-gcc -O0

kito-cheng, 2023-04-19:
> The compile option used is riscv64-unknown-elf-gcc -O0

Hmmm, you say rv64 here

TommyMurphyTM1234, 2023-04-19:
> > The compile option used is riscv64-unknown-elf-gcc -O0
> 
> Hmmm, you say rv64 here

I presume that they're actually using `riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 ...` and a multilib toolchain but, admittedly, insufficient info has been provided which makes it necessary to go back and forth interrogating it. I don't understand why people cannot provide adequate info and a clear/minimal reproducible test case up front when reporting issues.

xzt777, 2023-04-20:
> Try __builtin_bswap64 instead of __bswapdi2, my best guess is you use that without function declaration and then according c lang spec compiler will implicitly declare it as return a int.
> 
> See more: https://stackoverflow.com/questions/30542092/function-without-return-type-specified-in-c

Yes, adding __bswapdi2 declaration works.

Thanks for your advice.

xzt777, 2023-04-20:
> > Try __builtin_bswap64 instead of __bswapdi2, my best guess is you use that without function declaration and then according c lang spec compiler will implicitly declare it as return a int.
> > See more: https://stackoverflow.com/questions/30542092/function-without-return-type-specified-in-c
> 
> Yes, it's unknown what, if any, function declaration/prototype is being used for `__bswapdi2` in this case. It should be as described here:
> 
> * https://gcc.gnu.org/onlinedocs/gccint/Integer-library-routines.html
> 
> Again, this is why a minimal, self-contained, reproducible test case is useful and eliminates most of this inefficient back and forth.

Thanks for your advice.

closed 2023-04-20 by kito-cheng