riscv-collab/riscv-gnu-toolchain #1236: Cannot build baremetal/elf toolchain with bitmanip extension enabled

helloeagleyang, 2023-04-25:
Hello, here is my configuration:
```shell
./configure --with-abi=ilp32d --with-arch=rv32gc_zbb_zba_zbc_zbs 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow ' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow ' --target=$TARGET LDFLAGS="-static" --with-isa-spec=20191213
```

after 
`make -j8`

Some errors were reported:
```shell
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c: In function 'emdnorm':
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c:1413:1: error: unrecognizable insn:
 1413 | }
      | ^
(insn 277 276 278 48 (set (reg:SI 205)
        (const_int 2048 [0x800])) "/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c":1304:15 -1
     (nil))
during RTL pass: vregs
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c:1413:1: internal compiler error: in extract_insn, at recog.cc:2791
0x619df0 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:108
0x619e12 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:116
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c: In function 'emdnorm':
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c:1413:1: error: unrecognizable insn:
 1413 | }
      | ^
(insn 269 268 270 46 (set (reg:SI 198)
        (const_int 2048 [0x800])) "/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c":1304:15 -1
     (nil))
during RTL pass: vregs
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c:1413:1: internal compiler error: in extract_insn, at recog.cc:2791
0x6192dc extract_insn(rtx_insn*)
	../.././gcc/gcc/recog.cc:2791
0x8c4c4f instantiate_virtual_regs_in_insn
	../.././gcc/gcc/function.cc:1611
0x8c4c4f instantiate_virtual_regs
	../.././gcc/gcc/function.cc:1985
0x8c4c4f execute
	../.././gcc/gcc/function.cc:2034
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
make[6]: *** [Makefile:824: lib_a-ldtoa.o] Error 1
make[6]: *** Waiting for unfinished jobs....
0x619df0 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:108
0x619e12 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:116
0x6192dc extract_insn(rtx_insn*)
	../.././gcc/gcc/recog.cc:2791
0x8c4c4f instantiate_virtual_regs_in_insn
	../.././gcc/gcc/function.cc:1611
0x8c4c4f instantiate_virtual_regs
	../.././gcc/gcc/function.cc:1985
0x8c4c4f execute
	../.././gcc/gcc/function.cc:2034
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
make[6]: *** [Makefile:824: lib_a-ldtoa.o] Error 1
make[6]: *** Waiting for unfinished jobs....
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/fflush.c: In function '__sflush_r':
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/fflush.c:234:1: error: unrecognizable insn:
  234 | }
      | ^
(insn 16 15 17 4 (set (reg:HI 121)
        (const_int 2048 [0x800])) "/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/fflush.c":118:18 -1
     (expr_list:REG_EQUAL (const_int 2048 [0x800])
        (nil)))
during RTL pass: vregs
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/fflush.c:234:1: internal compiler error: in extract_insn, at recog.cc:2791
0x619df0 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:108
0x619e12 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:116
0x6192dc extract_insn(rtx_insn*)
	../.././gcc/gcc/recog.cc:2791
0x8c4c4f instantiate_virtual_regs_in_insn
	../.././gcc/gcc/function.cc:1611
0x8c4c4f instantiate_virtual_regs
	../.././gcc/gcc/function.cc:1985
0x8c4c4f execute
	../.././gcc/gcc/function.cc:2034
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
make[6]: *** [Makefile:940: lib_a-fflush.o] Error 1
make[6]: *** Waiting for unfinished jobs....
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/flags.c: In function '__sflags':
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/flags.c:94:1: error: unrecognizable insn:
   94 | }
      | ^
(insn 51 50 52 11 (set (reg:SI 89)
        (const_int 2048 [0x800])) "/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/flags.c":86:6 -1
     (expr_list:REG_EQUAL (const_int 2048 [0x800])
        (nil)))
during RTL pass: vregs
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/stdio/flags.c:94:1: internal compiler error: in extract_insn, at recog.cc:2791
0x619df0 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:108
0x619e12 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:116
0x6192dc extract_insn(rtx_insn*)
	../.././gcc/gcc/recog.cc:2791
0x8c4c4f instantiate_virtual_regs_in_insn
	../.././gcc/gcc/function.cc:1611
0x8c4c4f instantiate_virtual_regs
	../.././gcc/gcc/function.cc:1985
0x8c4c4f execute
	../.././gcc/gcc/function.cc:2034
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
make[6]: *** [Makefile:976: lib_a-flags.o] Error 1
make[6]: *** Waiting for unfinished jobs....
In file included from /home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/string/strstr.c:95:
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/string/str-two-way.h: In function 'two_way_long_needle':
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/string/str-two-way.h:410:1: error: unrecognizable insn:
  410 | }
      | ^
(insn 253 252 254 56 (set (reg:SI 205)
        (const_int 2048 [0x800])) "/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/string/str-two-way.h":326:14 -1
     (expr_list:REG_EQUAL (const_int 2048 [0x800])
        (nil)))
during RTL pass: vregs
/home/builder/repos/riscv-gnu-toolchain/newlib/newlib/libc/string/str-two-way.h:410:1: internal compiler error: in extract_insn, at recog.cc:2791
0x619df0 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:108
0x619e12 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	../.././gcc/gcc/rtl-error.cc:116
0x6192dc extract_insn(rtx_insn*)
	../.././gcc/gcc/recog.cc:2791
0x8c4c4f instantiate_virtual_regs_in_insn
	../.././gcc/gcc/function.cc:1611
0x8c4c4f instantiate_virtual_regs
	../.././gcc/gcc/function.cc:1985
0x8c4c4f execute
	../.././gcc/gcc/function.cc:2034
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.

```

helloeagleyang, 2023-04-25:
This is a known issue : `https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106532`

TommyMurphyTM1234, 2023-04-25:
> This is a known issue : `https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106532`

The failure is no longer present if the upstream GCC sources from the `master` branch (GCC 13.1 pre-release) are used.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-upstream
./configure --with-abi=ilp32d --with-arch=rv32gc_zbb_zba_zbc_zbs 'CFLAGS_FOR_TARGET=-Os -mcmodel=medlow ' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medlow ' --target=$TARGET LDFLAGS="-static" --with-isa-spec=20191213 --with-gcc-src=`pwd`/gcc-upstream --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log
```
I guess that once GCC 13 releases then the `.gitmodules` file will be updated to switch branches from `releases/gcc-12` to `releases/gcc-13`.

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.gitmodules#L8
```

helloeagleyang, 2023-04-27:
> > This is a known issue : `https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106532`
> 
> The failure is no longer present if the upstream GCC sources from the `master` branch ([GCC 13 pre-release](https://gcc.gnu.org/pipermail/gcc/2023-April/241140.html)) are used.
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> git clone https://github.com/gcc-mirror/gcc gcc-upstream
> ./configure --with-abi=ilp32d --with-arch=rv32gc_zbb_zba_zbc_zbs 'CFLAGS_FOR_TARGET=-Os -mcmodel=medlow ' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medlow ' --target=$TARGET LDFLAGS="-static" --with-isa-spec=20191213 --with-gcc-src=`pwd`/gcc-upstream --prefix=`pwd`/installed-tools
> make 2>&1 | tee build.log
> ```
> 
> I guess that once GCC 13 releases then the `.gitmodules` file will be updated to switch branches from `releases/gcc-12` to `releases/gcc-13`.
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.gitmodules#L8

Yes, I confirmed that this issue has been fixed in the gcc-13

TommyMurphyTM1234, 2023-04-28:
When this change is integrated the issue can be closed.

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239

closed 2023-10-12 by helloeagleyang