riscv-collab/riscv-gnu-toolchain #1240: RV64IMAC can't link soft-float modules with double-float modules

tomhepworth, 2023-04-28:
I'm trying to configure the tool chain for several architectures, none of which support hardware floating point.

My configuration command is:
```
./configure --with-multilib-generator="rv32ima-ilp32--;rv32imac-ilp32--;rv64ima-lp64--;rv64imac-lp64--;rv32ima-ilp32--" --enable-multilib

make 
```

Yet when i try and compile a simple hello world program with:
```
riscv64-unknown-elf-gcc -march=rv64im -mabi=lp64 hello.c -o hello
```
I get the error:

```
/usr/local/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccXkFVVn.o: can't link soft-float modules with double-float modules
/usr/local/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccXkFVVn.o
collect2: error: ld returned 1 exit status
```

Can anyone help me?

TommyMurphyTM1234, 2023-04-28:
You haven't built a toolchain with a multilib for `rv64im/lp64`.
If you only need to target that `arch/abi` then the simplest thing to do would be to build a toolchain specifically for it.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64im --with-abi=lp64
make 2>&1 | tee build.log
```

tomhepworth, 2023-04-28:
Apologies for the typo, the command that is failing is 
```
riscv64-unknown-elf-gcc -march=rv64ima -mabi=lp64 hello.c -o hello
```
but my config command does contain `rv64ima-lp64--`.

tomhepworth, 2023-04-28:
Realised from looking at #1196 that I wasn't rebuilding properly and something weird must have happened with my different configurations

closed 2023-04-28 by tomhepworth