riscv-collab/riscv-gnu-toolchain #1241: BUGS of gcc-riscv64-unknown-elf -O2

Toby-Shi-cloud, 2023-04-28:
I have found some bugs when use gcc-riscv64-unknown-elf with -O2.

I write some inline assemble to putchar.
something link this:
```c
static inline struct sbiret
sbi_ecall(u_int ext, u_int fid, u_int arg0, u_int arg1,
	  u_int arg2, u_int arg3, u_int arg4, u_int arg5) {
	struct sbiret res;
	asm volatile (
		"mv a0, %[a0]\n"
		"mv a1, %[a1]\n"
		"mv a2, %[a2]\n"
		"mv a3, %[a3]\n"
		"mv a4, %[a4]\n"
		"mv a5, %[a5]\n"
		"mv a6, %[fid]\n"
		"mv a7, %[eid]\n"
		"ecall\n"
		"mv %[error], a0\n"
		"mv %[value], a1\n"
		: [error] "=r" (res.error),
		[value] "=r" (res.value)
		: [a0] "r" (arg0),
		[a1] "r" (arg1),
		[a2] "r" (arg2),
		[a3] "r" (arg3),
		[a4] "r" (arg4),
		[a5] "r" (arg5),
		[fid] "r" (fid),
		[eid] "r" (ext)
		: "memory"
	);
	return res;
}

#define LegacyEcall(ext, a0, a1, a2) sbi_ecall((ext), 0, (a0), (a1), (a2), 0, 0, 0)

void printcharc(char ch) {
	panic_on(LegacyEcall(1, ch, 0, 0).error);
}
```

When I use gcc-riscv64-unknown-elf with flags `--std=gnu99 -march=rv32gc -mabi=ilp32 -Wall -fno-pic -ffreestanding -fno-stack-protector -fno-builtin -O2`, it compiled above code to
```
80200b60 <printcharc>:
printcharc():
80200b60:	4701                	li	a4,0
80200b62:	4785                	li	a5,1
80200b64:	852a                	mv	a0,a0
80200b66:	85ba                	mv	a1,a4
80200b68:	863a                	mv	a2,a4
80200b6a:	86ba                	mv	a3,a4
80200b6c:	873a                	mv	a4,a4
80200b6e:	87ba                	mv	a5,a4
80200b70:	883a                	mv	a6,a4
80200b72:	88be                	mv	a7,a5
80200b74:	00000073          	ecall
80200b78:	872a                	mv	a4,a0
80200b7a:	852e                	mv	a0,a1
80200b7c:	e311                	bnez	a4,80200b80 <printcharc+0x20>
80200b7e:	8082                	ret
80200b80:	802016b7          	lui	a3,0x80201
80200b84:	80201637          	lui	a2,0x80201
80200b88:	80201537          	lui	a0,0x80201
80200b8c:	18068693          	addi	a3,a3,384 # 80201180 <end+0xffc01180>
80200b90:	f1860613          	addi	a2,a2,-232 # 80200f18 <end+0xffc00f18>
80200b94:	02f00593          	li	a1,47
80200b98:	1b050513          	addi	a0,a0,432 # 802011b0 <end+0xffc011b0>
80200b9c:	a8ad                	j	80200c16 <_panic>
```

At `80200b6e: mv a5,a4`, a5 is set to a4, which is 0. This made a7, which is excepted to be 1, changed to wrong value at `80200b72: mv a7,a5`.

Note: this bug occurs as long as I turn on -Ox (including O1, O2, O3, ...). It obviously won't occur when I use -O0.

my gcc version:
> the newest version downloaded from apt
```
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/9.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 9.3.0 ()
```

my platform:
> GitHub Codespaces with 2 cores.

TommyMurphyTM1234, 2023-04-29:
> gcc version 9.3.0

This is quite old - `riscv-gnu-toolchain` is currently at GCC 12 and just about to change to 13.
You should try again with a more up to date toolchain - e.g. the latest "nightly" build here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

I tried with the latest build but your example does not seem to be a self-contained reproducible test case. I got this:
```
tommy@kingfisher:~/Downloads/issue-1241/riscv/bin$ ./riscv64-unknown-elf-gcc --std=gnu99 -march=rv32gc -mabi=ilp32 -Wall -fno-pic -ffreestanding -fno-stack-protector -fno-builtin -O2 -c test.c
test.c:2:11: error: unknown type name 'u_int'; did you mean 'int'?
    2 | sbi_ecall(u_int ext, u_int fid, u_int arg0, u_int arg1,
      |           ^~~~~
      |           int
test.c:2:22: error: unknown type name 'u_int'; did you mean 'int'?
    2 | sbi_ecall(u_int ext, u_int fid, u_int arg0, u_int arg1,
      |                      ^~~~~
      |                      int
test.c:2:33: error: unknown type name 'u_int'; did you mean 'int'?
    2 | sbi_ecall(u_int ext, u_int fid, u_int arg0, u_int arg1,
      |                                 ^~~~~
      |                                 int
test.c:2:45: error: unknown type name 'u_int'; did you mean 'int'?
    2 | sbi_ecall(u_int ext, u_int fid, u_int arg0, u_int arg1,
      |                                             ^~~~~
      |                                             int
test.c:3:11: error: unknown type name 'u_int'; did you mean 'int'?
    3 |           u_int arg2, u_int arg3, u_int arg4, u_int arg5) {
      |           ^~~~~
      |           int
test.c:3:23: error: unknown type name 'u_int'; did you mean 'int'?
    3 |           u_int arg2, u_int arg3, u_int arg4, u_int arg5) {
      |                       ^~~~~
      |                       int
test.c:3:35: error: unknown type name 'u_int'; did you mean 'int'?
    3 |           u_int arg2, u_int arg3, u_int arg4, u_int arg5) {
      |                                   ^~~~~
      |                                   int
test.c:3:47: error: unknown type name 'u_int'; did you mean 'int'?
    3 |           u_int arg2, u_int arg3, u_int arg4, u_int arg5) {
      |                                               ^~~~~
      |                                               int
test.c: In function 'printcharc':
test.c:35:9: warning: implicit declaration of function 'panic_on' [-Wimplicit-function-declaration]
   35 |         panic_on(LegacyEcall(1, ch, 0, 0).error);
      |         ^~~~~~~~
test.c:32:38: warning: implicit declaration of function 'sbi_ecall' [-Wimplicit-function-declaration]
   32 | #define LegacyEcall(ext, a0, a1, a2) sbi_ecall((ext), 0, (a0), (a1), (a2), 0, 0, 0)
      |                                      ^~~~~~~~~
test.c:35:18: note: in expansion of macro 'LegacyEcall'
   35 |         panic_on(LegacyEcall(1, ch, 0, 0).error);
      |                  ^~~~~~~~~~~
test.c:35:42: error: request for member 'error' in something not a structure or union
   35 |         panic_on(LegacyEcall(1, ch, 0, 0).error);
      |                                          ^
```

Toby-Shi-cloud, 2023-04-29:
Sorry, I will download the latest version and have a try again.

And my code given above is a part of my operator system, so you get some error when compiling my code directly.

`u_int` is a typedef of `unsigned int`

`struct sbiret` is defined to
```c
struct sbiret {
    u_int error;
    u_int value;
};
```

`panic_on ` is a external function which to do something else (usually when some unexceptional error happened).

Maybe you can try to compile the following code:
```c
typedef unsigned int u_int;

static inline u_int
sbi_ecall(u_int ext, u_int fid, u_int arg0, u_int arg1,
	  u_int arg2, u_int arg3, u_int arg4, u_int arg5) {
        u_int value, error;
	asm volatile (
		"mv a0, %[a0]\n"
		"mv a1, %[a1]\n"
		"mv a2, %[a2]\n"
		"mv a3, %[a3]\n"
		"mv a4, %[a4]\n"
		"mv a5, %[a5]\n"
		"mv a6, %[fid]\n"
		"mv a7, %[eid]\n"
		"ecall\n"
		"mv %[error], a0\n"
		"mv %[value], a1\n"
		: [error] "=r" (error),
		[value] "=r" (value)
		: [a0] "r" (arg0),
		[a1] "r" (arg1),
		[a2] "r" (arg2),
		[a3] "r" (arg3),
		[a4] "r" (arg4),
		[a5] "r" (arg5),
		[fid] "r" (fid),
		[eid] "r" (ext)
		: "memory"
	);
	return value << 16 | error;
}

u_int foo() {
	return sbi_ecall(1, 0, 0, 0, 0, 0, 0, 0);
}
```

I compile it using `riscv64-unknown-elf-gcc test.c --std=gnu99 -march=rv32gc -mabi=ilp32 -c -o test.elf -O2`.
The result of `riscv64-unknown-elf-objdump test.elf -aldS` is as follows:
```

test.elf:     file format elf32-littleriscv
test.elf

Disassembly of section .text:

00000000 <foo>:
foo():
   0:	4501                	li	a0,0
   2:	4785                	li	a5,1
   4:	852a                	mv	a0,a0
   6:	85aa                	mv	a1,a0
   8:	862a                	mv	a2,a0
   a:	86aa                	mv	a3,a0
   c:	872a                	mv	a4,a0
   e:	87aa                	mv	a5,a0
  10:	882a                	mv	a6,a0
  12:	88be                	mv	a7,a5
  14:	00000073          	ecall
  18:	852a                	mv	a0,a0
  1a:	87ae                	mv	a5,a1
  1c:	07c2                	slli	a5,a5,0x10
  1e:	8d5d                	or	a0,a0,a5
  20:	8082                	ret
```

Toby-Shi-cloud, 2023-04-29:
I now find the reason of above problem.

When I use `asm` in C, I should point out which register(s) or memory I am going to change, but unfortunately I forgot this rule.

If I change `"memory"` to `"a0", "a1", "a2", "a3", "a4", "a5", "a6", "a7"`, the compiler will use `t1` and `t3` (or other registers rather than `a0` and `a5`) to store immediate `0` and `1`, which will avoid this bug.

Thank you to answer my question. And I am going to close this issue as it is not the compiler's bug but my fault.

closed 2023-04-29 by Toby-Shi-cloud