riscv-collab/riscv-gnu-toolchain #1243: `make report-*` testing has been broken since dejagnu 1.6.2 rollback 2023-04-29 9a26689

patrick-rivos, 2023-05-03:
[dejagnu: Rollback version to 1.6.2 `9a26689`](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/9a26689a54dd43e0fc61ede063a0d6263d5d3f41) has broken `make report-gcc`.

I have also run into issues with `make report-binutils`. I expect that is the same issue.

```
...
		=== gcc tests ===

Schedule of variations:
    riscv-sim/-march=rv64imafdc/-mabi=lp64d/-mcmodel=medlow

Running target riscv-sim/-march=rv64imafdc/-mabi=lp64d/-mcmodel=medlow
ERROR: couldn't load description file for riscv-sim
...
riscv-gnu-toolchain/build/../scripts/testsuite-filter gcc newlib riscv-gnu-toolchain/build/../test/allowlist `find build-gcc-newlib-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
make[1]: Leaving directory 'riscv-gnu-toolchain/build'
```

Applying this patch to 1.6.2 fixes the issue: https://git.savannah.gnu.org/cgit/dejagnu.git/commit/?id=8c367b5ca4a79e8ab732ef7f237ab0153ce196bb

Relevant: #1201

kito-cheng, 2023-05-03:
Oh.......thanks for report that, I guess I didn't hit that because I already has that file.

patrick-rivos, 2023-05-03:
That sounds correct - I had some issues tracking down the root cause since the a dejagnu rebuild only appears to be re-triggered under certain conditions. Clean checkout/builds were required.

If you build with trunk, you then have to checkout [`26afe70`](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/8dd29548de6e9a34b8d0ee8e44c4f32a32c90f55) (right before [`8dd2954`](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/8dd29548de6e9a34b8d0ee8e44c4f32a32c90f55)) on a subsequent build in order to get dejagnu to actually work.

If you checkout/build clean, you can just use any commit before [`9a26689`](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/9a26689a54dd43e0fc61ede063a0d6263d5d3f41)

PR #1247 (2023-05-13): dejagnu: Bump dejagnu to include multilib fix
This changes Dejagnu to the most recent version that includes @cmuellner's fixes.

The Dejagnu commit with @cmuellner's fixes has not been tagged for release.

If we only want to use released versions of Dejagnu, I propose switching back to Dejagnu 1.6.3 to resolve #1243 since the rollback did not resolve #1201.

Resolves: #1201 and #1243.
closed 2023-05-13 by kito-cheng