riscv-collab/riscv-gnu-toolchain #1244: Floating Point Error

amithmath, 2023-05-03:
Hello, 

I am working on RSICV-32 based accelerator, when I  compare floating point (FP) addition result from RISC-V 32 with host as shown in the figure below 

[image]

where C[0], C[1], C[2] FP addition result from RISCV-32 and Expected is FP addition result from the host machine (x86), I am getting strange FP result C[0], C[1], C[1] from RISC-V 32, I am wondering, what is going wrong? It seems to me FP register file in RISC-V  getting wrong floating point numbers, but  I don't why. I thought that it could be issue of compiler optimization flags, I changed flags O0- O3 but of no help.

Can you please help me with is issue?

Thanks,
-Amit

TommyMurphyTM1234, 2023-05-03:
You need to provide a self-contained reproducible test case if you want assistance.

amithmath, 2023-05-03:
I am not using any special test cases, I am using following FP numbers for debugging purpose. 

Let's say C[i] = A[i] + B[i], where B[i]= 45.5505033, A[0]= 0.9823221699, A[1]=0.599975063, A[2]=0.6037020222, C[0] = 46.5328254699, C[1]= 46.1504783630, C[2] = 46.1542053222, where i=0, 1, and 2

TommyMurphyTM1234, 2023-05-03:
> I am not using any special test cases, I am using following FP numbers for debugging purpose.
> 
> Let's say C[i] = A[i] + B[i], where B[i]= 45.5505033, A[0]= 0.9823221699, A[1]=0.599975063, A[2]=0.6037020222, C[0] = 46.5328254699, C[1]= 46.1504783630, C[2] = 46.1542053222, where i=0, 1, and 2

You still need to provide a self-contained reproducible test case.
Otherwise you are expecting others to try to reinvent your code.

amithmath, 2023-05-03:
I have attached source code, in the following code: line #82-89, where B[i]= 45.5505033, in N=2, I am generating random numbers for A[i].  I am passing these inputs as arguments to kernel_float_vec_add at line #140 and #147. At line #191-208 (also line #8-13), I am comparing results from host and accelerator.

[test_float_vec_add.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11381586/test_float_vec_add.txt)

Following is the source code of kernel_float_vec_add.c, At line #17, I am computing FP addition in RISCV-32

[kernel_float_vec_add.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11381591/kernel_float_vec_add.txt)

TommyMurphyTM1234, 2023-05-03:
> I have attached source code, in the following code: line #82-89, where B[i]= 45.5505033, in N=2, I am generating random numbers for A[i]. I am passing these inputs as arguments to [...]

This is still not a reproducible test case as we don't know how exactly you're compiling the code.
You need to provide source code, instructions on how you're compiling your code (or, better still, a `Makefile`), details of the toolchain that you're using (version, how you built it or where you got it) etc.

amithmath, 2023-05-03:
Sorry, I have only attached snapshot of the code. Complete set-up is in our server, I am running it on FPGA board.  In order to understand exactly how we are compiling code, instructions, etc. I request to have conference call, and then you can replicate in your server? Because the set-up is complicated and lot of dependencies.  May I PM you email address: tommy_murphy@hotmail.com?

TommyMurphyTM1234, 2023-05-04:
> Sorry, I have only attached snapshot of the code. Complete set-up is in our server, I am running it on FPGA board. In order to understand exactly how we are compiling code, instructions, etc. I request to have conference call, and then you can replicate in your server? Because the set-up is complicated and lot of dependencies. May I PM you email address: [tommy_murphy@hotmail.com](mailto:tommy_murphy@hotmail.com)?

Please keep all discussion here.
It's up to you to create a self-contained reproducible test case that illustrates the issue that you are claiming. 
If you cannot or will not do that then the issue will have to be closed.

amithmath, 2023-05-05:
In that case, I am using this repo:  https://github.com/hwpeng/bsg_f1 and https://github.com/bespoke-silicon-group/bsg_manycore/tree/bc7ef08b343815ba0f0b10fe3327e5243e43d42c for riscv-32 based accelerator

In this source code, it is comparing floating result from x86 machine with riscv-32: https://github.com/hwpeng/bsg_f1/blob/master/regression/cuda/test_float_vec_add.c

riscv-32 FP add: https://github.com/bespoke-silicon-group/bsg_manycore/blob/bc7ef08b343815ba0f0b10fe3327e5243e43d42c/software/spmd/bsg_cuda_lite_runtime/float_vec_add/kernel_float_vec_add.cpp

Make recipes are at: https://github.com/bespoke-silicon-group/bsg_manycore/blob/bc7ef08b343815ba0f0b10fe3327e5243e43d42c/software/mk/Makefile.builddefs

TommyMurphyTM1234, 2023-05-05:
This is still not a self-contained reproducible test case.
You're asking others to do a lot of work just to reproduce your claimed issue.
Few, if any, people are going to have the bandwidth to do that for you.
You need to pare the issue down to a smaller, ideally a minimal, self-contained reproducible test case if you want others to look at it.

closed 2023-05-08 by amithmath