riscv-collab/riscv-gnu-toolchain #1250: GMP is missing or unusable

kgonidis, 2023-05-11:
machine: M1 mac

Ran brew install for dependencies

./configure --prefix=/opt/riscv

make

After downloads and checking I receive the following error.
checking for libexpat... yes
checking how to link with libexpat... -lexpat
checking for XML_StopParser... yes
checking for libgmp... no
configure: error: GMP is missing or unusable
make[2]: *** [configure-gdb] Error 1
make[1]: *** [all] Error 2
make: *** [stamps/build-gdb-newlib] Error 2

I verified that gmp is installed:

brew info gmp
==> gmp: stable 6.2.1 (bottled), HEAD
GNU multiple precision arithmetic library
https://gmplib.org/
/opt/homebrew/Cellar/gmp/6.2.1_1 (21 files, 3.2MB) *
  Poured from bottle on 2022-08-01 at 00:15:38
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/gmp.rb
License: LGPL-3.0-or-later or GPL-2.0-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 4,004 (30 days), 86,704 (90 days), 1,986,258 (365 days)
install-on-request: 123 (30 days), 3,975 (90 days), 56,102 (365 days)
build-error: 6 (30 days)

tried symlinking lib/include into /usr/{bin,include}

Not sure how to proceed. Any help would be appreciated.

Thanks

XYenChi, 2023-05-11:
Could you check your gcc version with:
```
gcc -version
```
The "gcc" installed by Xcode is different with "gcc" installed by brew. 
Maybe you could resolve this issue with:
```
export  LIBRARY_PATH=/opt/homebrew/lib
```

TommyMurphyTM1234, 2023-05-11:
> I verified that gmp is installed:

What about libgmp-dev?

TommyMurphyTM1234, 2023-05-13:
Any update @kgonidis?

TommyMurphyTM1234, 2023-05-17:
Closing due to lack of any update.
Reopen with further info if necessary.

EdvardSire, 2023-09-13:
I'm having the same problem as TommyMurphy, and I would like to troubleshoot it.

Without really understanding what's going on I have tried:
`export CC=/opt/homebrew/Cellar/gcc/13.2.0/bin/gcc-13`
`export LDFLAGS=-L/opt/homebrew/Cellar/gmp/6.2.1_1/include`
`./configure --prefix=/opt/riscv`
`make`
is producing
```
checking for libgmp... no
configure: error: GMP is missing or unusable
make[2]: *** [configure-gdb] Error 1
make[1]: *** [all] Error 2
make: *** [stamps/build-gdb-newlib] Error 2
```

there are some other warnings aswell, but I'm guessing the linking of gmp is the problem?:)

TommyMurphyTM1234, 2023-09-13:
> I'm having the same problem as TommyMurphy, and I would like to troubleshoot it.

I wasn't having the problem - that was @kgonidis.
I don't have Mac hardware to investigate here.

You should probably at least capture, compress and attach a full build log for a (non-parallel) build so that it could be compared with a successful build log on non-Mac hardware.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1250
cd issue-1250
./configure --prefix=`pwd`/installed-tools ...
make 2>&1 | tee build.log
```

EdvardSire, 2023-09-13:
I ran your commands and added
`.NOTPARALLEL:`
to the generated Makefile

[build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12598766/build.log.zip)

TommyMurphyTM1234, 2023-09-13:
> I ran your commands and added `.NOTPARALLEL:` to the generated Makefile

I don't understand why you did this.
Why not just run a non-parallel build by not passing any `-j` option to the `make` invocation?

TommyMurphyTM1234, 2023-09-13:
You could maybe look at how @ilg-ul builds the xPack RISC-V toolchain on/for Mac in case that helps at all:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
* https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/README-MAINTAINER.md#apple-silicon-macos

Maybe you could even use the xPack binary builds as-is?

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

ilg-ul, 2023-09-13:
Exactly to avoid such issues, I build all dependencies from sources.

Usually, if the libraries are not in system locations, you have to tell to the configure step the custom folders to search for.

EdvardSire, 2023-09-13:
> ```
> brew install gmp
> brew install gmp-dev
> brew install libgmp-dev
> ```

Only `gmp` is a brew package.

> Maybe you could even use the xPack binary builds as-is?
> https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

This solved my problem. Is there anything else i can help with?

TommyMurphyTM1234, 2023-09-13:
> This solved my problem.

OK - I'll close this issue again so.

closed 2023-09-13 by TommyMurphyTM1234