riscv-collab/riscv-gnu-toolchain #1251: README.md prerequisites section is incomplete

TommyMurphyTM1234, 2023-05-18:
See here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1238#issuecomment-1552866334

Omitted from the prerequistes seem to be:

* `git`
* `python` (Python 2)

TommyMurphyTM1234, 2023-05-26:
If using QEMU for the test suite:

* libglib2.0-dev

TommyMurphyTM1234, 2023-08-19:
Not sure about this but another potential dependency?
If GDB Python support is needed then one of the following pairs?

* `python` and `python-dev`
* `python2` and `python2-dev`
* `python3` and `python3-dev`

See here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/925
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/772

TommyMurphyTM1234, 2024-11-02:
A useful reference in this context is this script which installs the prerequisites for the GitHub CI Actions on Ubuntu:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/setup-apt.sh

However, it may not encompass all prerequisites mentioned above, e.g. for running the test suite etc.?

jjchico, 2025-12-23:
Hi! libncurses-dev is also missing from the pre-requisites. Without it, the GDB build will disable the TUI mode that is currently enabled by default.

cmuellner, 2025-12-23:
> Hi! libncurses-dev is also missing from the pre-requisites. Without it, the GDB build will disable the TUI mode that is currently enabled by default.

Thanks for reporting!
We also don't have it in the [setup-apt.sh](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/setup-apt.sh) that is used for the CI/CD builds.
It would be great if you could prepare a PR.

jjchico, 2025-12-25:
> Thanks for reporting! We also don't have it in the [setup-apt.sh](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/setup-apt.sh) that is used for the CI/CD builds. It would be [...]

Sure, here you are: #1817