riscv-collab/riscv-gnu-toolchain #1252: New `make report-newlib` errors since #1233

patrick-rivos, 2023-05-18:
#1233 introduced errors in `make report-newlib`.

```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |19208 /  3644 | 9845 /  2462 |      - |
```

<details>
    <summary>Subset of unexpected fails list</summary>
    
    ```
    		    === g++: Unexpected fails for rv64imafdc lp64d medlow ===
    [Removed some to avoid github's comment size limit]...
    FAIL: g++.old-deja/g++.pt/explicit42.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit42.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit42.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit43.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit43.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit43.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit43.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit50.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit50.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit50.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit50.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit51.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit51.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit51.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit51.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit52.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit52.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit52.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit52.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit53.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit53.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit53.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit53.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit54.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit54.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit54.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit54.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit55.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit55.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit55.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit55.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit56.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit56.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit56.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit56.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit57.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit57.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit57.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit57.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit58.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit58.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit58.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit58.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit59.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit59.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit59.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit59.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit60.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit60.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit60.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit60.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.pt/explicit61.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.pt/explicit61.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.pt/explicit61.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.pt/explicit61.C  -std=c++98 execution test
[... 615 lines cut ...]
    FAIL: g++.old-deja/g++.robertl/eh990323-3.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-4.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-4.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-4.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-4.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-5.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-5.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-5.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.robertl/eh990323-5.C  -std=c++98 execution test
    FAIL: g++.old-deja/g++.robertl/ice990323-2.C  -std=c++14 execution test
    FAIL: g++.old-deja/g++.robertl/ice990323-2.C  -std=c++17 execution test
    FAIL: g++.old-deja/g++.robertl/ice990323-2.C  -std=c++20 execution test
    FAIL: g++.old-deja/g++.robertl/ice990323-2.C  -std=c++98 execution test
    FAIL: g++.target/riscv/call-with-empty-struct-float.C execution test
    FAIL: g++.target/riscv/call-with-empty-struct-int.C execution test
    
                   ========= Summary of gcc testsuite =========
                                | # of unexpected case / # of unique unexpected case
                                |          gcc |          g++ |     gfortran |
     rv64imafdc/  lp64d/ medlow |19208 /  3644 | 9845 /  2462 |      - |
    make: *** [Makefile:980: report-gcc-newlib] Error 1
    ```
</details>

CC: @kito-cheng @cmuellner @vineetgarc

patrick-rivos, 2023-05-18:
`make report-linux` results:
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |19620 /  3743 | 9929 /  2481 |17972 /  3029 |
make: *** [Makefile:987: report-gcc-linux] Error 1
```

kito-cheng, 2023-05-19:
Could you provide some error log in gcc.log and see if https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1254 help?

patrick-rivos, 2023-05-19:
#1254 fixes the errors for both linux and newlib!
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |
```

commit 9134376a (2023-05-19): Install pyelftools before testing

We use some non-standard python package in #1233, so we must make sure
it available before run the test.

Fix #1252
PR #1254 (2023-05-23): Install pyelftools before testing
We use some non-standard python package in #1233, so we must make sure it available before run the test.

Fix #1252
closed 2023-05-23 by kito-cheng