riscv-collab/riscv-gnu-toolchain #1255: [multilib] compiler always defaults to IMAC libc on mismatch of `march` string to precompiled library set

jnk0le, 2023-05-20:
forwarding from xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/17

Whenever `march=` is provided with extension set that doesn't match exactly, one of the precompiled multilib set, it always silently links against IMAC (rv32i2p1_m2p0_a2p1_c2p0 ?)

- When IMAC is a subset of implemented extensions, it's a non issue, just a minor missing-opt.

- On targets like `RV32IC_Zba_Znotinprecompiledset` it silently links giving a false positive of properly compiling until A or M instruction is hit. (happy debugging...)

- On ch32v003 which is rv32ec_zicsr_zifencei it started causing build errors (https://www.eevblog.com/forum/microcontrollers/risc-v-compilation-errors-mismatched-libgcc/). Workaround is to use `-misa-spec=2.2` or increasing the permutation set of precompiled multilib, with extensions useless in libc code.

ie.
```
c:/pro[...]e-elf/lib\libg_nano.a(lib_a-nano-reallocr.o)
c:/pro[...]lf/lib\libg_nano.a(lib_a-readr.o) use 16-byte stack aligned but the output use 4-byte stack aligned
c:/pro[...]e-elf/bin/ld.exe: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order
```

I think it should pick the biggest common denominator instead. for `rv32ec_zicsr_zifencei` it should link against `rv32ec` one.

TommyMurphyTM1234, 2023-05-25:
> forwarding from [xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/17](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/17)
> 
> Whenever `march=` is provided with extension set that doesn't match exactly, one of the precompiled libc from multilib set, it always silently links against IMAC (rv32i2p1_m2p0_a2p1_c2p0 ?)

This is because the multilib matching the requested `arch/abi` does not exist so the toolchain falls back to linking the toolchain's default `arch/abi` libs - and the xPack RISC-V toolchain is configured with `--with-arch=rv32imac --with-abi=ilp32`.

* https://xpack.github.io/blog/2023/02/05/riscv-none-elf-gcc-v12-2-0-3-released/#supported-libraries

This is simply how the toolchain has always worked and is nothing to do with any sort of multilib matching smarts (which do not exist as far as I know) for matching the "closest" subset multilib which is what you seem to be suggesting/requesting?

TommyMurphyTM1234, 2023-05-31:
Closing this issue as there has been no update for a week and the toolchain is simply working as expected in this case.

closed 2023-05-31 by TommyMurphyTM1234