riscv-collab/riscv-gnu-toolchain #1257: Built newq cross-compiler from tip and have issues with csrr opcode?

InspireSemi, 2023-05-25:
Was updating our cross-compiler to 12.x.  Build went ok, but when using the new compiler it threw this error:

Error: unrecognized opcode `csrr a0,mhartid', extension `zicsr' required
I used the same ./configure line for our previous compiler (10.2)
./configure --prefix=/opt/riscv64_12.2 --with-cmodel=medany --with-arch=rv64imafdc --with-abi=lp64d

Do I need to add zicsr to the wth-arch flag?

InspireSemi, 2023-05-25:
I have added zicsr to the --with-arch string.

rebuilding the compiler

TommyMurphyTM1234, 2023-05-25:
Yes - this is down to work on canonicalized arch strings in recent months and what the specs say.
Previously, IIRC, `g` meant `imafd` and inplicitly included `zicsr` (and `zifencei`?).
But no longer. 
So if you want CSR access instructions then you need to explicitly specify `zicsr` (e.g. `rv64gc_zicsr`) in your toolchain (`configure --with-arch=...`) or compilation target (`riscv64-unknown-elf-gcc --march=...`) architecture string.

TommyMurphyTM1234, 2023-05-25:
Actually - I'm confused now...

The ratified Volume 1, Unprivileged Specification version 20191213 spec (https://riscv.org/technical/specifications/) says

> We have also defined an abbreviation “G” to represent the “IMAFDZicsr Zifencei” base and extensions, as this is intended to represent our standard general-purpose ISA.

But the latest draft (https://github.com/riscv/riscv-isa-manual/releases/tag/riscv-isa-release-1239329-2023-05-23) also says:

> we use the abbreviation G for the IMAFDZicsr_Zifencei combination of instruction-set extensions.

>

InspireSemi, 2023-05-25:
_zicsr or can I just use zicsr ? (adding jusat zicsr seems top work btw?)

I also need to add zifencei ?  or would it be _zfencei

InspireSemi, 2023-05-25:
No C in the G abbreviation

TommyMurphyTM1234, 2023-05-25:
> _zicsr or can I just use zicsr ? (adding jusat zicsr seems top work btw?)
> 
> I also need to add zifencei ? or would it be _zfencei

If you want both then you specify, say, `rv64gczicsr_zifencei`.

And additional confusion is the use of upper case 'Z' in the specs whereas, AFAIK, canonical architecture strings used when configuring or invoking the toolchain must (?) use lower case 'z'.

InspireSemi, 2023-05-25:
It would be helpful to add this to the readme on the github page...

InspireSemi, 2023-05-25:
I can take a shot at the readme in a few days and submit a PR...

closed 2023-05-25 by InspireSemi

TommyMurphyTM1234, 2023-05-25:
> No C in the G abbreviation

I don't understand your point.

TommyMurphyTM1234, 2023-05-25:
> I will update the readme with how to compile the current tools with the proper --with-arch options ... with some info like: If you are rv32ICMAZicsr then use this rv32gczicsr for the --with-arch option..

This doesn't look correct.

> rv32ICMAZicsr

The toolchain architecture string does not allow for upper case extension specifiers. The order may also be incorrect.

> rv32ICMAZicsr
> rv32gczicsr

These are not equivalent.
`g` = `imafd`.

TommyMurphyTM1234, 2023-05-25:
This mailing list thread might also be germane to this issue:

* https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/aE1ZeHHCYf4

lifong, 2023-06-22:
@InspireSemi 

Another compiler: [xuantie-gnu-toolchain](https://github.com/T-head-Semi/xuantie-gnu-toolchain)  that forked from riscv-gnu-toolchain can work with '-march=rv32imac'. Maybe it can give you some information.

```
RISC-V-Qemu-virt_GCC:lifong$ /opt/riscv-toolchain/xuantie/bin/riscv64-unknown-elf-gcc -v -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c main.c -o build/main.o
Using built-in specs.
COLLECT_GCC=/opt/riscv-toolchain/xuantie/bin/riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: /Volumes/MD/FreeRTOS/xuantie-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf CXXFLAGS='-g -O2 ' --disable-libgcctf --prefix=/opt/riscv-toolchain/xuantie --disable-shared --enable-threads=posix --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv-toolchain/xuantie/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medany'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-D' '__riscv_float_abi_soft' '-D' 'portasmHANDLE_INTERRUPT=handle_trap' '-I' '.' '-I' '../Common/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions' '-march=rv32imac' '-mabi=ilp32' '-mcmodel=medany' '-Wall' '-fmessage-length=0' '-ffunction-sections' '-fdata-sections' '-fno-builtin-printf' '-O2' '-MMD' '-MP' '-c' '-o' 'build/main.o' '-march=rv32imac'
 /opt/riscv-toolchain/xuantie/libexec/gcc/riscv64-unknown-elf/10.2.0/cc1 -quiet -v -I . -I ../Common/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -imultilib rv32imac/ilp32 -MMD build/main.d -MP -MQ build/main.o -D __riscv_float_abi_soft -D portasmHANDLE_INTERRUPT=handle_trap main.c -quiet -dumpbase main.c -march=rv32imac -mabi=ilp32 -mcmodel=medany -march=rv32imac -auxbase-strip build/main.o -O2 -Wall -version -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -o /var/folders/kq/d2j4fhss5dj1f_msqg3mp5zm0000gn/T//ccj8EPrD.s
GNU C17 (GCC) version 10.2.0 (riscv64-unknown-elf)
	compiled by GNU C version Apple LLVM 13.0.0 (clang-1300.0.29.30), GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version isl-0.26-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/riscv-toolchain/xuantie/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/riscv-toolchain/xuantie/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 .
 ../Common/include
 /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include
 /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V
 /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions
 /opt/riscv-toolchain/xuantie/lib/gcc/riscv64-unknown-elf/10.2.0/include
 /opt/riscv-toolchain/xuantie/lib/gcc/riscv64-unknown-elf/10.2.0/include-fixed
 /opt/riscv-toolchain/xuantie/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C17 (GCC) version 10.2.0 (riscv64-unknown-elf)
	compiled by GNU C version Apple LLVM 13.0.0 (clang-1300.0.29.30), GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version isl-0.26-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: abf1235dd2d23fe45464ce58cfae4512
COLLECT_GCC_OPTIONS='-v' '-D' '__riscv_float_abi_soft' '-D' 'portasmHANDLE_INTERRUPT=handle_trap' '-I' '.' '-I' '../Common/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions' '-march=rv32imac' '-mabi=ilp32' '-mcmodel=medany' '-Wall' '-fmessage-length=0' '-ffunction-sections' '-fdata-sections' '-fno-builtin-printf' '-O2' '-MMD' '-MP' '-c' '-o' 'build/main.o' '-march=rv32imac'
 /opt/riscv-toolchain/xuantie/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/as -v -I . -I ../Common/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/ [... cut ...]
COLLECT_GCC_OPTIONS='-v' '-D' '__riscv_float_abi_soft' '-D' 'portasmHANDLE_INTERRUPT=handle_trap' '-I' '.' '-I' '../Common/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions' '-march=rv32imac' '-mabi=ilp32' '-mcmodel=medany' '-Wall' '-fmessage-length=0' '-ffunction-sections' '-fdata-sections' '-fno-builtin-printf' '-O2' '-MMD' '-MP' '-c' '-o' 'build/main.o' '-march=rv32imac'
```

TommyMurphyTM1234, 2023-06-22:
> @InspireSemi
> 
> Another compiler: [xuantie-gnu-toolchain](https://github.com/T-head-Semi/xuantie-gnu-toolchain) that forked from riscv-gnu-toolchain can work with '-march=rv32imac'. Maybe it can give you some information.

There's no error or warning in the output that you posted.
What's the problem?
You're also replying to a closed issue which seems odd.

TommyMurphyTM1234, 2023-06-22:
> I think that the changes to split `zicsr` and `zifencei` out from the base integer ISA came in after this - maybe as recently as GCC 12.

See here for more info:

* https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/aE1ZeHHCYf4