riscv-collab/riscv-gnu-toolchain #1261: Problem with enabling ISA extension 'xw'

dwillmore, 2023-05-30:
I am trying to build a compiler for the WCH chip CH32V003 which has the xw extensions.  I am using:
```shell
../riscv-gnu-toolchain/configure --prefix=/home/willmore/bin/RISCV_64 --enable-multilib --with-arch=rv32ecxw --target=riscv64-unknown-elf
```
Have I configured incorrectly or is the extension not yet supported?  My errors are like:
```
Assembler messages:
Error: x ISA extension `xw' must be set with the versions
/tmp/ccEKJlZf.s:3: Error: x ISA extension `xw' must be set with the versions
```

TommyMurphyTM1234, 2023-05-30:
What is the "xw" extension?
I can't find any mention of it here:

* https://riscv.org/technical/specifications/

More specifically here:

* https://wiki.riscv.org/display/HOME/Recently+Ratified+Extensions
* https://wiki.riscv.org/display/HOME/Specification+Status
* https://wiki.riscv.org/display/HOME/RISC-V+Technical+Specifications

Where is it documented?

Maybe it's a custom extension of the WCH CM32V003 RISC-V implementation and is only supported by the accompanying MounRiver Studio toolchain?

* http://www.mounriver.com/

I don't know if/where MounRiver provide the sources for their GCC based toolchain as required by GPL and related open-source licenses.

dwillmore, 2023-05-30:
It looks like it's somthing like Zcb from what I can tell.  I saw that same stack overflow issue as well.  Looke like it's a WCH specific extension.  I'll ask them for details.  We've already asked for their compiler source or at least the changes--waiting for a reply.  I'll close this issue.  Thank you for your reply!

closed 2023-05-30 by dwillmore

TommyMurphyTM1234, 2023-05-30:
Yes - I just found these:

* https://www.eevblog.com/forum/microcontrollers/bizarre-problem-on-ch32v003-with-systick-isr-corrupting-uart-tx-data/25/

> Not sure how much of that is due to the proprietary 'XW' compact instructions

* http://www.wch-ic.com/search?t=downloads

> V2 series supports standard RV32EC instruction extensions, in addition to custom XW extension

So I guess that you'll have to persist with trying to get further info (and, ideally, code for any GCC et. al. changes to support this seemingly proprietary extension) from WCH/MounRiver. Good luck!

dwillmore, 2023-05-30:
> It looks like it's somthing like Zcb
>
> I'm not sure that this is correct - Zcb is cleary specified here:
>
>    -
>    https://wiki.riscv.org/display/HOME/Recently+Ratified+Extensions#:~:text=RISC%2DV%20Code,Zcf%2C%20Zcmp%2C%20Zcmt

It's likely not Zcb, but the instructions seem similar.  There's an effort
to fuzz out all the compressed instructions on the chip to see what's there
and what they do.  It'll take a little while, though. :)  If we end up with
a document describing the instructions, is there hope of contributing this?

TommyMurphyTM1234, 2023-05-30:
> If we end up with a document describing the instructions, is there hope of contributing this?

I don't understand - contribute it where?
Ideally WCH/MounRiver would make the necessary information available.
Certainly they are legally obliged to make available any code changes made to GCC et. al. to support a custom extension.

dwillmore, 2023-05-30:
Where does riscv-gnu-toolchain get code contributions from?  Upstream GCC?

TommyMurphyTM1234, 2023-05-30:
> Where does riscv-gnu-toolchain get code contributions from? Upstream GCC?

Support for draft standard extensions is done here because GCC does not accept support until they are ratified.
Other general and RISC-V changes/enhancements are done upstream in GCC and related projects.
So it's a two-way process.

dwillmore, 2023-05-30:
Ahh, so non-draft extensions live outside of both upstream GCC and this
project?

TommyMurphyTM1234, 2023-05-30:
> Ahh, so non-draft extensions live outside of both upstream GCC and this project?

"Non-draft" suggests "ratified" - in which case support that was developed here would be upstreamed and maintained in GCC etc.
As far as I see it there are three classes of extension:

1. Draft standard extensions - developed here while until ratified
2. Ratified standard extensions - support developmed here is upstreamed to GCC et. al. and maintained upstream from then on
3. Custom/proprietary extensions

This "xw" extension seems to fall into category 3 and I'm not sure how toolchain support is handled.

dwillmore, 2023-05-30:
Sorry, I used the wrong terminology.  Thank you for the correction.  Yes,
these are type 3.  Once we know more about them either from our RE efforts
or from WCH providing information, we can see if they overlap with an
existing either draft or ratified extensions and see where we can go from
there.

We simply want a toolchain that supports this chip (and its siblings)
fully.  If I have to maintain a set of patches to your repo or to upstream
GCC, then that's not optimal, but I fully understand that type 3 extensions
may not have a home in your repo.  If it turns out that they are, I would
be interested in contributing this to you.

AntonKrug, 2023-05-30:
[image]

TommyMurphyTM1234, 2023-05-30:
> QingKe V2 series microprocessor is a 32-bit general-purpose MCU microprocessor based on the standard RISC-V instruction set RV32I subset RV32E, with only 16 general-purpose registers, half of RV32I, and a more streamlined structure for deep embedded scenarios. V2 series supports standard RV32EC instruction extensions, **in addition to custom XW extensions**, Hardware Prologue/Epilogue (HPE), Vector Table Free (VTF), a more streamlined single-wire serial debug interface (SDI), and support for "WFE" instructions.
> 
> [image]

Thanks @AntonKrug.

dwillmore, 2023-05-30:
I appreciate your clarification as to the terminology.  I'm most concerned that we are understanding each other and your terminology solved that.

Yes, by 'fully' I mean we want to take advantage of all the instructions the chip offers--should ever manage to determine what those are. :)

They did alter GCC to support their chip and then then distributed that binary to their customers, so they are bound by the GPL to distribute their changes.  If we can manage to make them understand that and then secure their compliance is anyones guess.  We will make the effort, though.  They do seem to be in the middle of creating a repo for their tools.  I hope we'll see the code show up at some point as well.

Regardless, we're undertaking the effort to RE the timings of the chip and part of the will be to see what instructions are supported.  

Their toolset can be coaxed into decoding their instructions it seems, so that may help in the effort.  Assuming it can be trusted.

Thank you for all of your assistance, @TommyMurphyTM1234 and @AntonKrug

TommyMurphyTM1234, 2023-05-30:
Hopefully this has changed at this stage or will change soon...

* https://www.reddit.com/r/RISCV/comments/126262j/comment/jenk1gm/

> We're at a disadvantage of REALLY knowing because while we've sort of reverse engineered the behaviour of their chopped up GCC and found it "just" changes a ret to an mret, they continue to violate the GPL and won't provide the source to their GCC, even upon request from their users. This is a violation.

ilg-ul, 2023-06-01:
From a binary distribution maintainer point of view, my releases are based only on the upstream GNU GCC source code. If the upstream code includes support for `xw` or whatever extensions, my distribution will include it, otherwise it will not. It's that simple.

dwillmore, 2023-06-01:
@ilg-ul I'm looking to build the binaries that we can ship with a development environment for a specific chip.  If I were doing a distro, I'd probably think your way as well.

dwillmore, 2023-06-01:
And we're always going to provide source, because GPL is a thing. :)

ilg-ul, 2023-06-02:
> build the binaries that we can ship with a development environment for a specific chip

Sure, if you want to discourage people who use different development environments to use your specific chip, trying to lock them up into a specific environment is an efficient way to achieve it.

Otherwise toolchains should be distributed separately, not bundled within any development environment, and be available on all modern platforms, including for running unit tests in scriptable CI environments.

That's why there are 7 different xPack binaries, that can be automatically installed via `xpm`:

- <https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v12.2.0-3>

TommyMurphyTM1234, 2023-06-02:
> Sure, if you want to discourage people who use different development environments to use your specific chip, trying to lock them up into a specific environment is an efficient way to achieve it. [...]

Unfortunately, I think that the RISC-V (GCC - presumably the same for LLVM/clang?) toolchain fragmentation ship has already sailed just looking at the increasing instances of vendor custom or pre-ratification extensions and specific supporting tools - e.g. Alibaba T-Head, WCH etc.