riscv-collab/riscv-gnu-toolchain #1262: Extension `zicsr' required

rohanjuneja, 2023-05-31:
While executing 'riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32', am getting the following error.
'Error: unrecognized opcode `csrs mstatus,t0', extension `zicsr' required'

I use the following command to configure the toolchain.
'./configure --disable-gdb --disable-linux --enable-multilib'

TommyMurphyTM1234, 2023-05-31:
You're specifying `-march=rv32imac` but are using CSR instructions. You need instead to use `-march=rv32imaczicsr`.

TommyMurphyTM1234, 2023-05-31:
BTW - when `riscv-gnu-toolchain` is finally bumped to GCC 13:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239

The multilib matching rules will change again and it may be the case that `rv32imac` will automagically match `rv32imac_zicsr` but I'm not completely sure how the new multilib matching rules work yet.

TommyMurphyTM1234, 2023-05-31:
In your original post you said:

> While executing 'riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32', am getting the following error.
'Error: unrecognized opcode csrs mstatus,t0', extension zicsr' required'

In which case it doesn't make much sense to me that you're doing this:

> My ./configure line and compile lines from our build script for the tools.
> ./configure --prefix=/opt/riscv64_12.2 --with-cmodel=medany --enable-qemu-system --with-arch=rv64gc_zicsr_zifencei --with-abi=lp64d

I.e. configuring the tools for `rv64gc_zicsr_zifencei/lp64d`.

> I am getting the same error as above it does not recognize cssr opcode.

When? When building the toolchain? When attempting to compile code with the resulting toolchain? With what command line?

It would help if you posted more clearly about what you are doing, what problems you are having, provide exact instructions to reproduce your issue, attach logs where relevant etc.

> but the RVI and the Community without some type of release process for the tools that performs a minimal test suite on the compiler cannot move forward

There is a test suite and CI jobs that are run via GitHub.
But some changes - including the ABI change that resulted in `zicsr` and `zifencei` being split from the `I` base extension - can unfortunately impact the tools end user as is the case here.

InspireSemi, 2023-05-31:
I am NOT the original poster..

I am commenting on his post saying I am seeing the same thing on rv64 compiler building with what "should be" the proper ./configure line.

I had an open issue on this that I thought was closed when I was given the gc_zicsr_zifencei option for ./configure.  But it looks like this is not the correct option or the build is broken?

What is teh ./configure line in the CI that is run?  Maybe that has it right?

Marc Karasek

Principal Software Engineer

M: 678.770.3788

[A close up of a sign  Description automatically generated]

www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

TommyMurphyTM1234, 2023-05-31:
> I am NOT the original poster.. 

Apologies, my mistake.

InspireSemi, 2023-05-31:
I am building with --with-arch=rv64gczicsr_zifencei

I built with --with-arch=rv64gc_zicsr_zifencei and this still fails to compile csr opcodes.

Marc Karasek

Principal Software Engineer

M: 678.770.3788

[A close up of a sign  Description automatically generated]

www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

InspireSemi, 2023-05-31:
> This worked for me:
>
>   1.  Take the latest nightly build (specifically riscv64-elf-ubuntu-22.04-nightly-2023.05.31-nightly.tar.gz<https://url.avanan.click/v2/___https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2023.05.31/riscv64-elf-ubuntu-22.04-nightly-2023.05.31-nightly.tar.gz___.YXAzOmluc3BpcmVzZW1pOmE6bzphMDBmNGFiMmEwMzUwZWY2MTM0YTM2NjkzZTdmMGJkNTo2OjUxODk6ZDNhOTQ4ZjFiMzg5MTFlODZiMGRmZTg0MjA2YzdiNjFlMWRhMjVkYjllZWMxOTcyMmVkZjY0ZGNhZjcyMTQ4MDpoOlQ>) from here: https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.05.31<https://url.avanan.click/v2/___https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.05.31___.YXAzOmluc3BpcmVzZW1pOmE6bzphMDBmNGFiMmEwMzUwZWY2MTM0YTM2NjkzZTdmMGJkNTo2OjZiNzM6YThlYTNiNTkxNTE5MjFiZjg2ZTk5NTQ0ODZiZWI1YzBjZjg0ZTNhM2Q3NzQ1ZDBjY2VhODYzMTIyMjZmMGU4YzpoOlQ>
>   2.  Extracted that to a directory
>   3.  Note that riscv64-unknown-elf-gcc -dumpspecs shows that this toolchain was configured for rv64imafdc_zicsr_zifencei/lp64d so these will be the default arch/abi in the absence of any explicit -march=... -mabi=... options at compile time:
>
> *multilib_defaults:
> march=rv64imafdc_zicsr_zifencei mabi=lp64d
>
>   1.  Create a trivial example program:
>
> # test.c
> int main(int argc, char *argv[])
> {
>   asm volatile ("csrr t0, misa");
>   return 0;
> }
>
>   1.  Compile this:
>
> riscv64-unknown-elf-gcc test.c -o test.exe
>
>   1.  Disassemble the resulting executable:
>
> riscv64-unknown-elf-objdump -dS test.exe > test.dis
> [... 11 lines cut ...]
>    101b0:       4781                    li      a5,0
>    101b2:       853e                    mv      a0,a5
>    101b4:       6462                    ld      s0,24(sp)
>    101b6:       6105                    add     sp,sp,32
>    101b8:       8082                    ret

Ok 1 difference is I am building from the tip of the tree.

My -dumpspecs shows:

*multilib_defaults:
march=rv64imafdc_zicsr_zifencei mabi=lp64d

It still fails to compile a prog with a csr in it?
I am a bit behind the current tip at 2 commits past the tag of 20-23.05.19

I will git fetch/rebase and erbuild and see what happens..

Marc

Marc Karasek

Principal Software Engineer

M: 678.770.3788

[A close up of a sign  Description automatically generated]

www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

TommyMurphyTM1234, 2023-05-31:
I'm also trying the following to see about `rv32imac_zicsr`...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc -b releases/gcc-13 gcc-13
./configure --prefix=`pwd`/installed-tools --with-multilib-generator="rv32imac-ilp32--;rv32imac_zicsr-ilp32--;rv32imac_zicsr_zifencei-ilp32--" --with-gcc-src=`pwd`/gcc-13
make
```
Note that I am using the GCC 13 sources here because of changes to how the multilib stuff is done and in anticipation of this PR being finally integrated:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239

TommyMurphyTM1234, 2023-05-31:
> Ok 1 difference is I am building from the tip of the tree. My -dumpspecs shows: *multilib_defaults: march=rv64imafdc_zicsr_zifencei mabi=lp64d It still fails to compile a prog with a csr in it? I am [...]

I don't fully understand.
It would be clearer if you outlined all steps that you take to clone, configure and build your toolchain.

InspireSemi, 2023-05-31:
We need to be able to compile from source for a number of reasons..  So grabbing a tarball of a nightly build is not an option for us.

Marc Karasek

Principal Software Engineer

M: 678.770.3788

[A close up of a sign  Description automatically generated]

www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

  *   #1239<https://url.avanan.click/v2/___https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239___.YXAzOmluc3BpcmVzZW1pOmE6bzozMjljN2ViOGMxNGRiNjQ2Zjc0ODUzYTUxYTRiYzI5NDo2OjkxMjk6ZTg5Nzk0YTkzNzNhYjY5ZjgwODA2Y2I4YjJkMWU2OTAwMTQ5ZTkwZWViODEwYTliZDk5MjFhYzJhNzQzZmI2ZTpoOlQ>

TommyMurphyTM1234, 2023-05-31:
> We need to be able to compile from source for a number of reasons.. So grabbing a tarball of a nightly build is not an option for us.

I wasn't suggesting that it was but I am trying to tease out the issue.
And obviously the nightly build shows that it is possible to build a toolchain that can deal with `rvXX..._zicsr` and deal with CSR instructions.

TommyMurphyTM1234, 2023-05-31:
> This worked for me:
> 
> 1. Take the latest nightly build (specifically [`riscv64-elf-ubuntu-22.04-nightly-2023.05.31-nightly.tar.gz`](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2023.05.31/riscv64-elf-ubuntu-22.04-nightly-2023.05.31-nightly.tar.gz)) from here: https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.05.31 and extract that to a directory
> 2. Note that `riscv64-unknown-elf-gcc -dumpspecs` shows that this toolchain was configured for `rv64imafdc_zicsr_zifencei/lp64d` so these will be the default `arch/abi` in the absence of any explicit `-march=... -mabi=...` options at compile time:
> 
> ```
> *multilib_defaults:
> march=rv64imafdc_zicsr_zifencei mabi=lp64d
> ```
> 
> 3. Create a trivial example program that uses a CSR instruction:
> 
> ```
> # test.c
> int main(int argc, char *argv[])
> {
>   asm volatile ("csrr t0, misa");
>   return 0;
> }
> ```
> 
> 4. Compile this:
> 
> ```
> riscv64-unknown-elf-gcc test.c -o test.exe
> ```
> 
> 5. Disassemble the resulting executable:
> 
> ```
> riscv64-unknown-elf-objdump -dS test.exe > test.dis
> ```
> 
> 6. The `csrr` instruction is present and correct:
> 
> ```
> # test.dis
> ...
> 000000000001019c <main>:
>    1019c:       1101                    add     sp,sp,-32
>    1019e:       ec22                    sd      s0,24(sp)
>    101a0:       1000                    add     s0,sp,32
>    101a2:       87aa                    mv      a5,a0
>    101a4:       feb43023                sd      a1,-32(s0)
>    101a8:       fef42623                sw      a5,-20(s0)
>    101ac:       301022f3                csrr    t0,misa
>    101b0:       4781                    li      a5,0
>    101b2:       853e                    mv      a0,a5
>    101b4:       6462                    ld      s0,24(sp)
>    101b6:       6105                    add     sp,sp,32
>    101b8:       8082                    ret
> ```

BTW - with the same toolchain, this also works - because no matter what configuration time `--with-arch=... --with-abi=` is used. the resulting toolchain can always generate code for all standard extensions. But it may not be able to link with startup/library code if the required multilibs are not available.
```
riscv64-unknown-elf-gcc -march=rv32imac_zicsr -mabi=ilp32 -c test.c
riscv64-unknown-elf-objdump -dS test.o

test.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <main>:
   0:	1101                	add	sp,sp,-32
   2:	ce22                	sw	s0,28(sp)
   4:	1000                	add	s0,sp,32
   6:	fea42623          	sw	a0,-20(s0)
   a:	feb42423          	sw	a1,-24(s0)
   e:	301022f3          	csrr	t0,misa
  12:	4781                	li	a5,0
  14:	853e                	mv	a0,a5
  16:	4472                	lw	s0,28(sp)
  18:	6105                	add	sp,sp,32
  1a:	8082                	ret
```

InspireSemi, 2023-05-31:
Just tried the new toolchain built from the current tip and it fails

Error: unrecognized opcode `csrr a0,mhartid', extension `zicsr' required

-dumpspecs output

*multilib_defaults:
march=rv64imafdc_zicsr_zifencei mabi=lp64d

I am assuming Jenkins is building the toolchain for the tag releases?  What ./configure options is it using?

Marc Karasek

Principal Software Engineer

M: 678.770.3788

[A close up of a sign  Description automatically generated]

www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

TommyMurphyTM1234, 2023-06-01:
> Just tried the new toolchain built from the current tip and it fails Error: unrecognized opcode `csrr a0,mhartid', extension `zicsr' required
> […](#)
> -dumpspecs output *multilib_defaults: march=rv64imafdc_zicsr_zifencei mabi=lp64d 

Still doens't make sense to me that if your toolchain is configured for `rv64imafdc_zicsr_zifencei/lp64d` by default and you are not passing any `-march=... -mabi=...` options at compile time, you are still getting that error.
I'll try with my own build of the tools when it's finished (had to pause it overnight).

> I am assuming Jenkins is building the toolchain for the tag releases? 

The CI is done via GitHub actions.

You can see the actions here.

* rv64imafdc_zicsr_zifencei mabi=lp64d 

> What ./configure options is it using? 

Curiously the actions seem to be using, say, `--with-arch=rv64gc --with-abi=lp64d` 

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/workflows/nightly-release.yaml#L53
* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/workflows/nightly-release.yaml#L70

TommyMurphyTM1234, 2023-06-01:
I build the toolchain as per my earlier post:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1262#issuecomment-1570976300

Which gives me:
```
riscv64-unknown-elf-gcc -dumpspecs

...

*multilib:
. !march=rv32imac !march=rv32imac_zicsr !mabi=ilp32;rv32imac/ilp32 march=rv32imac !march=rv32imac_zicsr mabi=ilp32;rv32imac_zicsr/ilp32 !march=rv32imac march=rv32imac_zicsr mabi=ilp32;

*multilib_defaults:
march=rv64imafdc_zicsr mabi=lp64d

...

iscv64-unknown-elf-gcc -print-multi-lib
.;
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imac_zicsr/ilp32;@march=rv32imac_zicsr@mabi=ilp32
```
So the, in the absence of me providing any `--with-arch=... --with-abi=...` at configuration time, it defaults to `rv64imafdc_zicsr/lp64d`. I was kind of expecting `rv64imafdc_zicsr_zifencei/lp64d` but how and ever...

I then used the trivial `csrr` `test.c` here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1262#issuecomment-1570916261
````
riscv64-unknown-elf-gcc -c test.c
riscv64-unknown-elf-objdump -dS test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <main>:
   0:	1101                	add	sp,sp,-32
   2:	ec22                	sd	s0,24(sp)
   4:	1000                	add	s0,sp,32
   6:	87aa                	mv	a5,a0
   8:	feb43023          	sd	a1,-32(s0)
   c:	fef42623          	sw	a5,-20(s0)
  10:	301022f3          	csrr	t0,misa
  14:	4781                	li	a5,0
  16:	853e                	mv	a0,a5
  18:	6462                	ld	s0,24(sp)
  1a:	6105                	add	sp,sp,32
  1c:	8082                	ret

riscv64-unknown-elf-gcc test.c -o test.exe
riscv64-unknown-elf-objdump -h test.exe

test.exe:     file format elf64-littleriscv

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         00000398  00000000000100e8  00000000000100e8  000000e8  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .eh_frame     00000004  0000000000011480  0000000000011480  00000480  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  2 .init_array   00000010  0000000000011488  0000000000011488  00000488  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  3 .fini_array   00000008  0000000000011498  0000000000011498  00000498  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  4 .data         00000748  00000000000114a0  00000000000114a0  000004a0  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  5 .sdata        00000018  0000000000011be8  0000000000011be8  00000be8  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  6 .bss          00000038  0000000000011c00  0000000000011c00  00000c00  2**3
                  ALLOC
  7 .comment      0000003f  0000000000000000  0000000000000000  00000c00  2**0
                  CONTENTS, READONLY
  8 .riscv.attributes 0000004b  0000000000000000  0000000000000000  00000c3f  2**0
                  CONTENTS, READONLY

````
And then in 32-bit mode:
```
riscv64-unknown-elf-gcc -march=rv32imac_zicsr -mabi=ilp32 -c test.c
riscv64-unknown-elf-objdump -dS test.o

test.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <main>:
   0:	1101                	add	sp,sp,-32
   2:	ce22                	sw	s0,28(sp)
   4:	1000                	add	s0,sp,32
   6:	fea42623          	sw	a0,-20(s0)
   a:	feb42423          	sw	a1,-24(s0)
   e:	301022f3          	csrr	t0,misa
  12:	4781                	li	a5,0
  14:	853e                	mv	a0,a5
  16:	4472                	lw	s0,28(sp)
  18:	6105                	add	sp,sp,32
  1a:	8082                	ret

riscv64-unknown-elf-gcc -march=rv32imac_zicsr -mabi=ilp32 test.c -o test.exe
riscv64-unknown-elf-objdump -h test.exe

test.exe:     file format elf32-littleriscv

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000378  00010094  00010094  00000094  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .eh_frame     00000004  0001140c  0001140c  0000040c  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  2 .init_array   00000008  00011410  00011410  00000410  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  3 .fini_array   00000004  00011418  00011418  00000418  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  4 .data         00000428  00011420  00011420  00000420  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  5 .sdata        0000000c  00011848  00011848  00000848  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  6 .bss          0000001c  00011854  00011854  00000854  2**2
                  ALLOC
  7 .comment      0000003f  00000000  00000000  00000854  2**0
                  CONTENTS, READONLY
  8 .riscv.attributes 00000041  00000000  00000000  00000893  2**0
                  CONTENTS, READONLY
```
I don't get any unexpected behaviour here.

InspireSemi, 2023-06-01:
Something you said led me to root cause.

I have to go back to the riscv arch test guys.  This is an issue in the way the makefile is put together..

It is setting -march=rv64ic

which explains al lot

Marc Karasek

Principal Software Engineer

M: 678.770.3788

[A close up of a sign  Description automatically generated]

www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

  *   #1262 (comment)<https://url.avanan.click/v2/___https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1262%23issuecomment-1570976300___.YXAzOmluc3BpcmVzZW1pOmE6bzo5OGI3NGRlYjU0ZGQ1YTdlYjkzZmNjODA1NTk4OGQwMzo2OmI3NDQ6MWQ4YTFhMzA2ZTY1YmM5Y2RhZjljZmUzZDQ2NjEwMTY3MDY4OTI3MmMyNDhmNGU3NDYyYmViN2NkYjAyYWFhMjpoOlQ>

  *   #1262 (comment)<https://url.avanan.click/v2/___https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1262%23issuecomment-1570916261___.YXAzOmluc3BpcmVzZW1pOmE6bzo5OGI3NGRlYjU0ZGQ1YTdlYjkzZmNjODA1NTk4OGQwMzo2OjgyMzk6ZTQ3YjY2ZjI5NzZlYjA1MDgzZWJhMTdmMmUyMGFkMTQwY2MyOWRmZjVhZTlkODJlZThkZjBlNzFlOTc4YTU2NzpoOlQ>

InspireSemi, 2023-06-01:
From comments above gcc 13 is changing this again?

TommyMurphyTM1234, 2023-06-01:
> From comments above gcc 13 is changing this again?

I don't know.
I'm just building using the GCC 12 sources pointed to by this repo now to check that.

TommyMurphyTM1234, 2023-06-01:
> Something you said led me to root cause. I have to go back to the riscv arch test guys. This is an issue in the way the makefile is put together.. It is setting -march=rv64ic which explains al lot 

That's why it's good to post only self contained reproducible test cases and full instructions on how to reproduce the issue instead of depending on assumed info.

TommyMurphyTM1234, 2023-06-01:
Another piece of the puzzle:
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gc --with-abi=lp64d
make

cd installed-tools
./riscv-gnu-toolchain-gcc -dumpspecs

...

*multilib_defaults:
march=rv64imafdc_zicsr_zifencei mabi=lp64d

...
```
So, if `--with-arch=rv64gc` is specified at configuration time then it gets expanded somewhere to `rv64imafdc_zicsr_zifencei`.

TommyMurphyTM1234, 2023-06-01:
Ultimately, while some confusion can arise as a result of the splitting of `Zicsr` (and `Zifencei`) from the base `I` ISA, for tool changes to cater for this, and from the automatic expansion of, say, `rv64gc` to `rv64imafdc_zicsr_zifencei` in certain situations, it seems to me that this issue is not actually a valid bug and should be closed.

InspireSemi, 2023-06-01:
This is very confusing.... hopefully with the b new Linux foundation announcement this can be handled better

TommyMurphyTM1234, 2023-06-01:
> This is very confusing.... hopefully with the b new Linux foundation announcement this can be handled better Sent from my 

I don't know what Linux foundation announcement you are referring to.
And I'm not sure that this would be relevant to non Linux toolchains such as for bare metal or RTOSs?

InspireSemi, 2023-06-01:
Yes I agree

TommyMurphyTM1234, 2023-06-01:
> Ultimately, while some confusion can arise as a result of the splitting of `Zicsr` (and `Zifencei`) from the base `I` integer ISA, from tool changes to cater for this, and from the automatic expansion of, say, `rv64gc` to `rv64imafdc_zicsr_zifencei` in certain situations, it seems to me that this issue is not actually a valid bug and should be closed.

Closing the issue.