riscv-collab/riscv-gnu-toolchain #1272: fail to make cross compiler

JenniferBrana, 2023-06-11:
Hi,
I have been attempting to make both elf-gcc and linux-gcc. Both fail to build with the same error messages. 

My environment: Ubuntu 18.04.4 LTS, GCC version: 7.5.0.

After a long compile time, I received the error below. Does anyone know how to resolve this? 

make[5]: Entering directory '/h/jbrana/riscv/riscv-gnu-toolchain/build-binutils-linux/ld'
  GEN      configdoc.texi
  MAKEINFO ld.info
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ld.texi:903: warning: @pxref node name should not contain `.'
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ld.texi:9049: warning: @pxref node name should not contain `.'
  CC       libdep_plugin.lo
  GEN      libdep.la
libtool: link: warning: ignoring multiple `-rpath's for a libtool library
  CC       libldtestplug_la-testplug.lo
  CCLD     libldtestplug.la
  CC       libldtestplug2_la-testplug2.lo
  CCLD     libldtestplug2.la
  CC       libldtestplug3_la-testplug3.lo
  CCLD     libldtestplug3.la
  CC       libldtestplug4_la-testplug4.lo
  CCLD     libldtestplug4.la
  CC       ldgram.o
  LEX      ldlex.c
  CC       ldlex-wrapper.o
In file included from /h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex-wrapper.c:26:0:
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l: In function ‘yyrestart’:
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l:937:17: warning: unused variable ‘new_state_buf’ [-Wunused-variable]
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l:936:10: warning: unused variable ‘new_size’ [-Wunused-variable]
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l: In function ‘yy_switch_to_buffer’:
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l:965:17: warning: unused variable ‘new_state_buf’ [-Wunused-variable]
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l:964:10: warning: unused variable ‘new_size’ [-Wunused-variable]
In file included from /h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex-wrapper.c:26:0:
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l: In function ‘yy_create_string_buffer’:
/h/jbrana/riscv/riscv-gnu-toolchain/binutils/ld/ldlex.l:544:6: error: ‘struct yy_buffer_state’ has no member named ‘yy_at_bol’; did you mean ‘yyatbol’?
   b->yy_fill_buffer = 0;
      ^~~~~~~~~
      yyatbol
Makefile:2229: recipe for target 'ldlex-wrapper.o' failed
make[5]: *** [ldlex-wrapper.o] Error 1
make[5]: Leaving directory '/h/jbrana/riscv/riscv-gnu-toolchain/build-binutils-linux/ld'
Makefile:1835: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/h/jbrana/riscv/riscv-gnu-toolchain/build-binutils-linux/ld'
Makefile:1073: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/h/jbrana/riscv/riscv-gnu-toolchain/build-binutils-linux/ld'
Makefile:8159: recipe for target 'all-ld' failed
make[2]: *** [all-ld] Error 2
make[2]: Leaving directory '/h/jbrana/riscv/riscv-gnu-toolchain/build-binutils-linux'
Makefile:1004: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/h/jbrana/riscv/riscv-gnu-toolchain/build-binutils-linux'
Makefile:317: recipe for target 'stamps/build-binutils-linux' failed
make: *** [stamps/build-binutils-linux] Error 2

TommyMurphyTM1234, 2023-06-11:
Please provide full details on how your building the tools for both cases.

1. `git clone` command lines
2. `configure` command line
3. `make` command lines 

I built both the bare metal/newlib and Linux toolchains recently (last week) with no issues.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/newlib-installed-tools
make newlib 
make distclean
./configure --prefix=`pwd`/linux-installed-tools
make Linux
```

TommyMurphyTM1234, 2023-06-11:
> My environment: Ubuntu 18.04.4 LTS, GCC version: 7.5.0.

GCC 7.5.0 is very old.
Even if it's not the cause of your problems you should probably compile the RISC-V tools with a more recent host toolchain.

TommyMurphyTM1234, 2023-06-12:
I'm just trying my build instructions above on a clean Ubuntu 18.04 virtual machine and will post back with results as soon as I can.

TommyMurphyTM1234, 2023-06-12:
OK - the newlib toolchain build succeeded.

I installed a Ubuntu 18.04 VirtualBox virtual machine, installed all available updates, installed the VirtualBox guest additions, installed the `riscv-gnu-toolchain` build prerequisites and then ran the following:
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/newlib-installed-tools
date > make-newlib.log && make newlib 2>&1 | tee -a make-newlib.log && date >> make-newlib.log
```
Compressed build log: [make-newlib.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11721217/make-newlib.zip)

I'll try the Linux toolchain build now.

But it looks like the issue may be with your host setup or something.

TommyMurphyTM1234, 2023-06-13:
Invalid issue - not reproducible.

closed 2023-06-13 by TommyMurphyTM1234