riscv-collab/riscv-gnu-toolchain #1275: adding custom instruction to riscv gcc toolchain

eman877, 2023-06-13:
hi,  i am now adding push custom instruction to riscv32-gcc toolchain i will modify riscv-opc.c and riscv-opc.h and riscv.md for adding the functionalty of the push instruction and the functionality of the push is to saves the registers r1, [r5:r31] in the memory starting from location sp + 8 and with incrementing offset + 4 the rest of the register such that r5 will be saved in sp + 12 , r6 in sp + 16 , it does not take any operands or destination as every thisng is specified in the implementation of the hardware (processor ,memory) , so is these modifications are enough ? and when i modify the riscv-opc.c when i write the INSN_CLASS which one is suitable or do i need to define new one ?

TommyMurphyTM1234, 2023-06-13:
Please review existing (open and closed) issues that deal with the question of adding custom instructions to the toolchain. And upstream guides. Explaining how to do this is outside the scope of this repo.

closed 2023-06-13 by TommyMurphyTM1234

eman877, 2023-06-13:
please where can i ask my question?

TommyMurphyTM1234, 2023-06-13:
As I said, existing (closed) issues here probably have the answer.
But, if not, then you should ask on the upstream tools (binutils, gcc etc.) mailing lists or other forums.