riscv-collab/riscv-gnu-toolchain #1277: Missing riscv64-unknown-linux-gnu-pkg-config, cannot cross compile qemu

sunhaotian0122, 2023-06-15:
I run the following command to generate a cross compile toolchain for RISC-V.

```
./configure --prefix=/opt/riscv --enable-multilib
make linux
```
[image]

However, riscv64-unknown-linux-gnu-pkg-config does not exist, causing cross-compilation of qemu to fail. 
The error message is as follows:

```
./configure --target-list="riscv32-softmmu riscv64-softmmu" --cross-prefix=/opt/riscv/bin/riscv64-unknown-linux-gnu-
Using './build' as the directory for build output

ERROR: pkg-config binary '/opt/riscv/bin/riscv64-unknown-linux-gnu-pkg-config' not found
```
What should I do to compile riscv64-unknown-linux-gnu-pkg-config ?

TommyMurphyTM1234, 2023-06-15:
You'll need to post your full build steps - the `git clone`, `configure` and `make`` command lines that you use.
And maybe a full (compressed) build log.
I did a similar build and did not get that error - but neither did it build QEMU because I think that only happens if you run the test suite?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib
make linux 2>&1 | tee build.log
```
Build log (compressed): [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/11756125/build.log.tar.gz)

TommyMurphyTM1234, 2023-06-15:
Just to follow up - I did a toolchain build and then a test suite run using QEMU without any errors.
Here is exactly what I did on a clean Zorin OS Lite 16.2 (Ubuntu 20.04 derivative) VirtualBox virtual machine with all available updates applied.
```
# Install prerequisites- https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
sudo apt install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev

# Install more prerequisites - https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1251
sudo apt install git python pip expect device-tree-compiler libglib2.0-dev

# Build linux multilib toolchain
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib
make 2>&1 | tee build.log

# Run test suite - by default this builds QEMU and runs the test suite
make report-linux 2>&1 | tee test.log
```
Build log (compressed): same as previous post
Test log (compressed): TODO

I cannot reproduce your issue.

TommyMurphyTM1234, 2023-06-15:
> `./configure --target-list="riscv32-softmmu riscv64-softmmu" --cross-prefix=/opt/riscv/bin/riscv64-unknown-linux-gnu-`

This (`--cross-prefix`) suggests that you may be trying to build QEMU to run ON a RISC-V target.
Is that correct? If so - why? Seems odd.
Normally one would build QEMU to run on, say, x86 Linux or Windows, and emulate RISC-V there.

TommyMurphyTM1234, 2023-06-16:
> My current conclusion is that this is an invalid issue as far as this `riscv-gnu-toolchain` repo is concerned and, as such, liable for closure.

Closing.

closed 2023-06-16 by TommyMurphyTM1234