riscv-collab/riscv-gnu-toolchain #1278: debug ,I can not step through my code

peter-samir-luis, 2023-06-16:
this is the command i used 

riscv32-unknown-elf-gcc hello.c -o hello
qemu-system-riscv32 -machine virt -cpu rv32,pmp=false -smp 1 -s -S -nographic -bios none -kernel hello

in another session 
riscv32-unknown-elf-gdb hello -ex "target remote :1234"

==========

execution jump to address 0x80000000 which is empty, while main code is loaded in a different address which is not accessible.
so I can not step through my code

debug session :

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hello...
(No debugging symbols found in hello)
Remote debugging using :1234
0x00001000 in ?? ()
(gdb) display /i $pc

1: x/i $pc

=> 0x1000:	auipc	t0,0x0

(gdb) x/10i $pc

=> 0x1000:	auipc	t0,0x0

   0x1004:	addi	a2,t0,40

   0x1008:	csrr	a0,mhartid

   0x100c:	lw	a1,32(t0)

   0x1010:	lw	t0,24(t0)

   0x1014:	jr	t0

   0x1018:	.2byte	0x0

   0x101a:	.2byte	0x8000

   0x101c:	.2byte	0x0

   0x101e:	.2byte	0x0

(gdb) si 5

0x00001014 in ?? ()

1: x/i $pc

=> 0x1014:	jr	t0

(gdb) si

0x80000000 in ?? ()

1: x/i $pc

=> 0x80000000:	.2byte	0x0

(gdb) x/10i $pc

=> 0x80000000:	.2byte	0x0
   0x80000002:	.2byte	0x0
   0x80000004:	.2byte	0x0
   0x80000006:	.2byte	0x0
   0x80000008:	.2byte	0x0
   0x8000000a:	.2byte	0x0
   0x8000000c:	.2byte	0x0
   0x8000000e:	.2byte	0x0
   0x80000010:	.2byte	0x0
   0x80000012:	.2byte	0x0

(gdb) x/10i main
   0x1018c <main>:	Cannot access memory at address 0x1018c

TommyMurphyTM1234, 2023-06-16:
> `riscv32-unknown-elf-gcc hello.c -o hello`

If you're going to debug your code then you really should enable debug info (`-g`).

> `0x00001000 in ?? ()`

Isn't that in the OpenSBI boot ROM which is where the machine execution is going to start before vectoring to your program>

* https://github.com/qemu/qemu/blob/5692a39f329413a00020a61fff95aff6b9884a73/hw/riscv/virt.c#L79
* https://github.com/qemu/qemu/blob/5692a39f329413a00020a61fff95aff6b9884a73/hw/riscv/virt.c#L1462
* https://github.com/qemu/qemu/blob/5692a39f329413a00020a61fff95aff6b9884a73/include/hw/riscv/boot.h#L27

> execution jump to address 0x80000000 which is empty

Above you say that execution starts at `0x00001000` - which is it?
What do you mean by "empty"?

> while main code is loaded in a different address which is not accessible.

What address?
Where is your progam linked for?

> so I can not step through my code

On what basis do you consider this to be a `riscv-gnu-toolchain` issue/bug as opposed to incorrect use of the tools?

TommyMurphyTM1234, 2023-06-16:
If you compile like this:
```
riscv32-unknown-elf-gcc -g hello.c -o hello
```
Then the program will be linked such that the following addresses apply:
```
riscv32-unknown-elf-objdump -D hello

...

000100c0 <_start>:
   100c0:       00004197                auipc   gp,0x4
   100c4:       be818193                add     gp,gp,-1048 # 13ca8 <__global_pointer$>

...

0001013e <main>:
#include <stdio.h>

int main(int argc, const char *argv[])
{
   1013e:       1101                    add     sp,sp,-32
   10140:       ce06                    sw      ra,28(sp)
   10142:       cc22                    sw      s0,24(sp)
   10144:       1000                    add     s0,sp,32
   10146:       fea42623                sw      a0,-20(s0)
   1014a:       feb42423                sw      a1,-24(s0)
    printf("Hello, World\n");

...

````
So, I'm not sure where the address `0x8000000` comes in?

peter-samir-luis, 2023-06-16:
sorry, I'm very beginner with riscv and qemu

in the debug session in my first comment :
execution starts with (0x1000:	auipc	t0,0x0) to  (0x1014:	jr	t0)  then when i use step into (si) it jumps to address (0x80000000:	.2byte	0x0 ) 

perform this command (x/10i main) it says  (0x1018c <main>:	Cannot access memory at address 0x1018c) 

I'm confused it jumps to 0x80000000 and main lable at 0x1018c

in this blog, he used a simple linker script that loads the code at memory address 0x800000000 and it works well for me  
[(https://danielmangum.com/posts/risc-v-bytes-privilege-levels/)]

i tried this (riscv32-unknown-elf-gcc -g hello.c -o hello) and it is the same thing

TommyMurphyTM1234, 2023-06-16:
I built the toolchain and QEMU from this repo as follows:
```
# Install prerequisites: https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
# Install additional prerequisites: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1251

git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32d
make
make build-qemu
cd installed-tools/bin

# Create a simple "Hello, World!" example program hello.c

./riscv32-unknown-elf-gcc -g hello.c -o hello

./qemu-riscv32 -g 3333 hello

# In another terminal
./riscv32-unknown-elf-gdb hello
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv32-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hello...
(gdb) target remote :3333
Remote debugging using :3333
0x000100c0 in _start ()
(gdb) b main
Breakpoint 1 at 0x1014e: file hello.c, line 5.
(gdb) cont
Continuing.

Breakpoint 1, main (argc=1, argv=0x40800264) at hello.c:5
5	    printf("Hello, World\n");
(gdb) si
0x00010150	5	    printf("Hello, World\n");
(gdb) si
0x00010154	5	    printf("Hello, World\n");
(gdb) si
0x000102ec in puts ()
(gdb) 

...
```
So nothing wrong there.

You probably need to read up elsewhere on how to use RISC-V tools including the toolchain, gdb, QEMU etc.

closed 2023-06-16 by TommyMurphyTM1234

peter-samir-luis, 2023-06-23:
thanks, I followed your instructions and Qemu works well.
I used these configurations 
`./configure --prefix=home/installed-Tools --with-arch=rv32i_zicsr --with-abi=ilp32`

but Qemu reports this error when I use CSR instructions
`Program received signal SIGILL, Illegal instruction.
main () at main.S:2
2	main: csrw mepc, t0`

is there a way to enable CSR instructions in Qemu?

TommyMurphyTM1234, 2023-06-23:
How are you invoking QEMU?

peter-samir-luis, 2023-06-23:
I'm using the following commands :
`riscv32-unknown-elf-gcc -g test.S -o test
qemu-riscv32 -g 3333 test
 
// in another terminal
riscv32-unknown-elf-gdb test -ex "target remote :3333" `

TommyMurphyTM1234, 2023-06-24:
I can't seem to locate comprehensive documentation on the RISC-V specific options that QEMU supports, but this link suggests that it might be possible to use `Zicsr=on` to enable that ISA extension.

* https://gitlab.com/qemu-project/qemu/-/issues/1447

I've also seen it mentioned elsewhere that if the QEMU target architecture is specified (I'm not sure where exactly - e.g. whether this means at configuration/build time or runtime) with `g` then this will expand to `imafdc_zicsr_zifencei`.

peter-samir-luis, 2023-06-24:
thanks so much for your help,
i tried this :
> For example - something like:
> 
> `qemu-riscv32 -cpu rv32,Zicsr=on -g 3333 test`

but did not work 

> If still in doubt, you might want to ask about RISC-V QEMU issues in the relevant QEMU forums.

I'm not sure I can ask them through these links or how ?

> Contact
> The QEMU community can be contacted in a number of ways, with the two
> main methods being email and IRC
> 
> mailto:qemu-devel@nongnu.org
> https://lists.nongnu.org/mailman/listinfo/qemu-devel
> #qemu on irc.oftc.net
> 
> Information on additional methods of contacting the community can be
> found online via the QEMU website:
> 
> https://wiki.qemu.org/Contribute/StartHere

TommyMurphyTM1234, 2023-06-24:
> thanks so much for your help, i tried this :
> 
> > For example - something like:
> > `qemu-riscv32 -cpu rv32,Zicsr=on -g 3333 test`
> 
> but did not work

You don't say what happened.
 
> > If still in doubt, you might want to ask about RISC-V QEMU issues in the relevant QEMU forums.
> 
> I'm not sure I can ask them through these links 

Obviously you can contact them using those links.

> > Contact
> > The QEMU community can be contacted in a number of ways, with the two
> > main methods being email and IRC
> > [mailto:qemu-devel@nongnu.org](mailto:qemu-devel@nongnu.org)
> > https://lists.nongnu.org/mailman/listinfo/qemu-devel
> > #qemu on irc.oftc.net
> > Information on additional methods of contacting the community can be
> > found online via the QEMU website:
> > https://wiki.qemu.org/Contribute/StartHere

peter-samir-luis, 2023-06-24:
> You don't say what happened.
> E.g. any error messages?

the same error message
`Program received signal SIGILL, Illegal instruction. main () at main.S:2 2	main: csrw mepc, t0`

TommyMurphyTM1234, 2023-06-25:
I don't know why it doesn't seem possible to get the QEMU built by the `riscv-gnu-toolchain` bare-metal/newlib toolchain build to deal with CSR instructions.

I get the same behaviour as you do.

I build the toolchain and QEMU as follows:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32i_zicsr --with-abi=ilp32
make 
make build-sim
```
Then, using this trivial program:
```
/* hello.c */
#include <stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello World\n");
    asm("csrr t0, misa");
    return 0;
}
```
I get this:
```
cd installed-tools/bin

./riscv32-unknown-elf-gcc hello.c -o hello.exe

./qemu-riscv32 -cpu rv32,Zicsr=on hello.exe
Hello World
Illegal instruction

./qemu-riscv32 -cpu rv32,Zicsr=on hello.exe
Hello World
Illegal instruction
```
As shown by passing garbage, it's definitely parsing the `-cpu` options but they don't seem to be taking effect for some reason:
```
./qemu-riscv32 -cpu foo,Zicsr=on hello.exe
qemu-riscv32: unable to find CPU model 'foo'

./qemu-riscv32 -cpu rv32,bar=on hello.exe
qemu-riscv32: can't apply global rv32-riscv-cpu.bar=on: Property 'rv32-riscv-cpu.bar' not found
```
Perhaps QEMU needs to be configured and built in some way different to how the `riscv-gnu-toolchain` (bare-metal/newlib) build does it in order to cope with CSR instructions - I simply don't know. And. as I said, I find it very difficult to fine comprehensive and authoritative documentation on QEMU RISC-V support. And especially on the user mode (?) ``qemu-riscvXX` as opposed to the full-system (?) `qemu-system-riscvXX` variant.

TommyMurphyTM1234, 2023-06-25:
I build QEMU directly from the QEMU repo and still got the same result - illegal instruction when the `csrr t0, misa` instruction is part of the executable.

peter-samir-luis, 2023-06-25:
thanks for your effort to help,
I built qemu-system-riscv32 before i  know that i should use qemu-riscv32 (user mode) to be compatible with the toolchain .
I built qemu-system-riscv32 as follows  : 
```
wget https://download.qemu.org/qemu-8.0.2.tar.xz
tar xvJf qemu-8.0.2.tar.xz
cd qemu-8.0.2/
sudo apt-get install pkg-config
sudo apt-get install libglib2.0-dev libgthread-2.0-dev
sudo apt-get install libpixman-1-dev
./configure --target-list=riscv32-softmmu && make
make
sudo make install
```
**it supports CSR instructions .**
but I got  the problem that I tried to explain in the first two posts .

> this is the command i used
> 
> riscv32-unknown-elf-gcc hello.c -o hello
> qemu-system-riscv32 **-machine virt** -cpu rv32,pmp=false -smp 1 -s -S -nographic -bios none -kernel hello
> 
> in another session
> riscv32-unknown-elf-gdb hello -ex "target remote :1234"

i think the problem of qemu-system-riscv32 is that the default start address for the RISC-V **Virt machine** in QEMU system is 0x80000000 ,meanwhile the default linker of the toolchain loads the code in different address that fits in qemu user mode  (correct me if I'm wrong).

I think if I searched for linker fits with qemu-system-32 virt machine ,this may solve the problem with qemu system

TommyMurphyTM1234, 2023-06-25:
> i think the problem of qemu-system-riscv32 is that the default start address for the RISC-V **Virt machine** in QEMU system is 0x80000000

Can you clarify where this information comes from, please?

danielhb, 2023-06-26:
Hi! I'm posting here due to Tommy post in the QEMU mailing list  about the problem he noticed with qemu-riscv32, which is described a few comments above:

https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1278#issuecomment-1606108826

First, 'Zicsr' is always enabled in the 'rv32' CPU, so you don't need to enable it manually. What happened there is that QEMU linux-user (qemu-riscv32/64) doesn't have access to several CSRs. 'misa' is one of them. If you try with a  CSR that is enabled for linux-user mode, let's say 'cycle', it works. Feel fee to try it out this example:

```
#include <stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello World\n");
    asm("csrr t0, cycle");
    return 0;
}
```

This will not SIGILL using the 'qemu-riscv32' binary from the riscv-gnu-toolchain. We don't have a list of linux-user CSRs in the documentation, but you can check out the available CSRs for this mode in QEMU source, target/riscv/csr.c, csr_ops[]. I'll post here for convenience (based on upstream QEMU):

```
riscv_csr_operations csr_ops[CSR_TABLE_SIZE] = {
    /* User Floating-Point CSRs */
    [CSR_FFLAGS]   = { "fflags",   fs,     read_fflags,  write_fflags },
    [CSR_FRM]      = { "frm",      fs,     read_frm,     write_frm    },
    [CSR_FCSR]     = { "fcsr",     fs,     read_fcsr,    write_fcsr   },
    /* Vector CSRs */
    [CSR_VSTART]   = { "vstart",   vs,     read_vstart,  write_vstart },
    [CSR_VXSAT]    = { "vxsat",    vs,     read_vxsat,   write_vxsat  },
    [CSR_VXRM]     = { "vxrm",     vs,     read_vxrm,    write_vxrm   },
    [CSR_VCSR]     = { "vcsr",     vs,     read_vcsr,    write_vcsr   },
    [CSR_VL]       = { "vl",       vs,     read_vl                    },
    [CSR_VTYPE]    = { "vtype",    vs,     read_vtype                 },
    [CSR_VLENB]    = { "vlenb",    vs,     read_vlenb                 },
    /* User Timers and Counters */
    [CSR_CYCLE]    = { "cycle",    ctr,    read_hpmcounter  },
    [CSR_INSTRET]  = { "instret",  ctr,    read_hpmcounter  },
    [CSR_CYCLEH]   = { "cycleh",   ctr32,  read_hpmcounterh },
    [CSR_INSTRETH] = { "instreth", ctr32,  read_hpmcounterh },

    /*
     * In privileged mode, the monitor will have to emulate TIME CSRs only if
     * rdtime callback is not provided by machine/platform emulation.
     */
    [CSR_TIME]  = { "time",  ctr,   read_time  },
    [CSR_TIMEH] = { "timeh", ctr32, read_timeh },

    /* Crypto Extension */
    [CSR_SEED] = { "seed", seed, NULL, NULL, rmw_seed },

    /* Zcmt Extension */
    [CSR_JVT] = {"jvt", zcmt, read_jvt, write_jvt},
    
    #if !defined(CONFIG_USER_ONLY)
```
 
 All CSRs after the "!defined" are not available for qemu-riscv32.  Hope this clarifies why you're getting a SIGILL up there. Thanks,
 
 Daniel

TommyMurphyTM1234, 2023-06-26:
Hi @danielhb - thanks a lot for taking the time to look at this issue, for spotting my mistake (trying to access `misa` from user mode which, of course, isn't possible and is something that I really should have twigged myself), and explaining the behaviour in such detail. Much appreciated. Thanks again.

peter-samir-luis, 2023-06-26:
> > i think the problem of qemu-system-riscv32 is that the default start address for the RISC-V **Virt machine** in QEMU system is 0x80000000
> 
> Can you clarify where this information comes from, please?

I did not find clear documentation from Qemu but  found many links say that 
- [https://danielmangum.com/posts/risc-v-bytes-privilege-levels/](https://danielmangum.com/posts/risc-v-bytes-privilege-levels/)

Quote from the link:
> `.text` sections from all input object files we provide to the linker and combine them into our output executable .text section at address 0x80000000, which is where the virt machine we are going to use with QEMU [**maps its DRAM**](https://github.com/qemu/qemu/blob/1bd88c4542e97f49955c142f8dc04dd32df9e91f/hw/riscv/virt.c#L60).

- [https://osblog.stephenmarz.com/ch1.html](https://osblog.stephenmarz.com/ch1.html)

Quote from the link:
> That is--start at the physical memory address 0x8000_0000. Luckily, QEMU reads ELF files, so it will know to put our code at that address. Throughout this process, we will glean a lot of information by looking at the qemu/hw/riscv/virt.c source code contained in QEMU. First, the memory map is listed right in the beginning:
> ```
> static const struct MemmapEntry {
> 	hwaddr base;
> 	hwaddr size;
> } virt_memmap[] = {
> 	[VIRT_DEBUG] =       {        0x0,         0x100 },
> 	[VIRT_MROM] =        {     0x1000,       0x11000 },
> 	[VIRT_TEST] =        {   0x100000,        0x1000 },
> 	[VIRT_CLINT] =       {  0x2000000,       0x10000 },
> 	[VIRT_PLIC] =        {  0xc000000,     0x4000000 },
> 	[VIRT_UART0] =       { 0x10000000,         0x100 },
> 	[VIRT_VIRTIO] =      { 0x10001000,        0x1000 },
> 	[VIRT_DRAM] =        { 0x80000000,           0x0 },
> 	[VIRT_PCIE_MMIO] =   { 0x40000000,    0x40000000 },
> 	[VIRT_PCIE_PIO] =    { 0x03000000,    0x00010000 },
> 	[VIRT_PCIE_ECAM] =   { 0x30000000,    0x10000000 },
> };
> ```
> So, our machine starts at byte 0 of the DRAM (VIRT_DRAM), which starts at address 0x8000_0000.

- [http://rcore-os.cn/rCore-Tutorial-deploy/docs/lab-2/guide/part-2.html](http://rcore-os.cn/rCore-Tutorial-deploy/docs/lab-2/guide/part-2.html)

Quote from the link:
> we know that the starting physical address of DRAM physical memory specified by QEMU is 0x80000000. In QEMU, you can use to -m specify the size of RAM, the default is 128 MB. Therefore, the default DRAM physical memory address range is [0x80000000, 0x88000000).

- FreeRTOS Demo  RISC-V_RV32_QEMU_VIRT_GCC uses a linker with address 0x80000000

[https://github.com/FreeRTOS/FreeRTOS/blob/8f3233e0a0730b16f4fed3669d89954cbdd39ea2/FreeRTOS/Demo/RISC-V_RV32_QEMU_VIRT_GCC/build/gcc/fake_rom.ld#L7](https://github.com/FreeRTOS/FreeRTOS/blob/8f3233e0a0730b16f4fed3669d89954cbdd39ea2/FreeRTOS/Demo/RISC-V_RV32_QEMU_VIRT_GCC/build/gcc/fake_rom.ld#L7)
> ```
> MEMORY
> {
> 	/* Fake ROM area */
> 	rom (rxa) : ORIGIN = 0x80000000, LENGTH = 512K
> 	ram (wxa) : ORIGIN = 0x80080000, LENGTH = 512K
> }
> ```
and when i used qemu-system-riscv32 i found the same behavior that the main code is loaded at the address of 0x80000000 or higher.