riscv-collab/riscv-gnu-toolchain #1279: `README.md` is misleading on Linux/musl toolchain capabilities

TommyMurphyTM1234, 2023-06-19:
* https://github.com/riscv-collab/riscv-gnu-toolchain#installation-newliblinux-multilib

> The musl compiler (riscv64-unknown-linux-musl-) will only be able to target 64-bit systems due to limitations in the upstream musl architecture support. 

This is not strictly correct. Any RISC-V toolchain can compile code for any XLEN (and combination of ratified/formalised extensions). This remains the case for the Linux/musl toolchain - it can also compile code for rv32 and is not restricted to rv64. However [musl support is indeed currently restricted to rv64](https://wiki.musl-libc.org/supported-platforms.html#:~:text=riscv64%20(%20support%20added%20since%201.1.23%20release%20)%20currently%20softfp%2C%20single%2Dprecision%2C%20double%2Dprecision%20floating%20point%20supported) so when linking with the standard library/startup code this will only work when targeting rv64 (and the relevant rv64 multilib is available). If targeting rv32 then link errors will occur given the lact of musl rv32 support. But it is not correct to say that the Linux/musl toolchain _"will only be able to target 64-bit systems"_.

> The `--enable-multilib` flag therefore does not actually enable multilib support for musl libc.

This is also not strictly correct. Yes - the default set of multilibs defined by [the default `t-linux-multilib`](https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/t-linux-multilib) contains both rv64 and rv32 `arch/abi` libs so a toolchain build using that will fail on the attempted compilation of musl for rv32 (musl code contains rv64 specific instructions - [for example here](https://git.musl-libc.org/cgit/musl/tree/arch/riscv64/atomic_arch.h#n29)). But it is perfectly feasible to override the default `t-link-multilib` with a custom one that, for example, contains only rv64 `arch/abi` libs and this sort of multilib toolchain build will succeed for Linux/musl.

Also refer to this crosstool-NG issue where I first encountered the issue:

* https://github.com/crosstool-ng/crosstool-ng/issues/1981

A tangential issue - how and why was the default set of Linux multilibs arrived at - namely:

* `march=rv32imac/mabi=ilp32`
* `march=rv32imafdc/mabi=ilp32d`
* `march=rv64imac/mabi=lp64`
* `march=rv64imafdc/mabi=lp64d`

Other than the fact that it encompasses `rvXXgc` it seems slightly arbitrary.
And why both 32 and 64 bit when there are few, if any (?), RISC-V Linux systems that can support code for both XLENs?

cmuellner, 2024-08-26:
I am happy to merge a PR that improves the description in the readme.

I would probably add a simple statement that clearly states that "musl + rv32 => does not work and is not supported." on top.
Below, I would add the details you listed above and the effects that can be observed if one still attempts to go this route.

TommyMurphyTM1234, 2024-10-12:
This PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1571

renders this issue moot so I am closing it.

closed 2024-10-12 by TommyMurphyTM1234