riscv-collab/riscv-gnu-toolchain #1280: Assembler fault: unrecognized opcode....

lifong, 2023-06-21:
### **Operation description**
1. Successfully built the toolchain according to readme in macos:
    $ ./configure --prefix=/opt/riscv
    $ make
2. While compiling a FreeRTOS's Demo, the assembler fault occurred:
[image]

### **Compile log**
```
RISC-V-Qemu-virt_GCC:lifong$ riscv64-unknown-elf-gcc -v -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c main.c -o build/main.o
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: /Volumes/MD/FreeRTOS/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g2ee5e430018 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (g2ee5e430018) 
COLLECT_GCC_OPTIONS='-v' '-D' '__riscv_float_abi_soft' '-D' 'portasmHANDLE_INTERRUPT=handle_trap' '-I' '.' '-I' '../Common/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions' '-march=rv32imac' '-mabi=ilp32' '-mcmodel=medany' '-Wall' '-fmessage-length=0' '-ffunction-sections' '-fdata-sections' '-fno-builtin-printf' '-O2' '-MMD' '-MP' '-c' '-o' 'build/main.o' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imac' '-dumpdir' 'build/'
 /opt/riscv/libexec/gcc/riscv64-unknown-elf/12.2.0/cc1 -quiet -v -I . -I ../Common/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -MMD build/main.d -MP -MQ build/main.o -D __riscv_float_abi_soft -D portasmHANDLE_INTERRUPT=handle_trap main.c -quiet -dumpdir build/ -dumpbase main.c -dumpbase-ext .c -march=rv32imac -mabi=ilp32 -mcmodel=medany -mtune=rocket -misa-spec=20191213 -march=rv32imac -O2 -Wall -version -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -o /var/folders/kq/d2j4fhss5dj1f_msqg3mp5zm0000gn/T//ccGe2wSw.s
GNU C17 (g2ee5e430018) version 12.2.0 (riscv64-unknown-elf)
	compiled by GNU C version Apple LLVM 13.0.0 (clang-1300.0.29.30), GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version isl-0.26-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/riscv/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/riscv/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 .
 ../Common/include
 /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include
 /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V
 /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions
 /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/include
 /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/include-fixed
 /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C17 (g2ee5e430018) version 12.2.0 (riscv64-unknown-elf)
	compiled by GNU C version Apple LLVM 13.0.0 (clang-1300.0.29.30), GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version isl-0.26-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 5e51289261cd73db237fbea9eaf8cf6c
COLLECT_GCC_OPTIONS='-v' '-D' '__riscv_float_abi_soft' '-D' 'portasmHANDLE_INTERRUPT=handle_trap' '-I' '.' '-I' '../Common/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V' '-I' '/Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions' '-march=rv32imac' '-mabi=ilp32' '-mcmodel=medany' '-Wall' '-fmessage-length=0' '-ffunction-sections' '-fdata-sections' '-fno-builtin-printf' '-O2' '-MMD' '-MP' '-c' '-o' 'build/main.o' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imac' '-dumpdir' 'build/'
 /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/as -v -I . -I ../Common/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/include -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V -I /Volumes/MD/FreeRTOS/FreeRTOS_202212.01/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions --traditional-format -march=rv32imac -march=rv32imac -mabi=ilp32 -misa-spec=20191213 -o build/main.o /var/folders/kq/d2j4fhss5dj1f_msqg3mp5zm0000gn/T//ccGe2wSw.s
GNU assembler version 2.40.0 (riscv64-unknown-elf) using BFD version (GNU Binutils) 2.40.0.20230214
main.c: Assembler messages:
main.c:53: Error: unrecognized opcode `csrw mtvec,a5', extension `zicsr' required
main.c:84: Error: unrecognized opcode `csrc mstatus,8', extension `zicsr' required
main.c:111: Error: unrecognized opcode `csrc mstatus,8', extension `zicsr' required
main.c:125: Error: unrecognized opcode `csrc mstatus,8', extension `zicsr' required
```

TommyMurphyTM1234, 2023-06-21:
This is not a `riscv-gnu-toolchain` bug.

The error message is telling you exactly what the issue is:

> `main.c:53: Error: unrecognized opcode 'csrw mtvec,a5', extension 'zicsr' required`

The bulld script is specifying this arch:

> `-march=rv32imac`

This does not include the `zicsr` extension so CSR instructions such as `csrw` are not recognised.
In order to work with a more rececnt RISC-V gcc the build script/`Makefile` needs to be modified to specify:

> `-march=rv32imac_zicsr`

Previously the `zicsr` and `zifencei` extensions were part of the base integer `i` ISA but they were split out from it by the specs and then toolchain implementation in the past year or so,

TommyMurphyTM1234, 2023-06-21:
Closing as invalid.

closed 2023-06-21 by TommyMurphyTM1234