riscv-collab/riscv-gnu-toolchain #1281: floating point registers aren't included in riscv_save\restore routines

Hershkyman, 2023-06-22:
Hi, 
I'm building my simple code with a double floating point and using -Os and -msave_restore flags.

I am seeing the creation and usage of the save\restore functions such as: __riscv_save_0, __riscv_save_2,etc... 
But they do not contain the floating point registers... only the integer registers.

I was wondering if that's an existing option to include the floating point registers into the save\restore routines and maybe my code isn't complex enought to be included?

Or is it simply not supported?

Regards,

Hershkyman.

TommyMurphyTM1234, 2023-06-22:
`-msave-restore` only deals with integer registers as far as I can see:

* https://github.com/gcc-mirror/gcc/blob/master/libgcc/config/riscv/save-restore.S

TommyMurphyTM1234, 2023-06-24:
Closing as invalid.

closed 2023-06-24 by TommyMurphyTM1234