riscv-collab/riscv-gnu-toolchain #1286: Is `riscv-gnu-toolchain` `.gitmodules` file redundant?

TommyMurphyTM1234, 2023-06-30:
See here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1285
* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239#issuecomment-1541439054

This works:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

git submodule
-b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c binutils
-c298959ef991b389b64a825f70094738c6a48780 dejagnu
-2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 gcc
-e53a8e8685685c97588f8319d993ea6cd5635e47 gdb
-a704fd9a133bfb10510e18702f48a6a9c88dbbd5 glibc
-185b81e034ba60081023b6e59504dfffb560f3e3 llvm
-85e0e3519655220688e757b9d5bfd314923548bd musl
-415fdd4279b85eeec9d54775ce13c5c412451e08 newlib
-2efabd3e6604b8a9e8f70baf52f57696680c7855 pk
-621da7789083b80d6f1ff1c0fb499334007b4f51 qemu
-a0298a33e7b2091ba8d9f3a20838d96dc1164cac spike

rm .gitmodules

git submodule
-b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c binutils
-c298959ef991b389b64a825f70094738c6a48780 dejagnu
-2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 gcc
-e53a8e8685685c97588f8319d993ea6cd5635e47 gdb
-a704fd9a133bfb10510e18702f48a6a9c88dbbd5 glibc
-185b81e034ba60081023b6e59504dfffb560f3e3 llvm
-85e0e3519655220688e757b9d5bfd314923548bd musl
-415fdd4279b85eeec9d54775ce13c5c412451e08 newlib
-2efabd3e6604b8a9e8f70baf52f57696680c7855 pk
-621da7789083b80d6f1ff1c0fb499334007b4f51 qemu
-a0298a33e7b2091ba8d9f3a20838d96dc1164cac spike

git submodule update --init --recursive
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/binutils'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/dejagnu'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/gcc'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/gdb'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/glibc'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/llvm'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/musl'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/newlib'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/pk'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/qemu'...
Cloning into '/home/user/Downloads/riscv-gnu-toolchain-tommy/spike'...
Submodule path 'binutils': checked out 'b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c'
Submodule path 'dejagnu': checked out 'c298959ef991b389b64a825f70094738c6a48780'
Submodule path 'gcc': checked out '2ee5e4300186a92ad73f1a1a64cb918dc76c8d67'
Submodule path 'gdb': checked out 'e53a8e8685685c97588f8319d993ea6cd5635e47'
Submodule path 'glibc': checked out 'a704fd9a133bfb10510e18702f48a6a9c88dbbd5'
Submodule path 'llvm': checked out '185b81e034ba60081023b6e59504dfffb560f3e3'
Submodule path 'musl': checked out '85e0e3519655220688e757b9d5bfd314923548bd'
Submodule path 'newlib': checked out '415fdd4279b85eeec9d54775ce13c5c412451e08'
Submodule path 'pk': checked out '2efabd3e6604b8a9e8f70baf52f57696680c7855'
[... 82 lines cut ...]
Submodule path 'qemu/roms/ipxe': checked out '4bd064de239dab2426b31c9789a1f4d78087dc63'
Submodule path 'qemu/roms/openbios': checked out '0e0afae6579c1efe9f0d85505b75ffe989554133'
Submodule path 'qemu/roms/opensbi': checked out '48f91ee9c960f048c4a7d1da4447d31e04931e38'
Submodule path 'qemu/roms/qboot': checked out 'a5300c4949b8d4de2d34bedfaed66793f48ec948'
Submodule path 'qemu/roms/qemu-palcode': checked out '99d9b4dcf27d7fbcbadab71bdc88ef6531baf6bf'
Submodule path 'qemu/roms/seabios': checked out 'd239552ce7220e448ae81f41515138f7b9e3c4db'
Submodule path 'qemu/roms/seabios-hppa': checked out '458626c4c6441045c0612f24313c7cf1f95e71c6'
Submodule path 'qemu/roms/sgabios': checked out 'cbaee52287e5f32373181cff50a00b6c4ac9015a'
Submodule path 'qemu/roms/skiboot': checked out '24a7eb35966d93455520bc2debdd7954314b638b'
Submodule path 'qemu/roms/u-boot': checked out '840658b093976390e9537724f802281c9c8439f5'
Submodule path 'qemu/roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'qemu/roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'qemu/slirp': checked out '9d59bb775d6294c8b447a88512f7bb43f12a25a8'
Submodule path 'qemu/subprojects/libvfio-user': checked out '0b28d205572c80b568a1003db2c8f37ca333e4d7'
Submodule path 'qemu/tests/fp/berkeley-softfloat-3': checked out 'b64af41c3276f97f0e181920400ee056b9c88037'
Submodule path 'qemu/tests/fp/berkeley-testfloat-3': checked out '5a59dcec19327396a011a17fd924aed4fec416b3'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '324355cf62e86fb551408575afb123bac989ac37'
Submodule path 'qemu/ui/keycodemapdb': checked out 'd21009b1c9f94b740ea66be8e48a1d8ad8124023'
Submodule path 'spike': checked out 'a0298a33e7b2091ba8d9f3a20838d96dc1164cac'
```
Similarly, leaving `.gitmodules` in place and changing the details of the branch to be used does not actually affect what branch is used.

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239#issuecomment-1527228397

So, is the `.gitmodules` file redundant, the submodules info stored somewhere else (I can't find any trace of it unless it's in some binary index file or something?), and consequently `.gitmodules` may be a candidate for removal?

patrick-rivos, 2023-06-30:
From here: https://git-scm.com/docs/gitmodules it seems like the `.gitmodules` file could be used to store `update` strategy, `shallow`, `fetchRecurseSubmodules`, but we don't use any of those options at the moment. (And we use the `branch` option but it doesn't have an effect on the submodules?)

The submodule info is actually stored as a tree object, not just the `.gitmodules` file
https://devops.stackexchange.com/questions/5371/where-is-the-git-submodule-revision-actually-stored

kito-cheng, 2023-07-07:
Just found a stackoverflow entry about that: https://stackoverflow.com/questions/59633536/git-submodules-without-gitmodules-file

It says: `What you have here is a broken submodule.`

TommyMurphyTM1234, 2023-07-07:
> Just found a stackoverflow entry about that: https://stackoverflow.com/questions/59633536/git-submodules-without-gitmodules-file
> 
> It says: `What you have here is a broken submodule.`

Thanks @kito-cheng . However I wouldn't trust a StackOverflow answer with only 3 upvotes as being necessarily authoritative. Thus, I don't see this as categorical evidence that the `.gitmodules` file is required (and, in practice, I have shown that it's not) and should be retained.

kito-cheng, 2023-07-07:
What if you just create a branch, remove the file, push to github, and create a PR to see if the github action still work or not?

TommyMurphyTM1234, 2023-07-07:
> What if you just create a branch, remove the file, push to github, and create a PR to see if the github action still work or not?

I was thinking of doing that as a "strawman" PR alright 🙂 so I will look at that later and we can see what happens. Thanks.

TommyMurphyTM1234, 2023-07-11:
I created a `rm_gitmodules` branch off my forked repo https://github.com/TommyMurphyTM1234/riscv-gnu-toolchain and then tried to make the change required for the aforementioned "strawman" PR.

`git submodules` works fine even in the absence of the `.gitmodules` file until after the `git push at which point it fails. 
This doesn't make much sense to me.
```
git clone git@github.com:TommyMurphyTM1234/riscv-gnu-toolchain.git issue-1286
cd issue-1286
git checkout rm_gitmodules
Branch 'rm_gitmodules' set up to track remote branch 'rm_gitmodules' from 'origin'.
Switched to a new branch 'rm_gitmodules'
git pull
Already up to date.
git submodule
-b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c binutils
-c298959ef991b389b64a825f70094738c6a48780 dejagnu
-2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 gcc
-e53a8e8685685c97588f8319d993ea6cd5635e47 gdb
-a704fd9a133bfb10510e18702f48a6a9c88dbbd5 glibc
-185b81e034ba60081023b6e59504dfffb560f3e3 llvm
-85e0e3519655220688e757b9d5bfd314923548bd musl
-415fdd4279b85eeec9d54775ce13c5c412451e08 newlib
-2efabd3e6604b8a9e8f70baf52f57696680c7855 pk
-621da7789083b80d6f1ff1c0fb499334007b4f51 qemu
-71f5a8fd1b1185b7c15151f116aa3fef058bf210 spike
rm .gitmodules 
git submodule
-b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c binutils
-c298959ef991b389b64a825f70094738c6a48780 dejagnu
-2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 gcc
-e53a8e8685685c97588f8319d993ea6cd5635e47 gdb
-a704fd9a133bfb10510e18702f48a6a9c88dbbd5 glibc
-185b81e034ba60081023b6e59504dfffb560f3e3 llvm
-85e0e3519655220688e757b9d5bfd314923548bd musl
-415fdd4279b85eeec9d54775ce13c5c412451e08 newlib
-2efabd3e6604b8a9e8f70baf52f57696680c7855 pk
-621da7789083b80d6f1ff1c0fb499334007b4f51 qemu
-71f5a8fd1b1185b7c15151f116aa3fef058bf210 spike
git add .gitmodules
git submodule
-b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c binutils
-c298959ef991b389b64a825f70094738c6a48780 dejagnu
-2ee5e4300186a92ad73f1a1a64cb918dc76c8d67 gcc
-e53a8e8685685c97588f8319d993ea6cd5635e47 gdb
-a704fd9a133bfb10510e18702f48a6a9c88dbbd5 glibc
-185b81e034ba60081023b6e59504dfffb560f3e3 llvm
-85e0e3519655220688e757b9d5bfd314923548bd musl
-415fdd4279b85eeec9d54775ce13c5c412451e08 newlib
-2efabd3e6604b8a9e8f70baf52f57696680c7855 pk
-621da7789083b80d6f1ff1c0fb499334007b4f51 qemu
-71f5a8fd1b1185b7c15151f116aa3fef058bf210 spike
git commit -m "Remove .gitmodules file - see https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1286"
[rm_gitmodules 2553eab] Remove .gitmodules file - see https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1286
 1 file changed, 42 deletions(-)
 delete mode 100644 .gitmodules
git push
Enumerating objects: 3, done.
Counting objects: 100% (3/3), done.
Delta compression using up to 4 threads
Compressing objects: 100% (2/2), done.
Writing objects: 100% (2/2), 289 bytes | 289.00 KiB/s, done.
Total 2 (delta 1), reused 0 (delta 0)
remote: Resolving deltas: 100% (1/1), completed with 1 local object.
To github.com:TommyMurphyTM1234/riscv-gnu-toolchain.git
   d0d0730..2553eab  rm_gitmodules -> rm_gitmodules
git submodule
fatal: no submodule mapping found in .gitmodules for path 'binutils'
```

TommyMurphyTM1234, 2023-07-15:
While I still don't fully understand the role of `.gitmodules` here - and, for example, why changes to its contents such as the submodule branch seem to have no obvious effect - it does look like it is still required so I am closing this issue as redundant and will not proceed with the aforementioned PR since it would seem to jue break things. I guess that contributors just need to remember to update `.gitmodules` when necessary when updating submodules in order to avoid confusing mismatched info.

closed 2023-07-15 by TommyMurphyTM1234