riscv-collab/riscv-gnu-toolchain #1288: Are Zc* extensions supported ?

zengderui, 2023-07-04:
I'm getting the following errors when enabling Zc*:

```
% riscv32-unknown-elf-gcc -march=rv32gc_zcmt -mabi=ilp32 -static -nostdlib -nostartfiles -Tlink.ld test.s -o test.elf
Assembler messages:
Error: rv32imafdc_zicsr_zifencei_zcmt: unknown prefixed ISA extension `zcmt'
test.s:89: Error: unrecognized opcode `cm.jt 3'
```

TommyMurphyTM1234, 2023-07-04:
You need to explain how exactly you're building your toolchain by providing the `git clone`, `configure` and `make` commands that you're using.

TommyMurphyTM1234, 2023-07-05:
Is there anybody (@kito-cheng, @cmuellner, @simonpcook perhaps?) who might be able to clarify the situation with regard to GCC toolchain support for the `Zc*` extensions and if/when it will be available from upstream GCC et. al.?

pz9115, 2023-07-05:
> Is there anybody (@kito-cheng, @cmuellner, @simonpcook perhaps?) who might be able to clarify the situation with regard to GCC toolchain support for the `Zc*` extensions and if/when it will be available from upstream GCC et. al.?

We are doing some work to support ZC* into upstream recently, but still exists some Zcmt bugs fixing now, hopefully, it will take one month for all ZC* extension upstream I think.

TommyMurphyTM1234, 2023-07-05:
> We are doing some work to support ZC* into upstream recently, but still exists some Zcmt bugs fixing now, hopefully, it will take one month for all ZC* extension upstream I think.

Thanks a lot @pz9115 - 

> We are doing some work to support ZC* into upstream recently, but still exists some Zcmt bugs fixing now, hopefully, it will take one month for all ZC* extension upstream I think.

Thanks a lot @pz9115. So what repo and branch is the `Zc*` work being done on?

simonpcook, 2023-07-05:
> Thanks a lot @pz9115. So what repo and branch is the `Zc*` work being done on?

The Zc work is being done in the OpenHW Group fork of [gcc](https://github.com/openhwgroup/corev-gcc) and [binutils-gdb](https://github.com/openhwgroup/corev-binutils-gdb). The `development` branch has the latest support for all the Zc* extensions, other than zcmt, which is still being reported in binutils-gdb. For that, the `development-eddf4096b97` branch has the latest work that was done on that extension (it does support the other zc* extensions, its just based on an earlier version of binutils).

TommyMurphyTM1234, 2023-07-05:
> The Zc work is being done in the OpenHW Group fork of [gcc](https://github.com/openhwgroup/corev-gcc) and [binutils-gdb](https://github.com/openhwgroup/corev-binutils-gdb). The `development` branch [...]

Thanks a lot @simonpcook. So this should build the latest `Zc*/Zcmt` support?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/openhwgroup/corev-gcc corev-gcc
git clone https://github.com/openhwgroup/corev-binutils-gdb corev-binutils-gdb -b development-eddf4096b97
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/corev-gcc --with-binutils-src=`pwd`/corev-binutils-gdb
make 
```

simonpcook, 2023-07-07:
Yes I believe that should work.

TommyMurphyTM1234, 2023-07-10:
> -march=rv32gc_zcmt -mabi=ilp32

Just to confirm, that worked and I was able to do this:
```
# test.S
cm.jt 3
```
```
./riscv64-unknown-elf-gcc arch=rv32i_zcmt -mabi=ilp32 -c test.S 
./riscv64-unknown-elf-objdump -D test.o

test.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <.text>:
   0:	a00e                	cm.jt	3

Disassembly of section .riscv.attributes:

00000000 <.riscv.attributes>:
   0:	2841                	jal	90 <.riscv.attributes+0x90>
   2:	0000                	unimp
   4:	7200                	.2byte	0x7200
   6:	7369                	lui	t1,0xffffa
   8:	01007663          	bgeu	zero,a6,14 <.riscv.attributes+0x14>
   c:	001e                	c.slli	zero,0x7
   e:	0000                	unimp
  10:	7205                	lui	tp,0xfffe1
  12:	3376                	.2byte	0x3376
  14:	6932                	.2byte	0x6932
  16:	7032                	.2byte	0x7032
  18:	5f31                	li	t5,-20
  1a:	637a                	.2byte	0x637a
  1c:	3161                	jal	fffffca4 <.riscv.attributes+0xfffffca4>
  1e:	3070                	.2byte	0x3070
  20:	7a5f 6d63 3174      	.byte	0x5f, 0x7a, 0x63, 0x6d, 0x74, 0x31
  26:	3070                	.2byte	0x3070
	...
```
Note that this - mentioned in the original post - does not work:
```
./riscv64-unknown-elf-gcc -march=rv32gc_zcmt -mabi=ilp32 -c test.S 
Assembler messages:
Error: Zcm* is not compatible with the `c' extension.
```
Closing this issue since, once `Zc*` support has been fully implemented in the OpenHW Group fork it will presumably eventually be upstreamed to the master GCC/binutils projects.

junambi, 2023-07-24:
Thank you for the information, this helped, i am able to compile with the extensions.

Micronana, 2023-08-07:
Hi，Pz9115
I just compile the gcc toolchain which support Zc* as you mentioned there:  https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1288.  
As Zc is used to reduce code size,  but when I use this Zc* gcc to compile coremark with   rv32imc,  the code size is 2% larger than than the normal gcc in master branch.    So even I compile the program with rv32im_cmt,  the code size is still larger than normal gcc with rv32imc option. 
In summary,  the Zc toolchain I tried has not reduce the code size.  Is it  expected?

TommyMurphyTM1234, 2023-08-07:
> Hi, @pz9115 I just compile the gcc toolchain which support Zc* as you mentioned there: #1288. As Zc is used to reduce code size, but when I use this Zc* gcc to compile coremark with rv32imc, the [...]

You might want to cross check your results against the info here under "Benchmark results" 

* https://github.com/riscv/riscv-code-size-reduction

and, if they still don't make sense to you, maybe open an issue here:

* https://github.com/riscv/riscv-code-size-reduction/issues

pz9115, 2023-08-07:
> Hi, @pz9115 I just compile the gcc toolchain which support Zc* as you mentioned there: #1288. As Zc is used to reduce code size, but when I use this Zc* gcc to compile coremark with rv32imc, the [...]

For Zc* extension, it's an addition to the C extension. Since C extension will compress some instructions,
you should compare single Zc* extension without C extension, like imafd_zcb vs imafd, or both enable C extension to dismiss the compress feature influence, like imafdc vs imafdc_zcb.

TommyMurphyTM1234, 2023-08-07:
> So even I compile the program with rv32im_cmt 

BTW I presume that this was meant to be `rc32im_zcmt` otherwise it's an invalid `arch` specifier?