riscv-collab/riscv-gnu-toolchain #1289: why i get 16bit instruction?

androny1012, 2023-07-05:
RISCV_GCC_OPTS = -static -Wa,-march=rv32i_zicsr -mabi=ilp32 -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf

the result in dump
000003c4 <divide>:
 3c4:	4785                	li	a5,1
 3c6:	00a05863          	blez	a0,3d6 <divide+0x12>
 3ca:	8d0d                	sub	a0,a0,a1
 3cc:	0785                	add	a5,a5,1
 3ce:	fea04ee3          	bgtz	a0,3ca <divide+0x6>
 3d2:	853e                	mv	a0,a5
 3d4:	8082                	ret
 3d6:	4505                	li	a0,1
 3d8:	8082                	ret

code in .c
int divide(int x, int y) {
	if (x<=0) return 1;
	else return (1+divide(x-y,y));
}

Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/12.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: ./riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g2ee5e430018 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32imca_zicsr --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (g2ee5e430018)

i want to compile the program and get instruction 32bit but not 16bit, what should i do?

TommyMurphyTM1234, 2023-07-05:
You don't clearly say how exactly you're compiling and linking your program, where in the disassembly listing that snippet comes from, where you obtained or how you built your toolchain etc. so it's difficult to offer any useful comment.

androny1012, 2023-07-06:
Thank you very much!  rv32imca_zicsr/ilp32 contain compressed instructions, i reconfigure and get right instruction.

closed 2023-07-06 by androny1012

TommyMurphyTM1234, 2023-07-06:
> Thank you very much! rv32imca_zicsr/ilp32 contain compressed instructions, i reconfigure and get right instruction.

You're welcome. Glad it helped.