riscv-collab/riscv-gnu-toolchain #1290: Decimal to floating point and floating point to decimal conversion

amithmath, 2023-07-10:
Hello,

Can you please point me codes where numbers in human readable format to floating point conversion and floating point to human readable format conversion?  For example: 3.2444422 -->404FA4F1-->3.2444422

Regards,
Amith

kito-cheng, 2023-07-10:
You need online one https://www.h-schmidt.net/FloatConverter/IEEE754.html or in your C code https://stackoverflow.com/questions/45228925/how-to-print-float-as-hex-bytes-format-in-c ?

closed 2023-07-10 by TommyMurphyTM1234

amithmath, 2023-07-10:
> You need online one https://www.h-schmidt.net/FloatConverter/IEEE754.html or in your C code https://stackoverflow.com/questions/45228925/how-to-print-float-as-hex-bytes-format-in-c ?

I am aware of the above link. This is not problem, well, I am looking for a code in compiler how decimal to FP are converted and vice versa.

for example:

int main ()
{
float a;

a= 3.2444422;

return 0;

}
when you generate asm code, a in assembler code converted to 404FA4F1. So I  am looking for a code which part of compiler and where it is converted.

TommyMurphyTM1234, 2023-07-10:
Have you familiarised yourself with the RISC-V F/D floating point extension specifications and the IEEE 754-2008 standard?

* https://riscv.org/technical/specifications/
* https://ieeexplore.ieee.org/document/4610935