riscv-collab/riscv-gnu-toolchain #1291: Are P* extensions supported ?

manju1205, 2023-07-10:
I am getting below error ::

Assembler messages:
Error: cannot find default versions of the ISA extension `p'
/tmp/ccKNSLNd.s:3: Error: cannot find default versions of the ISA extension `p'

I have used below build commands:: 

./configure --prefix=$RISCV --with-arch=rv32gp --with-abi=ilp32 
make

TommyMurphyTM1234, 2023-07-10:
I'm not sure if anything has changed since this?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1190

kito-cheng, 2023-07-10:
Not supported, this repo are using upstream GNU toolchain source tree, and upstream GNU toolchain not supported yet since it never get ratified, but you could use external GCC and binutils source tree from PLCT or Andes https://github.com/andestech/gcc.

closed 2023-07-10 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-07-12:
Please clarify and provide...

1. Details of how you're building your toolchain
2. Your example code and details of how you're compiling it - ideally a small self-contained source file and associated `Makefile`

manju1205, 2023-07-12:
My sincere apologies. Test output is generated, "ll" command was not listing it.
I am getting below error with spike simulator ::
spike pk Test
error: bad --isa option 'rv64imafd_zpn'. unsupported extension: zpn

Please suggest me how to add "zpn" to spike?

TommyMurphyTM1234, 2023-07-12:
> My sincere apologies. Test output is generated, "ll" command was not listing it. 

Do you mean that your previous post can be ignored after all?

> I am getting below error with spike simulator :: spike pk Test error: bad --isa option 'rv64imafd_zpn'. unsupported extension: zpn

I suspect that Spike only supports ratified extensions.

I don't know if the PLCT fork of Spike might support `Zpn` like their GCC and binutils forks?

* https://github.com/plctlab/plct-spike

If not and there is no other Spike fork that implements `Zp*` support then you'll probably have to ask over at the Spike repo about how to approach implementing support yourself:

* https://github.com/riscv-software-src/riscv-isa-sim

manju1205, 2023-07-12:
yes, my post wrt output not generated can be ignored.
"P extension, v0.9.2" is listed among supported ISA in https://github.com/riscv-software-src/riscv-isa-sim.
If not spike can I try any other simulator?

TommyMurphyTM1234, 2023-07-12:
> "P extension, v0.9.2" is listed among supported ISA in https://github.com/riscv-software-src/riscv-isa-sim.

You'll have to ask about this and why your attempt to use it fails over at the Spike and/or P extension pages:

* https://github.com/riscv-software-src/riscv-isa-sim
* https://github.com/riscv/riscv-p-spec

> If not spike can I try any other simulator?

Sorry - I've no idea myself.
Maybe check with some of the other RISC-V simulator projects to see if they offer what you're looking for?

* https://michaeljclark.github.io/software.html

TommyMurphyTM1234, 2023-07-12:
Looks to me like `Zpn` should be supported by Spike:

* https://github.com/riscv-software-src/riscv-isa-sim/blob/ed5dccb2911f3b2dda731e5fe65e6c03ad0b46fb/riscv/isa_parser.cc#L74

How exactly did you build it?
Maybe you need to pass the requested ISA string in a different form to Spike?

TommyMurphyTM1234, 2023-07-12:
> I am getting below error with spike simulator :: 
> ```
> spike pk Test 
> error: bad --isa option 'rv64imafd_zpn'. unsupported extension: zpn
> ```
> 
> Please suggest me how to add "zpn" to spike?

Looks like you need to do this:
```
spike --isa=rv64imafdp ...
```

manju1205, 2023-07-16:
Thanks, with "--isa" option I could get spike recognizing the "zpn" sub-ext. 
Below branch do they support zbpbo and zpsfoperand extns? 

git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
git clone https://github.com/plctlab/riscv-binutils-gdb -b riscv-binutils-p-ext riscv-binutils-p-ext

TommyMurphyTM1234, 2023-07-16:
> Thanks, with "--isa" option I could get spike recognizing the "zpn" sub-ext.

How? You said earlier that you got this:

> error: bad --isa option 'rv64imafd_zpn'. unsupported extension: zpn

> Below branch do they support zbpbo and zpsfoperand extns?

Yes - as I said here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1291#issuecomment-1633103738

manju1205, 2023-07-24:
I used to get error on running below command without --isa option::

spike  pk Test  --- error
spike --isa=rv64imafdp pk Test -- NO error...

TommyMurphyTM1234, 2023-07-24:
> I used to get error on running below command without --isa option::
> 
> spike pk Test --- error spike --isa=rv64imafdp pk Test -- NO error...

Yes, that's basically what I said above:

> Yes - as I said here, if you use `--isa=rvXX...p`:
> 
> * [Are P* extensions supported ? #1291 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1291#issuecomment-1633103738)