riscv-collab/riscv-gnu-toolchain #1292: RV32IM softcore stalling on new versions of the toolchain.

HakamAtassi, 2023-07-10:
A softcore used for a research project has seemed to stop functioning correctly after an update to the toolchain (not entirely sure which aspect of the toolchain). The issue is demonstrated by a stall in the RTL simulation after building and running the project using the latest version of the toolchain (but does not occur with older checkouts). Here is a useful thread regarding the issue: 

https://github.com/pphilippos/simodense/issues/1

Hope this is of use. Take care.

Hakam

pphilippos, 2023-07-10:
Hello, (I wrote the softcore) As I just mentioned in the issue you linked, there is a peculiarity in the way the softcore gets the start of the text segment (manually) as it was for research purposes. I plan to make it more consistent/automated, but this does not relate to the toolchain itself. Hope this helps, Best regards

TommyMurphyTM1234, 2023-07-10:
You should post (attach) the verbose log (`openocd -d3`) for the working and non working OpenOCD. That should narrow down where the issue is.

TommyMurphyTM1234, 2023-07-10:
You need to debug the issue with your core in order to narrow down what's going on/wrong. There really isn't much to go on in the linked thread. And there's nothing at all to indicate any issue/bug with the toolchain as far as I can see. In particular there's no standalone self-contained test case that illustrates a problem.

HakamAtassi, 2023-07-10:
Hello. I stand corrected- there was a change in the compilation that required a change in the testbench but no real bug to be fixed. Thanks for the help. Take care

closed 2023-07-10 by HakamAtassi