riscv-collab/riscv-gnu-toolchain #1294: submodule gcc clone failed

HUJIYONG, 2023-07-12:
While doing `make linux` on ubuntu, a fatal error saying "fetch-pack:invalid index-output...clone of .../gcc.git into submodule path .../gcc failed" occurs. Maybe it's resulted from the pack size, but I have no way to fix it.

TommyMurphyTM1234, 2023-07-12:
Is this issue reproducible? Every time? I did several builds in the past few days and never encountered this problem. Are you sure that it's not due to local internet connectivity issues? I'm in Europe (Ireland, to be specific) in case that matters.

TommyMurphyTM1234, 2023-07-12:
> ```
> git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain
> ```

That worked fine for me just now.
I can only assume that your problem is a transient one or realted to local network issues?
```
git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
POST git-upload-pack (gzip 3915 to 1947 bytes)
remote: Enumerating objects: 8462, done.
remote: Counting objects: 100% (8458/8458), done.
remote: Compressing objects: 100% (3755/3755), done.
remote: Total 8462 (delta 4220), reused 8396 (delta 4197), pack-reused 4
Receiving objects: 100% (8462/8462), 5.24 MiB | 8.63 MiB/s, done.
Resolving deltas: 100% (4220/4220), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/binutils'...
remote: Enumerating objects: 1178814, done.        
remote: Counting objects: 100% (35806/35806), done.        
remote: Compressing objects: 100% (35502/35502), done.        
remote: Total 1178814 (delta 2177), reused 319 (delta 292), pack-reused 1143008        
Receiving objects: 100% (1178814/1178814), 421.88 MiB | 11.30 MiB/s, done.
Resolving deltas: 100% (979549/979549), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/dejagnu'...
remote: Counting objects: 9018, done.        
remote: Compressing objects: 100% (2007/2007), done.        
remote: Total 9018 (delta 6891), reused 9010 (delta 6883)        
Receiving objects: 100% (9018/9018), 4.66 MiB | 3.72 MiB/s, done.
Resolving deltas: 100% (6891/6891), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/gcc'...
remote: Enumerating objects: 2890711, done.        
remote: Counting objects: 100% (198555/198555), done.        
remote: Compressing objects: 100% (16509/16509), done.        
remote: Total 2890711 (delta 191152), reused 185240 (delta 181964), pack-reused 2692156        
Receiving objects: 100% (2890711/2890711), 1.11 GiB | 10.74 MiB/s, done.
Resolving deltas: 100% (2376354/2376354), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/gdb'...
remote: Enumerating objects: 1178834, done.        
remote: Counting objects: 100% (35826/35826), done.        
remote: Compressing objects: 100% (35522/35522), done.        
remote: Total 1178834 (delta 2217), reused 319 (delta 292), pack-reused 1143008        
Receiving objects: 100% (1178834/1178834), 421.74 MiB | 11.58 MiB/s, done.
Resolving deltas: 100% (979589/979589), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/glibc'...
remote: Enumerating objects: 701793, done.        
remote: Counting objects: 100% (23450/23450), done.        
remote: Compressing objects: 100% (7576/7576), done.        
remote: Total 701793 (delta 17106), reused 17892 (delta 15591), pack-reused 678343        
Receiving objects: 100% (701793/701793), 227.62 MiB | 11.39 MiB/s, done.
Resolving deltas: 100% (597843/597843), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/llvm'...
remote: Enumerating objects: 5489419, done.        
remote: Counting objects: 100% (2009/2009), done.        
remote: Compressing objects: 100% (714/714), done.        
remote: Total 5489419 (delta 1345), reused 1702 (delta 1289), pack-reused 5487410        
Receiving objects: 100% (5489419/5489419), 1.89 GiB | 12.31 MiB/s, done.
Resolving deltas: 100% (4487541/4487541), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/musl'...
remote: Counting objects: 37180, done.        
[... 278 lines cut ...]
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/pyca-cryptography': checked out '09403100de2f6f1cdd0d484dcb8e620f1c335c8f'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Library/BrotliCustomDecompressLib/brotli': checked out 'f4153a09f87cbb9c826d8fc12c74642bb2d879ea'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Universal/RegularExpressionDxe/oniguruma': checked out 'abfc8ff81df4067f309032467785e06975678f0d'
Submodule path 'qemu/roms/edk2/RedfishPkg/Library/JsonLib/jansson': checked out 'e9ebfa7e77a6bee77df44e096b100e7131044059'
Submodule path 'qemu/roms/edk2/UnitTestFrameworkPkg/Library/CmockaLib/cmocka': checked out '1cc9cde3448cdd2e000886a26acf1caac2db7cf1'
Submodule path 'qemu/roms/ipxe': checked out '4bd064de239dab2426b31c9789a1f4d78087dc63'
Submodule path 'qemu/roms/openbios': checked out '0e0afae6579c1efe9f0d85505b75ffe989554133'
Submodule path 'qemu/roms/opensbi': checked out '48f91ee9c960f048c4a7d1da4447d31e04931e38'
Submodule path 'qemu/roms/qboot': checked out 'a5300c4949b8d4de2d34bedfaed66793f48ec948'
[... cut ...]
Submodule path 'qemu/subprojects/libvfio-user': checked out '0b28d205572c80b568a1003db2c8f37ca333e4d7'
Submodule path 'qemu/tests/fp/berkeley-softfloat-3': checked out 'b64af41c3276f97f0e181920400ee056b9c88037'
Submodule path 'qemu/tests/fp/berkeley-testfloat-3': checked out '5a59dcec19327396a011a17fd924aed4fec416b3'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '324355cf62e86fb551408575afb123bac989ac37'
Submodule path 'qemu/ui/keycodemapdb': checked out 'd21009b1c9f94b740ea66be8e48a1d8ad8124023'
Submodule path 'spike': checked out '71f5a8fd1b1185b7c15151f116aa3fef058bf210'
```

HUJIYONG, 2023-07-12:
I am working on an Azure Ubuntu Server located on East US, so it's not probably resulted from internet connectivity.

TommyMurphyTM1234, 2023-07-12:
> I am working on an Azure Ubuntu Server located on East US, so it's not probably resulted from internet connectivity.

So...

> Is this issue reproducible? Every time? I did several builds in the past few days and never encountered this problem. Are you sure that it's not due to local internet connectivity issues? I'm in Europe (Ireland, to be specific) in case that matters.
> 
> What happens if you do the following?
> 
> ```
> git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain
> ```

HUJIYONG, 2023-07-12:
I'm sorry! I tried to buy a server with better performance today, and the problem was fixed now. It seems to be a problem of git, who costs too much RAM and some other resources  while cloning. **Thank you for the answer!!! 👍 ** And I am going to make an issue for git ~

HUJIYONG, 2023-07-12:
But ..., I remember if specify the branch, and set --depth=1, I can clone gcc successfully:
`git clone -b branch master --single-branch --depth=1 https://gcc.gnu.org/git/gcc.git`.
I wonder if these parameter can be added while cloning your project? I tried but failed in my old server. Maybe I made mistakes...

TommyMurphyTM1234, 2023-07-12:
> And I am going to make an issue for git ~

It may not be Git that is the problem but maybe Azure or something else.

> But ..., I remember if I specify the branch, and set --depth=1, I can clone gcc successfully in my old server:
> git clone -b branch master --single-branch --depth=1 https://gcc.gnu.org/git/gcc.git.
> I wonder if these parameter can be added while cloning your project?

That may be OK in restricted circumstances but many users/developers will want to have the complete log/history of changes available on cloning so I don't think that it would be a good idea to force this on everybody. 

By all means edit your local `riscv-gnu-toolchain` `Makefile` to add this if it makes things more reliable for you and works around limitations of the systems that you use.

TommyMurphyTM1234, 2023-07-12:
Closing this issue as invalid.

closed 2023-07-12 by TommyMurphyTM1234