riscv-collab/riscv-gnu-toolchain #1295: Any compiler switch for align the data variable into 32 bit address boundary

dineshannayya, 2023-07-12:
Hi 

I am looking for way to force the memory location for data variable in 32 bit address boundary

1. Example: Two character variable case
       char  data1 = 0x55;
       char  data2 = 0x66;
              Address: 0x0000   0xXXXXXX55
              Address: 0x0004   0xXXXXXX66
    Not as: 
         Address: 0x0000   0xXXXX6655

2. Example: Two Short variable case
       short  data1 = 0x5566;
       short  data2 = 0x7788;
              Address: 0x0000   0xXXXX5566
              Address: 0x0004   0xXXXX7788
    Not as: 
         Address: 0x0000   0x77885566
Is there way to control the RISCV Gcc to force 32 bit data variable allocation?

cmuellner, 2023-07-12:
Using `struct`s could help you:

```c
typedef struct {
    uint8_t b1;
    uint8_t b2;
    uint8_t b3;
    uint8_t b4;
} myunion_t;

void set_b1(myunion_t *p, uint8_t v)
{
    p->b1 = v;
}

uint8_t get_b3(myunion_t *p)
{
    return p->b3;
}
```

...compiles to:
```
set_b1(myunion_t*, unsigned char):
        sb      a1,0(a0)
        ret
get_b3(myunion_t*):
        lbu     a0,2(a0)
        ret
```

Also, have a look at [the `packed` variable attribute](https://gcc.gnu.org/onlinedocs/gcc/Common-Variable-Attributes.html).

I'm closing this ticket, because this question is a general compiler usage question and does not affect this repo. A better place to ask you question is the [gcc-help mailing list](https://gcc.gnu.org/mailman/listinfo/gcc-help).

closed 2023-07-12 by cmuellner

TommyMurphyTM1234, 2023-07-12:
The GCC `alignment` attribute may also be of use:

* https://gcc.gnu.org/onlinedocs/gcc/Common-Variable-Attributes.html

Or you may be able to force non-standard alignment via a custom linker script.

However, in some (many?) working against the toolchain's default tendancies may result in suboptimal code. So you should double check that what you're attempting to do here is actually absolutely necessary.

dineshannayya, 2023-07-12:
@TommyMurphyTM1234 
I have made open source based Riscv SOC : [Link](https://github.com/dineshannayya/riscduino) , Chip showing promising result but unaligned Data write are failing and it's blocking the progress.

TommyMurphyTM1234, 2023-07-12:
Perhaps a simpler way to do what you want?

> 1. Example: Two character variable case
>    char  data1 = 0x55;
>    char  data2 = 0x66;
>    Address: 0x0000   0xXXXXXX55
>    Address: 0x0004   0xXXXXXX66
>    Not as:
>    Address: 0x0000   0xXXXX6655
```
uint8_t data1 = 0x55;
uint8_t dummy1_1;
uint8_t dummy1_2;
uint8_t dummy1_3;

uint8_t data2 = 0x66;
uint8_t dummy2_1;
uint8_t dummy2_2;
uint8_t dummy2_3;
```
> 2. Example: Two Short variable case
>    short  data1 = 0x5566;
>    short  data2 = 0x7788;
>    Address: 0x0000   0xXXXX5566
>    Address: 0x0004   0xXXXX7788
>    Not as:
>    Address: 0x0000   0x77885566
```
uint16_t data1 = 0x5566;
uint16_t dummy1_1;

uint16_t data2 = 0x7788;
uint16_t dummy2_1;
```
Depending on the size in bytes of C standard integer types as per your original code isn't a good idea.