riscv-collab/riscv-gnu-toolchain #1296: ABI Mismatch Error | Unable to link for RV32I_ZICSR with ilp32 ABI

shaurya0406, 2023-07-15:
I have built the toolchain with multilib option for **riscv64-unknown-elf** by running the `make` command

Here is the output for `riscv64-unknown-elf-gcc --print-multi-lib`

```
.;
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
rv64imafdc/lp64d;@march=rv64imafdc@mabi=lp64d
```

I am able to compile all my individual C and ASM files with these flags:

```
PREFIX=riscv64-unknown-elf-
LINKER_SCRIPT=rv32i_linkerscript.ld
FPIC="-fpic"       
MARCH="rv32i_zicsr"
MABI="ilp32"
GCC_FLAGS_COMPILE+="-march=$MARCH -mabi=$MABI -ffunction-sections -fdata-sections $FPIC "
```

But I am not able to link all the source files with entry.s

`GCC_FLAGS_LINKER+="-march=$MARCH -mabi=$MABI -Wl,--gc-sections -Wl,-melf32lriscv -Wl,-Ttext,0 -Wl,-Map,linker.map -lm -lc -lgcc -lc -nostartfiles $FPIC"`

The final Linking GCC Command looks like this:

`${PREFIX}gcc ${GCC_FLAGS_LINKER} -T ${LINKER_SCRIPT} ./obj/entry.out ./obj/*.o  -o ${ONAME}.bin -lm`

Here is the link to the linker script: [Linker Script](https://github.com/shaurya0406/RISCV_C_tests/blob/main/rv32i_linkerscript.ld)

I am getting ABI mismatch error:

```
PROCESSING: test_gpio.c
	compiling assembly file.....
	compiled all files in library and extra

/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(divsf3.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(divsf3.o)
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(divdf3.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(divdf3.o)
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(div.o)
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(_clzsi2.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(_clzsi2.o)
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(_clz.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(_clz.o)
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: test.bin(.text): relocation ".L9+0x0 (type R_RISCV_RVC_JUMP)" goes out of range
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(divsf3.o): file class ELFCLASS64 incompatible with ELFCLASS32
/opt/riscv_multilib/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
```

I can see, that it is defaulting to 64-bit, but how should I set the target to rv32i with zicsr extension? Its not looking in the rv32i/ilp32 folder for libgcc.a

Platform: Ubuntu : Linux Virtual Machine running via Parallels on a M2 Macbook.

Any help would be much appreciated. Thanks!

TommyMurphyTM1234, 2023-07-15:
Your toolchain does not have any multilibs for `rv32i_zicsr/ilp32` so is falling back on the toolchain defaults for `rv64gc/lp64d` hence the link errors.

If you want a toolchain with support for `rv32i_zicsr/ilp32` then you need to do something like the following:
```
# riscv32-unknown-elf toolchain supporting rv32i_zicsr/ilp32 by default
./configure ... --with-arch=rv32i_zicsr --with-abi=ilp32

# riscv64-unknown-elf toolchain supporting rv64gc/lp64d by default and rv32i_zicsr/ilp32 via multilib
./configure ... --with-multilib-generator="rv32i_zicsr-ilp32--"

# riscv64-unknown-elf toolchain supporting rv64gc/lp64d by default and various other arch/abis including rv32i_zicsr/ilp32 via multilibs
./configure ... --with-multilib-generator="rv32i_zicsr-ilp32--;rv32i-ilp32--;rv32im-ilp32--;rv32iac-ilp32--;rv32imac-ilp32--;rv32imafc-ilp32f--;rv64imac-lp64--;rv64imafdc-lp64d--"
```

shaurya0406, 2023-07-15:
Thanks for the reply, will rebuild the toolchain and get back!

I will use the second command for configuration:

```
# riscv64-unknown-elf toolchain supporting rv64gc/lp64d by default and rv32i_zicsr/ilp32 via multilib
./configure ... --with-multilib-generator="rv32i_zicsr-ilp32--"
```

shaurya0406, 2023-07-15:
I tried rebuilding the toolchain with this command:
`./configure --prefix=/opt/riscv32i_zicsr --with-multilib-generator="rv32i_zicsr-ilp32--"`

And after linking, it gave these errors:

```
riscv64-unknown-elf-gcc: error: unrecognized command-line option '-fpic-march=rv32i_zicsr'
riscv64-unknown-elf-gcc: error: unrecognized command-line option '-fpic-march=rv32i_zicsr'
riscv64-unknown-elf-gcc: error: unrecognized command-line option '-fpic-march=rv32i_zicsr'
riscv64-unknown-elf-gcc: error: unrecognized command-line option '-nostartfiles-march=rv32i_zicsr'
```

TommyMurphyTM1234, 2023-07-15:
> `-fpic-march=rv32i_zicsr`

This is not a valid GCC flag.
The build script or `Makefile` for your program code must be erroneous.

shaurya0406, 2023-07-15:
I am able to generate binary from assembly files. But with C files, I am facing issues.
These are my commands for C src files.

```
PREFIX=riscv64-unknown-elf-
LINKER_SCRIPT=rv32i_linkerscript.ld
FPIC="-fpic"       
MARCH="rv32i_zicsr"
MABI="ilp32"
GCC_FLAGS_COMPILE+="-march=$MARCH -mabi=$MABI -ffunction-sections -fdata-sections $FPIC "
```

`GCC_FLAGS_LINKER+="-march=$MARCH -mabi=$MABI -Wl,--gc-sections -Wl,-melf32lriscv -Wl,-Ttext,0 -Wl,-Map,linker.map -lm -lc -lgcc -lc -nostartfiles $FPIC"`

The final Linking GCC Command looks like this:

`${PREFIX}gcc ${GCC_FLAGS_LINKER} -T ${LINKER_SCRIPT} ./obj/entry.out ./obj/*.o  -o ${ONAME}.bin -lm`

Here is the link to the linker script: [Linker Script](https://github.com/shaurya0406/RISCV_C_tests/blob/main/rv32i_linkerscript.ld)

TommyMurphyTM1234, 2023-07-15:
> I am able to generate binary from assembly files. But with C files, I am facing issues. These are my commands for C src files.

What issues?
You don't describe any.

> ```
> PREFIX=riscv64-unknown-elf-
> LINKER_SCRIPT=rv32i_linkerscript.ld
> FPIC="-fpic"
> MARCH="rv32i_zicsr"
> MABI="ilp32"
> GCC_FLAGS_COMPILE+="-march=$MARCH -mabi=$MABI -ffunction-sections [...]

I don't see how the above can be generating the erroneous command line options that you posted previously.

Seems to me that you are not describing what you're doing exactly so some info is missing.

TommyMurphyTM1234, 2023-07-15:
> I tried rebuilding the toolchain with this command: `./configure --prefix=/opt/riscv32i_zicsr --with-multilib-generator="rv32i_zicsr-ilp32--"`
>
> And after linking, it gave these errors:
>
> ``` [...]

If you are using this to generate the command line options:

> `GCC_FLAGS_LINKER+="-march=$MARCH -mabi=$MABI -Wl,--gc-sections -Wl,-melf32lriscv -Wl,-Ttext,0 -Wl,-Map,linker.map -lm -lc -lgcc -lc -nostartfiles $FPIC"`

Then perhaps `$GCC_FLAGS_LINKER` already contains some options and they are being prepended to the additional options with no space before `-march`?

But when you don't post all relevant info it's all guesswork on my part.

Either way there is no issue with the toolchain evident here as far as I can see.

shaurya0406, 2023-07-15:
Yes, you pointed that correctly, I just removed the + form the GCC_FLAGS_LINKER variable.
And the toolchain is now working.

Thank you so much for your time. I was stuck at this for the last 2 days.

closed 2023-07-15 by TommyMurphyTM1234