riscv-collab/riscv-gnu-toolchain #1298: Are P* extensions supported for 32-bit toolchain ?

vedanttalnikar, 2023-07-24:
I am getting below error:
riscv32-unknown-elf-ar rc ../libc.a *.o
riscv32-unknown-elf-ar: ../stdlib/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../time/lib.a: No such file or directory
riscv32-unknown-elf-ranlib libc.a

I used following build procedure:

$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
$ cd riscv-gnu-toolchain/
as mentioned in #1291 I used following repositories
$ git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
$ git clone https://github.com/plctlab/riscv-binutils-gdb -b riscv-binutils-p-ext riscv-binutils-p-ext

$ ./configure --prefix=$RISCV32  --with-arch=rv32imafd_zpn_zifencei --with-abi=ilp32d --with-gcc-src=`pwd`/riscv-gcc-p-ext --with-binutils-src=`pwd`/riscv-binutils-p-ext
$ make

I have attached the make.log
where you can spot the actual error as mentioned below :
[make.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12146128/make.log)

```
riscv-gnu-toolchain/newlib/newlib/libc/stdlib/rand.c:82:1: error: unrecognizable insn:
   82 | }
      | ^
(insn 19 18 20 2 (set (reg:DI 89)
        (mult:DI (zero_extend:DI (reg:SI 90))
            (zero_extend:DI (reg:SI 91)))) "/manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/newlib/newlib/libc/stdlib
     (nil))
during RTL pass: vregs
/manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/rand.c:82:1: internal compiler error: in ex
0x59ada8 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
    /manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/rtl-error.c:108
0x59adca _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
    /manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/rtl-error.c:116
0x59a2d0 extract_insn(rtx_insn*)
    /manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/recog.c:2294
0x8109ff instantiate_virtual_regs_in_insn
    /manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/function.c:1607
0x8109ff instantiate_virtual_regs
    /manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/function.c:1977
0x8109ff execute
    /manjup/vedant/projects/RISCV/riscv32/riscv-gnu-toolchain/ri
(https://github.com/riscv-collab/riscv-gnu-toolchain/files/12146110/make.log)
scv-gcc-p-ext/gcc/function.c:2026
Please submit a full bug report,
```

Is the current configuration mentioned in configure command supported by the toolchain or anything missing in the procedure ?

kito-cheng, 2023-07-24:
Seem like you hit bug in their compiler implementation, cc @pz9115

TommyMurphyTM1234, 2023-07-24:
> Seem like you hit bug in their compiler implementation, cc @pz9115

It would also be good if PLCT enabled the GitHub Issues section of their repo so that the issue could be reported there rather than here.

pz9115, 2023-07-24:
I had repeated this bug, it seems exist some compatible problems with newlib version. I will check it soon.

pz9115, 2023-07-25:
@vedanttalnikar I had fixed this bug, now you can try build again on RV32, thanks for your report.

vedanttalnikar, 2023-07-25:
@pz9115 Thanks for the quick support. 
Now the build is working fine for me :)

TommyMurphyTM1234, 2023-10-18:
@cmuellner / @kito-cheng / others ...

Can/should this issue be closed? 

It doesn't really seem to be a valid `riscv-gnu-toolchain` issue as far as I can see.
It seems to be an issue with the PCLT fork of the tools:

* https://github.com/plctlab/riscv-gcc

And/or this long open PR

* https://github.com/riscv-collab/riscv-gcc/pull/258

on the `riscv-gcc` repo (`riscv-gcc-experiment` branch):

* https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-experiment

It's still unclear to me what the precise purpose of the `riscv-gcc` repo is, what active development is ongoing there, and why/when anybody would use it instead of upstream gcc.

cmuellner, 2023-10-18:
Yes, we can close this issue because it is not directly related to the riscv-gnu-toolchain repo or repositories that are referenced by the submodules.

It is a valid issue (thank you for reporting), but the issue tracker of this repo is not the right place. We don't need to be too strict on this, but after three months it is reasonable to close it here. Especially since the issue got recognition from the owner of the affected code.

closed 2023-10-18 by cmuellner