riscv-collab/riscv-gnu-toolchain #1300: LLVM build failed on macOS

jerryzj, 2023-07-30:
Somehow I can't build LLVM compiler-rt on macOS...
```
FAILED: CMakeFiles/clang_rt.builtins_i386_osx.dir/i386/fp_mode.c.o
/Volumes/Develop/riscv-gnu-toolchain/build/build-llvm-linux/./bin/clang --target=riscv64-unknown-linux-gnu   -O3 -DNDEBUG -arch i386 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -mmacosx-version-min=10.7 -fPIC -O3 -fvisibility=hidden -DVISIBILITY_HIDDEN -Wall -fomit-frame-pointer -Werror=format-nonliteral -DDONT_DEFINE_EPRINTF -arch i386 -MD -MT CMakeFiles/clang_rt.builtins_i386_osx.dir/i386/fp_mode.c.o -MF CMakeFiles/clang_rt.builtins_i386_osx.dir/i386/fp_mode.c.o.d -o CMakeFiles/clang_rt.builtins_i386_osx.dir/i386/fp_mode.c.o -c /Volumes/Develop/riscv-gnu-toolchain/llvm/compiler-rt/lib/builtins/i386/fp_mode.c
clang-16: warning: argument unused during compilation: '-arch i386' [-Wunused-command-line-argument]
clang-16: warning: argument unused during compilation: '-mmacosx-version-min=10.7' [-Wunused-command-line-argument]
clang-16: warning: argument unused during compilation: '-arch i386' [-Wunused-command-line-argument]
/Volumes/Develop/riscv-gnu-toolchain/llvm/compiler-rt/lib/builtins/i386/fp_mode.c:37:38: error: invalid output constraint '+t' in asm
  __asm__ __volatile__ ("fdivs %1" : "+t" (f) : "m" (g));
                                     ^
1 error generated
```

TommyMurphyTM1234, 2023-07-30:
You need to clarify exactly how you're trying to build the toolchain - i.e your `git clone`, `configure` and `make` commands. And attach a (compressed) full build log.

This looks strangen if you're trying to build a RISC-V toolchain:

> `-arch i386`
> `

closed 2023-07-30 by jerryzj

TommyMurphyTM1234, 2023-07-30:
Sorry, I mistakenly closed the issue earlier so reopened it.
But I see that you've now closed it yourself.
Did you figure out what the issue was?