riscv-collab/riscv-gnu-toolchain #1302: Specify that root is required to write in /opt/riscv

ReFruity, 2023-08-07:
Since people usually just run commands you provide, it would be best to specify that `sudo` in `sudo make linux` is required in order to use the provided path of `/opt/riscv`. I am aware that it says the path must be writable but still.

TommyMurphyTM1234, 2023-08-07:
Not everybody will install into a "system" directory.
Some people will install into a local directory that is writeable without `sudo`.
Does the readme really need to remind Linux users that `sudo` is needed in certain cases?
If you think it does then please submit a PR to modify the instructions appropriately.
For what it's worth, this PR was integrated recently in an attempt to clarify the issue that you raise here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1171

ReFruity, 2023-08-07:
True, some people would want a non-system directory, in that case specifying `sudo` for everyone would not be the best decision. Maybe we should specify that using `sudo` is required in case you want to use the directory provided in readme? The only thing that irks me is that following the default path readme provides gives errors.

TommyMurphyTM1234, 2024-06-06:
> The only thing that irks me is that following the default path readme provides gives errors.

When a build has completed and reached the install stage then, if `root` privileges are required in order to write to the `prefix` directory specified at `configure` time then this will be reported to the user who can then just do an incremental `sudo make ...` to complete the installation.

closed 2024-06-06 by TommyMurphyTM1234