riscv-collab/riscv-gnu-toolchain #1303: custom encode/decode

amithmath, 2023-08-08:
Hi,

I have posit hardware for addition, multiplication, division, etc. for 32 bit risc-v. In order to encode and decode posit numbers in compiler, one has to modify encode_ieee_single() and  decode_ieee_single() functions in real.c?

Thanks,
Amit

TommyMurphyTM1234, 2023-08-08:
Your post is unclear.
What exactly are you asking?
What exactly is the problem with the `riscv-gnu-toolchain` that you're encountering?

TommyMurphyTM1234, 2023-08-08:
If the question is how to add Unum/Posit support to GCC then that's not really a RISC-V specific question and should be raised in the GCC mailing list or other relevant forum.

TommyMurphyTM1234, 2023-08-08:
This closed issue may contain some relevant information:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/890

closed 2023-08-09 by amithmath