riscv-collab/riscv-gnu-toolchain #1305: Superfluous AS_HELP_STRING in configure.ac

jeremybennett, 2023-08-08:
Commit bb74af5a874c8b521ae0159ba4d892af9df93d09 enabled LLVM & QEMU build flow, but introduced a bug in changing `configure.ac`.  It also seems that `configure` was not correctly regenerated.

In the following block
```m4
AS_IF([test "x$enable_debug_info" != xyes],
        [AS_HELP_STRING([--disable-debug-info],
		[build glibc and musl without debug infromation])],
        [disable_debug_info=yes],
        [disable_debug_info=no]
        )

```
The `AS_HELP_STRING` is not required, and is treated as the "if" expression with ` [disable_debug_info=yes]` as the "else" expression.  The code should be
```m4
AS_IF([test "x$enable_debug_info" != xyes],
        [disable_debug_info=yes],
        [disable_debug_info=no]
        )
```

As a side note, it would be useful to add a comment at the top of the file, noting that it needs to be regenerated using Autoconf 2.69.  Newer versions will complain!

I shall submit a PR to fix this very shortly.

jeremybennett, 2023-08-08:
PR #1306 submitted.

kito-cheng, 2023-08-10:
Fixed via #1306

closed 2023-08-10 by kito-cheng

jeremybennett, 2023-08-14:
Thanks @kito-cheng