riscv-collab/riscv-gnu-toolchain #1307: make linux cannot generate RISCV64-unknown-linux-gnu-g ++

Lzingnow, 2023-08-11:
The compilation process under centos is as follows:

../configure --prefix=tool_chain/install/riscv-gnu-toolchain-rvv1p0 --with-arch=rv64gcv --with-abi=lp64d --with-multilib-generator="rv64gcv-lp64d--"

make -j16

make linux -j16

But the make linux command can only generate the gcc compiler and not g++, and the erf compiler is complete.

[image]

What is the reason for this？

TommyMurphyTM1234, 2023-08-11:
I'm not sure that the build scripts are set up to do sequential builds like that as opposed to doing completely separate builds such as like this:
```
# Configure and build the newlib toolchain
./configure --prefix=`pwd`/newlib-installed-tools --with-arch=rv64gcv --with-abi=lp64d --with-multilib-generator="rv64gcv-lp64d--"
make -j16 2>&1 | tee build-newlib.log

# Clean
make distclean

# Configure and build the Linux toolchain
./configure --prefix=`pwd`/linux-installed-tools --with-arch=rv64gcv --with-abi=lp64d --with-multilib-generator="rv64gcv-lp64d--"
make linux -j16 2>&1 | tee build-linux.log
```
Maybe capture, compress and attach your build logs in case they help shed some light on why the Linux C++ compiler doesn't get built when you try the sequential builds with no `make distclean` between them?

TommyMurphyTM1234, 2023-08-11:
I actually tried this (no parallel building) but hit a different problem:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1307
cd issue-1307
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d --with-multilib-generator="rv64gcv-lp64d--"
make 2>&1 | tee build-newlib.log
make linux 2>&1 | tee build-linux.log
...
make[3]: Entering directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
# If this is the top-level multilib, build all the other
# multilibs.
make[4]: Entering directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
make[5]: Entering directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/lib32/ilp32/libgcc'
# If this is the top-level multilib, build all the other
# multilibs.
ln -s ../../../../.././gcc/libgcc/enable-execute-stack-empty.c enable-execute-stack.c
ln -s ../../../../.././gcc/libgcc/unwind-generic.h unwind.h
ln -s ../../../../.././gcc/libgcc/config/riscv/linux-unwind.h md-unwind-support.h
ln -s ../../../../.././gcc/libgcc/config/riscv/sfp-machine.h sfp-machine.h
ln -s ../../../../.././gcc/libgcc/gthr-posix.h gthr-default.h
DEFINES='' HEADERS='' \
	../../../../.././gcc/libgcc/mkheader.sh > tmp-libgcc_tm.h
/bin/bash ../../../../.././gcc/libgcc/../move-if-change tmp-libgcc_tm.h libgcc_tm.h
echo timestamp > libgcc_tm.stamp
/home/user/Downloads/issue-1307/build-gcc-linux-stage2/./gcc/xgcc -B/home/user/Downloads/issue-1307/build-gcc-linux-stage2/./gcc/ -B/home/user/Downloads/issue-1307/installed-tools/riscv64-unknown-linux-gnu/bin/ -B/home/user/Downloads/issue-1307/installed-tools/riscv64-unknown-linux-gnu/lib/ -isystem /home/user/Downloads/issue-1307/installed-tools/riscv64-unknown-linux-gnu/include -isystem /home/user/Downloads/issue-1307/installed-tools/riscv64-unknown-linux-gnu/sys-include    -O2 -mcmodel=medlow -march=rv32imac -mabi=ilp32 -O2  -O2    -mcmodel=medlow -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -fPIC -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector   -fPIC -I. -I. -I../../../.././gcc -I../../../../.././gcc/libgcc -I../../../../.././gcc/libgcc/. -I../../../../.././gcc/libgcc/../gcc -I../../../../.././gcc/libgcc/../include  -DHAVE_CC_TLS   -o _negdi2.o -MT _negdi2.o -MD -MP -MF _negdi2.dep -DL_negdi2 -c ../../../../.././gcc/libgcc/libgcc2.c -fvisibility=hidden -DHIDE_EXPORTS
In file included from /home/user/Downloads/issue-1307/installed-tools/sysroot/usr/include/features.h:515,
                 from /home/user/Downloads/issue-1307/installed-tools/sysroot/usr/include/bits/libc-header-start.h:33,
                 from /home/user/Downloads/issue-1307/installed-tools/sysroot/usr/include/stdio.h:27,
                 from ../../../../.././gcc/libgcc/../gcc/tsystem.h:87,
                 from ../../../../.././gcc/libgcc/libgcc2.c:27:
/home/user/Downloads/issue-1307/installed-tools/sysroot/usr/include/gnu/stubs.h:8:11: fatal error: gnu/stubs-ilp32.h: No such file or directory
    8 | # include <gnu/stubs-ilp32.h>
      |           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:501: _negdi2.o] Error 1
make[5]: Leaving directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/lib32/ilp32/libgcc'
make[4]: *** [Makefile:1211: multi-do] Error 1
make[4]: Leaving directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
make[3]: *** [Makefile:127: all-multi] Error 2
make[3]: Leaving directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
make[2]: *** [Makefile:13336: all-target-libgcc] Error 2
make[2]: Leaving directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2'
make[1]: *** [Makefile:1030: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/issue-1307/build-gcc-linux-stage2'
make: *** [Makefile:456: stamps/build-gcc-linux-stage2] Error 2
```
Build logs:

* [build-newlib.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12322762/build-newlib.log.tar.gz)
* [build-linux.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12322763/build-linux.log.tar.gz)

Lzingnow, 2023-08-11:
Sorry, I can't download the log at the moment, but I found critical errors in the log file when I was making linux, as follows:

`/riscv-gnu-toolchain/build-glibc-linux-rv64gcv-lp64d/intl/plural.c:69:25: error: static declaration of '__gettextlex' follows non-static declaration
   69 | #define yylex           __gettextlex
      |                         ^~~~~~~~~~~~
plural.y:57:12: note: in expansion of macro 'yylex'
   57 | static int yylex (YYSTYPE *lval, struct parse_args *arg);
      |            ^~~~~
`
I think this is the reason why g++ cannot be generated.

Above is the result of my use of "make distclean", which is consistent with the continuous results.

TommyMurphyTM1234, 2023-08-11:
Slightly off topic but this:
```
./configure --prefix=tool_chain/install/riscv-gnu-toolchain-rvv1p0 --with-arch=rv64gcv --with-abi=lp64d --with-multilib-generator="rv64gcv-lp64d--"
```
is unnecessary. In particular `--with-multilib-generator="rv64gcv-lp64d--"` is pointless since `--with-arch=rv64gcv --with-abi=lp64d` is already specified so `rv64gcv/lp64d` libs/startup code objects will be built automatically without the need to additionally specify `rv64gcv/lp64d` as a multilib.

In relation to the original issue I too am having problems building a Linux toolchain even without doing any other steps first. But I seem to be getting different results/problems to you. I will post back when I have some further info.

TommyMurphyTM1234, 2023-08-11:
This works fine and builds both the Newlib and Linux toolchains - including C++ front ends - and installs them in the one place:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1307
cd issue-1307
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
make
make linux
s -1 installed-tools/bin/*c++
installed-tools/bin/riscv64-unknown-elf-c++
installed-tools/bin/riscv64-unknown-linux-gnu-c++
```

Lzingnow, 2023-08-12:
Thank you very much. Now the normal master branch works fine, but now I want to compile a toolchain that supports rvv1.0. The order of the commands I execute is as follows:

`git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git --recurse-submodules`
`cd riscv-gnu-toolchain`
`cd gcc`
`git checkout tags/releases/gcc-13.1.0`
`cd ..`
`./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d`
`make linux -j$(nproc) 2>&1 | tee build-linux.log`

But it stopped working.
[build-linux.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12327172/build-linux.zip)

TommyMurphyTM1234, 2023-08-12:
> I want to compile a toolchain that supports rvv1.0. 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git --recurse-submodules
> cd riscv-gnu-toolchain
> cd gcc
> git checkout tags/releases/gcc-13.1.0
>```

This doesn't seem to make sense.
GCC 13 only supports RVV 0.11 - and only intrinsics at that.

* https://gcc.gnu.org/gcc-13/changes.html#:~:text=RISC%2DV,most%20of%20implementation.

> **RISC-V**
> 
> - Support for vector intrinsics as specified in [version 0.11 of the RISC-V vector intrinsic specification](https://github.com/riscv-non-isa/rvv-intrinsic-doc/tree/v0.11.x), thanks Ju-Zhe Zhong from [RiVAI](https://rivai-ic.com.cn/) for contributing most of implementation.
> - ...

TommyMurphyTM1234, 2023-08-12:
If you want to build a vector extension enabled toolchain then you should probably review this closed issue:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1301

and, in particular, this post:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1301#issuecomment-1674382150

TommyMurphyTM1234, 2023-08-12:
Since the original issue (failure to build the Linux C++ front end) seems to have been addressed I am closing this and you can open a new issue if you have further build problems. However, please be as detailed and specific as possible about what the issue is, provide (non-parallel build) log files to illustrate it, clarify all steps taken to perform the build and arrive at the anomalous situation etc.

closed 2023-08-12 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-08-12:
> This works fine and builds both the Newlib and Linux toolchains - including C++ front ends - and installs them in the one place:
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1307
> cd issue-1307
> ./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d --enable-multilib
> make
> make linux
> ls -1 installed-tools/bin/*c++
> installed-tools/bin/riscv64-unknown-elf-c++
> installed-tools/bin/riscv64-unknown-linux-gnu-c++
> ```

Just for the record, this also works fine:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1307
cd issue-1307
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d
make
make linux
ls -1 installed-tools/bin/*c++
installed-tools/bin/riscv64-unknown-elf-c++
installed-tools/bin/riscv64-unknown-linux-gnu-c++
```