riscv-collab/riscv-gnu-toolchain #1308: Should the `rvv-next` branch be removed?

TommyMurphyTM1234, 2023-08-12:
It seems to be inactive:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1301#issuecomment-1666601034

and perhaps stale/redundant?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1301#issuecomment-1670942246

Removing it may avoid confusion with people building a vector enabled toolchain from the "wrong" sources?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1301#issuecomment-1670942246

Ditto for the `riscv-gcc` `riscv-gcc-rvv-next` branch?

* https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-rvv-next

TommyMurphyTM1234, 2023-08-23:
Anybody know about this? In particular:

* @cmuellner 
* @kito-cheng 
* @joe-img

TommyMurphyTM1234, 2023-08-23:
> My comment regarding the staleness of that branch was purely based on observation - I'm afraid I don't know the 'official' status on it.

OK - thanks @joe-img.

kito-cheng, 2023-08-23:
I think it's time to retire/remove rvv-next for this repo, upstream gcc has support rvv instrinsic and auto vec now, but I would prefer keep that on riscv-gcc repo, it still useful to compare what's missed on upstream, and it seems used by many different groups.

cmuellner, 2023-10-05:
Yes, the [`rvv-next`](https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next) branch should be removed (in this repo).

@TommyMurphyTM1234: Do you see any reasons why I should not simply remove that branch (i.e. will something break that you are aware of)? If not, I will remove it.

TommyMurphyTM1234, 2023-10-05:
> Yes, the [`rvv-next`](https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next) branch should be removed (in this repo).
> 
> @TommyMurphyTM1234: Do you see any reasons why I should not simply remove that branch (i.e. will something break that you are aware of)? 

No, but, then again, I'm not sure that I'm the best person to ask!  🙂

cmuellner, 2023-10-05:
You are the best to ask, because you will most likely have to deal with the complaints in the issue tracker... ;)

cmuellner, 2023-10-05:
The `rvv-next` branch of this repo has been deleted.

closed 2023-10-09 by TommyMurphyTM1234