riscv-collab/riscv-gnu-toolchain #1309: fatal error: stdio.h: No such file or directory

alanmasu, 2023-08-14:
Hi,
I've setup the toolchain on WSL2 (Ubuntu 22.04.3 LTS) accordingly to README for linux, as architecture I've setted `rv32ia ` and the default ABI.
Now, I'm not able to include standard libraries and every time that I try to include those files an "No such file or directory" error occurs.

It's my configuration that cause this problem or in any case it's true?

PS: I'm not a linux user so, if anyone can explain me in a simple way how to fix it, I'll be grateful
Alan

TommyMurphyTM1234, 2023-08-14:
You need to clarify exactly how/where you obtained the toolchain or how you built it if that's the case - and how you installed it.

Using `...-gcc -v ...` for verbose output will tell you where the toolchain is looking for header files and maybe help investigate why they're not there.

A `tree` mapping of the layout of your installed toolchain might help. If you do generate one then please attach it rather than embedding it in a reply.

What is the code that you're trying to preprocess?

This is obviously not an issue with the toolchain per se, or it would have been reported more widely, but probably with how you built or installed it.

TommyMurphyTM1234, 2023-08-14:
This worked for me:
```
# Install WSL2/Ubuntu on Windows 10
# See: https://learn.microsoft.com/en-us/windows/wsl/install

# Install all available Ubuntu updates
sudo apt update
sudo apt upgrade

# Install riscv-gnu-toolchain prerequisites
# See here: https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
# And here: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1251
sudo apt install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev git python2 expect device-tree-compiler libglib2.0-dev pip

# Clone, configure (for a default arch/abi of rv32ia/ilp32) and make riscv-gnu-toolchain bare-metal/newlib toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32ia --with-abi=ilp32
make

# Compile "Hello world"
cd installed-tools/bin
# Create hello-world.c - see below
./riscv32-unknown-elf-gcc hello-world.c -o hello-world
file hello*
hello-world:   ELF 32-bit LSB executable, UCB RISC-V, soft-float ABI, version 1 (SYSV), statically linked, with debug_info, not stripped
hello-world.c: C source, ASCII text
```
```
# Hello world
#include <stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello world\n");
    return 0;
}
```

alanmasu, 2023-08-14:
Thanks for help, now I'll try to clarify all: 
 > You need to clarify exactly how/where you obtained the toolchain or how you built it if that's the case - and how you installed it.

I've cloned the repo by the command:

```
$ git clone https://github.com/riscv/riscv-gnu-toolchain
```
than I've built it using
```
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv --with-arch=rv32i
sudo make linux
```

> Using ...-gcc -v ... for verbose output will tell you where the toolchain is looking for header files and maybe help investigate why they're not there.

```
Using built-in specs.
COLLECT_GCC=/opt/riscv/bin/riscv32-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-linux-gnu/12.2.0/lto-wrapper
Target: riscv32-unknown-linux-gnu
Configured with: /home/alanmasu/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-newlib --without-headers --disable-shared --disable-threads --with-system-zlib --enable-tls --enable-languages=c --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-bootstrap --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32i --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-O2    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (GCC)
```

> A tree mapping of the layout of your installed toolchain might help. If you do generate one then please attach it rather than embedding it in a reply.

I'm not sure about your request, but I've saved the output of `tree ` in a txt that's attached to this reply: [res.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12339571/res.txt)

> What is the code that you're trying to preprocess?

``` C
#include <stdio.h>
void funzione(char a, char b, char c, char d, char e, char f, char g, char h, char i, char* str);

int main(){
	char str[10]; 
	funzione('a', 'b', 'c','d','e','f','g','h','i', str);
        printf("%s\n", str);
        return 0;
}

void funzione(char a, char b, char c, char d, char e, char f, char g, char h, char i, char* str){
	str[0] = a;
	str[1] = b;
        str[2] = c;
        str[3] = d;
        str[4] = e;
        str[5] = f;
        str[6] = g;
        str[7] = h;
        str[8] = i;
	str[9] = '\0';
}
```
I hope that's enough, in the negative case I'll remain available for others information
Alan

TommyMurphyTM1234, 2023-08-14:
I did what I posted in the previous post and was able to successfully compile/link your test code.

alanmasu, 2023-08-15:
> I've setted rv32ia and the default ABI

Now I'm not sure about the A extension, because I've get some errors about the minimum extension for the ABI, and by heart the default abi needed the extension A, but now with the verbose log seems that I've used only the base architecture.

Anyway now I'm trying to use your suggestion! Later the results!

alanmasu, 2023-08-15:
Ok, as I remember the A extension is required! (although by glibc, and not by ABI)

```
mips nios2 or1k powerpc riscv configure: error: glibc requires the A extension
make: *** [Makefile:389: stamps/build-glibc-linux-rv32i-ilp32] Error 1
```
I'll attach the whole log here [build.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12341948/build.log)

Alan

TommyMurphyTM1234, 2023-08-15:
Sorry - my previous instructions were incorrect. I transcribed the `configure` command incorrectly. It should be:
```
./configure --prefix=/opt/riscv --with-arch=rv32ia --with-abi=ilp32
````
and not
```
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
```
As you discovered, the Linux toolchain or some part of it may have a dependency on the A extension.

Please try again with these instructions:
```
# Install WSL2/Ubuntu on Windows 10
# See: https://learn.microsoft.com/en-us/windows/wsl/install

# Install all available Ubuntu updates
sudo apt update
sudo apt upgrade

# Install riscv-gnu-toolchain prerequisites
# See here: https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
# And here: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1251
sudo apt install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev git python2 expect device-tree-compiler libglib2.0-dev pip

# Clone, configure (for a default arch/abi of rv32ia/ilp32) and make riscv-gnu-toolchain Linux/glibc toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv --with-arch=rv32ia --with-abi=ilp32
sudo make linux 2>&1 | tee build.log

# Compile "Hello world"
# Create hello-world.c - see above: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1309#issuecomment-1678050852
export PATH=/opt/riscv/bin:$PATH
riscv32-unknown-elf-gcc hello-world.c -o hello-world
file hello*
hello-world:   ELF 32-bit LSB executable, UCB RISC-V, soft-float ABI, version 1 (SYSV), statically linked, with debug_info, not stripped
hello-world.c: C source, ASCII text
```

alanmasu, 2023-08-15:
In the meanwhile I'm trying to build with NewLib on WSL (just tried in another virtual machine and seems working).
After that, and if still not working I'll try whit new configuration!

@TommyMurphyTM1234 thanks for your time and help

TommyMurphyTM1234, 2023-08-15:
> In the meanwhile I'm trying to build with NewLib on WSL (just tried in another virtual machine and seems working). After that, and if still not working I'll try whit new configuration!
> 
> @TommyMurphyTM1234 thanks for your time and help

I don't understnad why you are chopping and changing what you're trying to do here - it just complicates things.
You originally wanted to build the Linux toolchain for `rv32ia/lp32`, then mentioned `rv32i`, but now you are switching to the Newlib toochain for some reason and using a different VM.

What exactly are you trying to do and what toolchain do you actually need?

alanmasu, 2023-08-15:
The exactly goal is to build simple applications to test a RISC-V RV32I that I've created in VHDL and now running on a FPGA device; I wrote some code in C, that's compiled for RV32I, disassembled using ...-objdump and loaded into instructions's BRAM on the FPGA. So I'm confused about witch toolchain I need to. 
I would like to "debug" this small CPU and try to develop new features compatible with standard compilers. 

I'm not an expert of compilers/toolchains/operating systems so excuse me for the confusing exposition of problems.

> You originally wanted to build the Linux toolchain
I'm not so sure what I actually wanted to do, "I'm on a Linux VM, build it for Linux no?..." I was think"...No, wasn't the best choice!"

Now, what do you think about what I need to use?

Excuse me another time for my doubts / shortcomings in the subject!
Alan

TommyMurphyTM1234, 2023-08-15:
> The exactly goal is to build simple applications to test a RISC-V RV32I that I've created in VHDL and now running on a FPGA device; I wrote some code in C, that's compiled for RV32I, disassembled [...]

If your target is not running Linux - i.e. is bare-metal or running a lightweight [RT]OS - then the bare-metal/Newlib toolchain is almost certainly the most appropriate:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
sudo make 2>&1 | tee build.log
...
export PATH=/opt/riscv/bin:$PATH
riscv32-unknown-elf-gcc -v
...
```
> I'm not so sure what I actually wanted to do, "I'm on a Linux VM, build it for Linux no?..." I was think, but "...No, wasn't the best choice!"

Bear in mind that you are building a cross compiler in this case.
For example, the above in this post build a toolchain that runs ON Linux (e.g. WSL2/Ubuntu in my case) but generates code FOR a RISC-V bare-metal/Newlib target.

Are you sure that you actually need to build your own toolchain as opposed to using an existing pre-built one that supports `rv32i/ilp32` (possibly via multilibs) such as @ilg-ul's xPack toolchain?

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

Note that if you use this toolchain then you'll need to use something like this:
```
riscv-none-elf-gcc -march=rv32i -mabi=ilp32 ...
```
because it supports `rv32i/ilp32` via multilibs rather than as the native `arch/abi` (if I recall correctly) and this toolchain uses a different prefix (`riscv-none-elf-*` rather than `riscvXX-unknown-elf-`).

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#riscv-none-elf-gcc

TommyMurphyTM1234, 2023-08-15:
Closing this because no issue with `riscv-gnu-toolchain` has been identified.

closed 2023-08-15 by TommyMurphyTM1234

alanmasu, 2023-08-15:
Thanks for your time @TommyMurphyTM1234