riscv-collab/riscv-gnu-toolchain #1311: Compiled program cannot access environment list

ChinYikMing, 2023-08-17:
config of `riscv32-unknown-elf-gcc` : `$ ./configure --prefix=/opt/riscv32 --with-arch=rv32ia --with-abi=ilp32`

I think `a2` register should be point to starting address of environment variable (some address of stack), but `a2` is set to zero before calling `main`, is it normal?

C source file:
```
#include <stdio.h>

int main(int argc, char *argv[], char *envp[]){
        while(*envp){
                printf("%s\n", *envp);
                envp++;
        }

        return 0;
}
```

compiled `_start` label:
```
000100dc <_start>:
   100dc:       00005197                auipc   gp,0x5
   100e0:       dcc18193                add     gp,gp,-564 # 14ea8 <__global_pointer$>
   100e4:       04418513                add     a0,gp,68 # 14eec <__malloc_max_total_mem>
   100e8:       09c18613                add     a2,gp,156 # 14f44 <__BSS_END__>
   100ec:       40a60633                sub     a2,a2,a0
   100f0:       00000593                li      a1,0
   100f4:       194000ef                jal     10288 <memset>
   100f8:       00000517                auipc   a0,0x0
   100fc:       4f450513                add     a0,a0,1268 # 105ec <atexit>
   10100:       00050863                beqz    a0,10110 <_start+0x34>
   10104:       00001517                auipc   a0,0x1
   10108:       89050513                add     a0,a0,-1904 # 10994 <__libc_fini_array>
   1010c:       4e0000ef                jal     105ec <atexit>
   10110:       0dc000ef                jal     101ec <__libc_init_array>
   10114:       00012503                lw      a0,0(sp)
   10118:       00410593                add     a1,sp,4
   1011c:       00000613                li      a2,0
   10120:       06c000ef                jal     1018c <main>
   10124:       f71ff06f                j       10094 <exit>
```

TommyMurphyTM1234, 2023-08-17:
You're using the bare-metal toolchain so it's up to you to provide any OS capabilities, such as environment/variables, as they're not going to be present automatically. What is your target - hardware, [RT]OS etc.?

TommyMurphyTM1234, 2023-08-17:
Using bare-metal GCC toolchain is similar to using the Linux GCC toolchain in "freestanding" mode as outlined here:

* http://cs107e.github.io/guides/gcc/

ChinYikMing, 2023-08-17:
> You're using the bare-metal toolchain so it's up to you to provide any OS capabilities, such as environment/variables, as they're not going to be present automatically. What is your target - hardware, [RT]OS etc.?

I see. I am developing an emulator of RV32 ISA on Linux. Does "Linux cross-compiler" is gonna help?

TommyMurphyTM1234, 2023-08-17:
> I see. I am developing an emulator of RV32 ISA on Linux.

You mean the emulator will run on Linux?
Or the emulated RV32 target will run Linux?
Or both?
Or something else?

> Does "Linux cross-compiler" is gonna help?

Only if your emulated RISC-V runs Linux.

Nothing is going to magic up OS capabilities such as the environment/variables on a bare-metal target.
As I said, and as per the link that I posted, these need to be provided by you or by an [RT]OS etc. if needed.

You might need to read up on and better understand the nature of and differences between bare-metal, RTOS and rich OS (e.g. Linux) systems.

Anyway -this issue does not identify any issue with the `riscv-gnu-toolchain` tools so I am closing it.
Please use other forums such as the mailing lists for general questions/support.

closed 2023-08-17 by TommyMurphyTM1234

ChinYikMing, 2023-08-17:
I think I got it, thanks a lot!